MarlinFirmware/Marlin #3558: Speed movement for each motor

CoderSquirrel, 2016-04-18:
Hi i wanna know if its possible, in code, to set the movement speed for each motor.
e.g: i want the E to do 80 mm/s, the X 255 mm/s and the Y 15mm/s.
I tried to use  `plan_buffer_line(current_position[(X_AXIS)], current_position[(Y_AXIS)], current_position[(Z_AXIS)], current_position[E_AXIS], 100.0, active_extruder);` and changing the feed rate but, at least at my p.o.v, the feedrate is the speed for all together, therefore i wasn't accomplishing what i want.
No, the use of it isn't for a 3D printer so theres no need of them to be coordinated.
So now i'm wondering if it is even possible.

Blue-Marlin, 2016-04-18:
Simple answer: No!
In Marlin the axes __are__ coordinated.
If you want to have determined speeds for every axis __you__ have to calculate a mating target point

CoderSquirrel, 2016-04-18:
Thanks @Blue-Marlin

closed 2016-04-18 by CoderSquirrel