MarlinFirmware/Marlin #3562: changing feedrate by the command of "M203"

WillSmithInChina, 2016-04-19:
when i changed feedrate with the command of "M203 X200 Y200" (default feedrate: x300 y300) ,then tested, i found it didn't work, so do you meet the same problem and how do you solve it ?

manianac, 2016-04-19:
Curious, how are you sure it did not work?  I've changed acceleration and max feedrate before without issue.

WillSmithInChina, 2016-04-19:
@manianac  test as following:
I>
G1 X0 Y0
M203 X300 Y300 
G1 X800 Y800
calculating the time of 0 to 800

II>
G1 X0 Y0
M203 X200 Y200
G1 X800 Y800
calculating the time of 0 to 800

Result:
the time of "M203 X300 Y300" is about 12.12s
the time of "M203 X200 Y200" is about 11.99s

so i think it doestn't work no matter how i change the feedrate of xy

thinkyhead, 2016-04-19:
@WillSmithInChina Try these tests to get a bit more contrast, and to actually set a max feedrate for the moves.
```
G1 X0 Y0
M203 X500 Y500 ; typical high feed rate
G1 X800 Y800 F5000
```
…then…
```
G1 X0 Y0
M203 X60 Y60 ; makibox slow feed rate mm/s
G1 X800 Y800 F5000
```

WillSmithInChina, 2016-04-20:
@thinkyhead changing speed by using "M220" and "M221" ,thanks again !

manianac, 2016-04-20:
@WillSmithInChina Whats your acceleration settings?

I find M203 X500 Y500 to fly as it should, and X60 Y60 all work properly.

Realize if you're testing with only a G1 command, you need to set the feedrate with the F var in mm/min as it will use the last set speed (usually a slow move).

So when I do
`G28 X0 Y0`
`M203 X500 Y500`
`G1 X800 Y800 F30000`

It sails amazingly fast.

thinkyhead, 2016-04-21:
@manianac [Three backticks](https://guides.github.com/features/mastering-markdown/) is your friend for multi-line code.

closed 2016-05-04 by jbrazio