MarlinFirmware/Marlin #3570: E step handling, calibration and extruding with USB

Alex9779, 2016-04-19:
I am trying to calibrate my E steps for a day now and I have strange strange issues.
I used the bleeding edge of last nights RCBgFix, I used RC5 and I used RC4 all with the same results.

What I do is I mark the filament measured from a fixed point, in my case the top of the extruder, then I extrude and measure where the mark moved.
My calculation for the new E step is [current e steps] * [target extrusion distance] / [real extrusion distance].

So my first issue or question is how the firmware does handle a change of E steps?
I doe what described above, then I move the E axis then I calculate and change the E steps and move the axis again.
Apparently there is something going on because a test showed me this.
I changed the E steps to a too high value and moved the axis to 10mm, the real length extruded was 12mm. Then I change the steps and move the axis again but the extruder retracted and my mark was at the now set 11mm.
Is that an intended behaviour?

My second issue is that when doing this process with USB over OctoPrint I am somehow not able to get any reliable results.
I used the built in extrusion method of OctoPrint, it uses two commands to extrude, first "G91" for relative followed by the extrusion command "G1 Ex F300" in my case. And also I tried with scripts like this one:
```
G21 ; set to millimeters
G90 ; use absolute positioning
T0 ; select extruder 0
G92 E0 ; zero the extrusion distance
G1 E100 F400 ; extrude 100mm
G92 E0 ; zero the extrusion distance
```
or this one:
```
G21 ; set to millimeters
G90 ; use absolute positioning
M83 ; relative extrusion
T0 ; select extruder 0
G1 E100 F400 ; extrude 100mm
```
Both had the same result. With the shown 100mm extrusion I get to a maximum of about 92mm extruded no matter how high I set my E steps. I didn't do it very high only for about three iterations, for example I started with
417.5
417,5*100/91=458.8
458.8*100/99=463.4
463.4*100/91=....
I increased but the extruded distance decreased...

If I try extrusion from OctorPrint and set the E steps to 417.5 which seems to be right in my case at least with LCD and changing nothing I get right that what I want, and I extrude 10mm with OctoPrint I get 10mm, if I extrude 50mm I get less, about only 41mm, and if I extrude 100mm I get much less...

Am I doing something wrong? Are there calculations made I do not know of?
How am I supposed to do this process?

jbrazio, 2016-04-19:
Could it be slippage on the drive gear for your feeder ?

Alex9779, 2016-04-19:
No I checked that twice, triple, infinite times. I am sure the gear has a good grip on the filament. I am guiding the filament when extruding to be sure it is centered on the robbed gear and the pushing bearing.
Believe me, I am working on this for two days now. I wrote this here because I am out of options and ideas...

Alex9779, 2016-04-19:
BTW it is the new Titan by E3D, a geared extruder, the motor has a D shaft, the pinion can't slip. The main gear has full engagement to the pinion. The robbed gear feeding the filament sits tight in the main gear. The tension to the filament is good. Replay no slippage...

jbrazio, 2016-04-19:
Just for kicks, have you tried lower values of `F` in `G1 E100 F400` ?

Alex9779, 2016-04-19:
Oh not yet. Good hint. Will try that and report...

Alex9779, 2016-04-19:
Ok did G1 E50 F50 and then the extrusion is the same and with the 417.5 E steps also 50mm.
So does the motor skip steps? But the sound of it is ok, power too low?

Alex9779, 2016-04-19:
So because of my previous comment I tried the same with the second extruder. Didn't do that yet because it was not really setup properly.
I just inserted filament, activate cold extrusion and tried to extruder 50mm at various speeds.
The distance was a bit off for that extruder an filament, I got 54mm but at ANY speed...
I tried F50 -> 54mm, F300 -> 54mm, F400 -> 54mm...

Any advise? Motor, stepper driver, current setting?

Alex9779, 2016-04-19:
Ok I think the motor is not ok. I tried on the second extruder, same motor, now the same filament and again the extrusion at the speeds I tested is ok...
Going to switch the just the motors now and recheck...

Alex9779, 2016-04-19:
Well I think I found it but it is not that obvious. It it somehow filament related and speed. It is not the motor or the firmware...
Thanks for posting me in a direction and sorry for bothering...

closed 2016-04-19 by Alex9779

jbrazio, 2016-04-19:
You're not bothering, always welcome.