MarlinFirmware/Marlin #3573: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN doesnt compile alone

Sungod3000, 2016-04-20:
Hi,

im just reinstalling my bed level probe and I enabled #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN the comments say that only one of these #define probe options should be enabled, but just having z min endstop doesnt compile and gives the "A probe needs a pin!" error. Im using ramps which should be the default or do I have to redefine the z min endstop pin?
or is there another prerequisite that I overlooked to make this work?

Blue-Marlin, 2016-04-20:
Please check if you have
```
#define USE_ZMIN_PLUG
```

Sungod3000, 2016-04-20:
You mean #define Z_MIN_PROBE_PLUG instead of #define Z_MIN_PROBE_ENDSTOP

The example below also compiles with the same error
```

// WARNING:
// Setting the wrong pin may have unexpected and potentially disastrous consequences.
// Use with caution and do your homework.
//
#define Z_MIN_PROBE_PLUG

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// To use a probe you must enable one of the two options above!
```

Blue-Marlin, 2016-04-20:
No. I thaught about this area
```
// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
```

Sungod3000, 2016-04-20:
Im still not finding the define_plug BUT i found that #define DISABLE_MIN_ENDSTOPS was enabled. after commenting it, it compiles now

Blue-Marlin, 2016-04-20:
That has gone now
```
#elif defined(DISABLE_MAX_ENDSTOPS) || defined(DISABLE_MIN_ENDSTOPS)
  #error DISABLE_MAX_ENDSTOPS and DISABLE_MIN_ENDSTOPS deprecated. Use individual USE_*_PLUG options instead.
```

A good opportunity to remember everyone that we urgently need your version/branch with __every__ issue.

closed 2016-05-07 by jbrazio