MarlinFirmware/Marlin #3574: starting print doesn't work

WillSmithInChina, 2016-04-20:
Before starting print, when i set extrude temperature which is the same as 
the  command (M109 S230.000000) in gcode file, starting print doesn't work ;

But when i set extrude temperature not the same as "M109 S230.000000" before printing,
starting print works fine.

Do you  meet the same problem ? if you konw that, please tell me,
thank you, guys !

WillSmithInChina, 2016-04-21:
when the current extrude temperature is the same as the command(M109 S230.000000) in gcode file,
starting print will not work

thinkyhead, 2016-04-21:
I thought we fixed this in the latest code. We'll have to have another look…

WillSmithInChina, 2016-04-21:
@thinkyhead thank you for all you did

jbrazio, 2016-04-21:
@WillSmithInChina do you still have the issue ? If so could you please provide a few lines of serial output. Thanks.

thinkyhead, 2016-04-21:
I looked over the code in `M109` (`RCBugFix` of course) last night. I couldn't find any condition where it would get stuck. So maybe @WillSmithInChina is using an older version of Marlin…? He seems to be seeing various bugs that I'm sure we fixed after RC5.

jbrazio, 2016-04-21:
I have the feeling we may have something going on with the bed PID or the recent `HISTERISYS` stuff as we have at least a couple is reports about printing never starting. One of them (I am in mobile can't link to issue) the serial log shows bed temp is always 2C behind target.

thinkyhead, 2016-04-21:
So you think it might be in the conversion between celsius and "thermistor stuff"? But I believe the heater is turned on and off based on the celsius value, so it should stay on until it reaches the target.

jbrazio, 2016-04-22:
That's exactly the strange part.. Data is not sufficient but the PID seems to settle the temp a couple of degrees of the target with a stable PWM value.

thinkyhead, 2016-04-22:
That just sounds like poor PID tuning. There should be a little overshoot in a proper PID. Apparently we have a new `M303` in the queue to allow just that.

closed 2016-05-04 by jbrazio