MarlinFirmware/Marlin #3575: Bed Levelling not relevelling properly after retry

Grogyan, 2016-04-20:
Prusa I2
RAMPS 1.4
Full Graphic LCD
Max31855 thermocouple
OctoPrint 1.2.10, RPI2
Tried even with RCBugFix 20-4-2016

Start Gcode
`G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M106       ;start with the fan on
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G29        ;auto level bed
G1 Z15.0 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E15              ;extrude 15mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...`

When starting out for the first time of a full reboot of the printer and OctoPrint, I manually do,
G28 X0 Y0
G28 Z0

On starting a print then cancelling it during a G29
Then restarting the print again, After a G28, it often does not reach the first touch point during a G29 as it comes down at the slow feedrate of 2mm/s, instead of the higher 5mm/s, for the first touch of the first touch point, touch points 2, 3 and 4 home properly and probe at the slow 2mm/s

Same thing occurs when executing G29 manually after G28.

If doing a G29 during a print, the nozzle with crash into the bed as that what it sees as it adjusted planar.

Grogyan, 2016-04-21:
After another round of testing this, I have come to the conclusion that after starting print;
G28 X0 Y0
G28 Z0
G29
Cancel the print during G29
Rest of the gcodes in the buffer for printing execute
Manually Raise Z axis 10mm
Start print again, which then does,
G28 X0 Y0
G28 Z0
G29
Print starts again 10mm above the bed.
This means that the buffer for G29 is getting corrupted where it uses a test point of the manual bed raise from a previous print job.
I can now reproduce this every time.

thinkyhead, 2016-04-21:
> Cancel the print during G29
Rest of the gcodes in the buffer for printing execute

It would be good to purge the command buffer when an SD print is canceled.

> I can now reproduce this every time.

Excellent! So, if you enable `DEBUG_LEVELING_FEATURE` and then do `M111 S32` to display debugging, can you see a clear difference in the output? Where in the process is it losing track of the position?

Grogyan, 2016-04-22:
This is the output of the terminal with Bed level debug.
<script src="https://gist.github.com/Grogyan/0917920773d6c90ee0b32d2e78b9936a.js"></script>
In the end of the test I just a simple G28 Z0, the probe did not touch the bed.

thinkyhead, 2016-04-23:
> This is the output of the terminal with Bed level debug.

Did you glean anything yourself from the output? What lines should we pay the most attention to? I will check it out shortly

thinkyhead, 2016-04-23:
> In the end of the test I just a simple G28 Z0, the probe did not touch the bed.

When you do a `G28 Z` the probe should always reach the bed. Or at least it should do the usual thing, which is to probe the bed once quickly, then probe the bed again slowly. If it fails to do this, then something weird is going on.

I've been combing through the code and your log output, and it looks like the issue might be occurring in the `G28` routine, not the `G29` routine. I will add more logging and make some adjustments, then post some new code for you to try.

Grogyan, 2016-04-23:
Yes, if probing is at slow speed, it will timeout.
The only workaround I know of is to reduce 
Z_RAISE_BEFORE_PROBING
Z_RAISE_AFTER_PROBING 
to 10mm
That however doesn't prevent the G29 from ignoring the G1 Z10 (The Z height could be anything) when it begins probing. Not to be confused with Z_RAISE_BEFORE_PROBING  and Z_RAISE_AFTER_PROBING 
And on many occasions fast initial Z zero probe is at it's slow speed setting, it isn't as easy to replicate though

thinkyhead, 2016-04-23:
When you say it "times out" is it invoking `kill` and halting the machine?

thinkyhead, 2016-04-23:
I've added more logging in this branch:
https://github.com/thinkyhead/Marlin/tree/rc_z_safe_homing_deps

See if the added output gives any clues!

Grogyan, 2016-04-27:
Sorry, I haven't had time to test his with RC6

thinkyhead, 2016-04-29:
@Grogyan Be sure to test with the branch I linked, for the extra logging.

Grogyan, 2016-04-30:
I have tried with the extra logging
Z_OFFSET is  -1.15mm

Just G28
X and Y homed, then moved to (50, 50) Z probed ok then raised
Recv: X:50.00 Y:90.00 Z:16.15 E:0.00 Count X: 8000 Y:14400 Z:51680
Recv: ok

Doesn't really say much

Ran G28 Z0 a couple more times, but with doing a G28 X0 Y0 first.
Recv: X:0.00 Y:0.00 Z:16.15 E:0.00 Count X: 0 Y:0 Z:51680
Recv: X:0.00 Y:0.00 Z:16.15 E:0.00 Count X: 0 Y:0 Z:51680
Z never actually touched probe

I do feel now though it does seem to be a G28 problem, seems to be timing out when there is a slow feedrate, this is because with a toggled G28 Z0 from the host interface instead of typed into the terminal, G28 Z0 probes consistently well, up until it decides to use slow feedrate mode.

http://pastebin.com/TspwfrYc

thinkyhead, 2016-05-01:
`M111 S1` allows you to see all commands sent from the host. Perhaps the host is itself sending a command to use a slow feedrate?

Grogyan, 2016-05-06:
Adding my configuration.h file for reference
[Configuration.h](https://gist.github.com/Grogyan/595daca66f37101fb2fed315ac985754)

gddeen, 2016-05-10:
You have a Z-Probe on a servo?  You have defined 3 servos.

Are there 2 other servos? (I'm going through all of this... I finally have all of my issues worked out...
with a dual extruder, mesh bed leveling, auto bed leveling, and a Z-Probe with Min/Max on all stops.

thinkyhead, 2016-05-11:
Maybe this is a bug in need of fixing. The current (default) configuration defines…
```cpp
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command
...
//#define X_ENDSTOP_SERVO_NR 1
//#define Y_ENDSTOP_SERVO_NR 2
//#define Z_ENDSTOP_SERVO_NR 0
//#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {70,0}} // X,Y,Z Axis Extend and Retract angles
```
It looks like it really should be just set up for Z…
```cpp
//#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
...
//#define Z_ENDSTOP_SERVO_NR 0
//#define X_ENDSTOP_SERVO_NR 1
//#define Y_ENDSTOP_SERVO_NR 2
//#define SERVO_ENDSTOP_ANGLES {{70,0}, {0,0}, {0,0}} // Z,X,Y Axis Extend and Retract angles
```

gddeen, 2016-05-11:
They go through the array with the common enumerations, x, y, z... So if you do this, marlin main
needs to also change... (duh). What you have done of course would make much more sense to a person modifying
configuration.h.

Blue-Marlin, 2016-05-11:
No, no, no.
SERVO_ENDSTOP_ANGLES [ x, y, z]

But x an y servo_endsops are not tested anymore in `Endstops::update()`, so they can go away here as well as in ` homeaxis()`.

Grogyan, 2016-05-11:
As I use the BLTouch probe, which uses the Z-probe servo, using the same probe commands as a servo.
```#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {10,90}} // X,Y,Z Axis Extend and Retract angles```
I added even more debugging code to Marlin_main.cpp to determine where the slow Z feedrate occurs, but was unable to test it last night, sleep got in the way.

thinkyhead, 2016-05-11:
Oh right, sorry. This even confuses me sometimes…

`#define Z_ENDSTOP_SERVO_NR 0` actually means Z is `SERVO_0_PIN`, of course…
```cpp

  const int servo_endstop_id[] = SERVO_ENDSTOP_IDS;

  . . .

  #if NUM_SERVOS >= 1 && HAS_SERVO_0
    servo[0].attach(SERVO0_PIN);
    servo[0].detach(); // Just set up the pin. We don't have a position yet. Don't move to a random position.
  #endif

  . . .

  . . . servo[servo_endstop_id[i]] . . .
```

> But x and y servo_endsops are not tested anymore in `Endstops::update()`

They are still moved, if defined, in the `homeaxis` function:

```cpp
    #if HAS_SERVO_ENDSTOPS
      // Engage an X or Y Servo endstop if enabled
      if (_Z_SERVO_TEST && servo_endstop_id[axis] >= 0) {
        servo[servo_endstop_id[axis]].move(servo_endstop_angle[axis][0]);
        if (_Z_PROBE_SUBTEST) endstops.z_probe_enabled = true;
      }
    #endif
```
…not that this isn't in many ways obsolete, since it doesn't have any added code to make room for the deployment or stowing.

Grogyan, 2016-05-15:
After much retesting over the weekend, I was unable to reproduce the bug, firmware hadn't been changed when it did  show, and even with a new Arduino Mega, last week the bug appeared.
So right now I don't know what is happening.

Currently there are a lot of developments with regards to MBL.

jbrazio, 2016-05-24:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-24 by jbrazio

gddeen, 2016-05-24:
@Grogyan FYI, I noticed similar weird behavior when I failed to load eeprom values, then write them back
after a G29 S1. Things just acted crazy. So I did this and everything started working.
M502 ; load factory
G28 ; go play with the probe for fun.
G29 S1 ;do it again
G1 Z0 ; twiddle until nozzle at 0
G29 S2 ; next point
G1 Z0 ; twiddle until nozzle at 0
G29 S2 ; next point
G1 Z0
G29 S2
G1 Z0
M500 ; save as EEPROM values

jbrazio, 2016-05-24:
@gddeen you brought up a good point.. we never delete the EEPROM values unless we have a new layout version, this means we can have a special condition where the EEPROM may be messed up and the robot starts acting weird unless we replace the "random" data with true data.

@Blue-Marlin @thinkyhead what do you think, this can actually explain some weird/random behaviors.

Grogyan, 2016-05-24:
@gddeen I've done a M502 followed by a M500, the problem still randomly crops up.

@jbrazio I've not commented on this thread  for a while due to knuckling down the #3548 issue, which was of much more serious concern to me, whereas  this issue is just annoying.
There is a few other PRs in the RCBugFix branch that I have not had time to retest, for the same reason as above.

thinkyhead, 2016-05-25:
> explain some weird/random behaviors

For disabled features we try to write sane default values to EEPROM. As far as I know, none of the default values should cause any problems when a feature is enabled later. But it should be noted that they may not match the defaults set in the configuration. So it's always a good idea when re-flashing to do `M502` to reset your settings to the defaults, redo and resave your mesh bed leveling, etc.

gddeen, 2016-05-25:
I doubled checked store. There is no problem in getting off sync. Values are set (mostly to 0). MBL does same at reset. So... My weird acting printer was something else. (this was RC6 4-21). I'll get more detail if it happens again...