MarlinFirmware/Marlin #3580: bed level smashes into bed from g28 to g29.. does not stop

locutus40, 2016-04-21:
just uploaded it to my arduino to test.. the new release failed..

first i use a prusia i3 with mk 8 extruder with a z probe on a aluminum bed

when you g28 its fine.. then g29 right from g28 it smashes into the bed see its already homed so 0.001 mm from the bed then it smashes right down where before it would raise and do its thing

ive adjusted the start g code to lift the nozzle after homing and it still does the same thing.. 
 so i had to emergency stop it.. i went back to rc3 and it works just fine someone take a look at it

Blue-Marlin, 2016-04-21:
Some important information s missed.
What version/branch did you test?
Where is your probe connected? (pin)
Have you done a M119 test in mid air? Can you trigger the probe?
Your failing Configuration.h may help us to find the cause if appended.

jbrazio, 2016-04-21:
@locutus40 Test all your endstops with `M119`.

locutus40, 2016-04-21:
the end stop i use is an inductive probe.. it homes just fine red light comes on and stop where it should.. while homed aka almost touching the surface i do a g29 it should just go to the back and lift and then probe and so on but it dosent .. from the home position with the red light triggered it just smashes rightinto the bed....

even when i changed my start code to say after g28 lift nozzel to 10 mm it just smashes the nozzle

go to print right from a g28  homeing and it does the same thing.. its really weird.. all the sensors are working..

and rc3 i used before rc5 and rc3 worked just fine.. so im at a loss for words..

thinkyhead, 2016-04-21:
@locutus40 Have you tried [the current `RCBugFix` code](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix)? We might have already fixed this issue.

Roxy-3DPrintBoard, 2016-04-22:
> it should just go to the back and lift and then probe and so on but it (does'nt) .. from the home position with the red light triggered it just smashes right into the bed....

My guess is we can change a few Configuration.h settings and it won't do those bad things.   It might be helpful to post your current Configuration.h settings.

locutus40, 2016-04-22:
like rc3 works just fine.. its only in rc5 that.. it goes right past the sensor triggering and keeps going into the bed like it does not detect it at all..

so youve just g28 hot end touching bed slightly then g29 poof smashes into bed.. before it would raise up or slide across the bed from the same position then probe but now...

so im going back to the old firmware gunna try rc4

jbrazio, 2016-04-22:
Config files have changed a lot for RC5

locutus40, 2016-04-22:
i tried rc4 and it works just fine now.. so it has to be somthing up with rc5 could damage alot of hot ends im lucky i had my cursor over the stop button

jbrazio, 2016-04-22:
Could you post your config file ?

jbrazio, 2016-04-22:
You should try `RCBugFix`.

locutus40, 2016-04-23:
what the my thread was closed.. thats not cool its a ongoing thing until fixed.. i dont appreciate who ever closed it..

thinkyhead, 2016-04-23:
@locutus40 Was it closed without comment? Whoever closed it should have been clearer about why they were closing it.

thinkyhead, 2016-04-23:
I know that some maintainers here will close an issue if it's too hard to understand. But they usually ask the author to post a new issue with more clear language before they close it.

thinkyhead, 2016-04-23:
Sorry @locutus40 I mis-read your post and thought you had resolved the issue.

locutus40, 2016-04-23:
the issue is with marlin fw revision rc5 so until you guys fix it it remains open i just wanted people to see that theres a problem wiht it.. closing the issue ...ok well then move the thread this whole thread to rcbugfix.. i cant do that i dont have the power

thinkyhead, 2016-04-23:
@locutus40 You are confused. **_This is the `RCBugFix` thread._**

Here's the code you need to test: https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip

locutus40, 2016-04-23:
ok i will try it out.. and get back to you...  rc6 ok well i guess ill try both out ..

thinkyhead, 2016-04-23:
> ok i will try it out

Great, thanks!

> i guess ill try both out

Please understand, when I say "we don't support `RC5`" what I mean is that `RC5` isn't going to get patched and then re-posted as `RC5`. Any bugs that exist in `RC5` will _always_ exist in `RC5` forevermore.

The code in `RC5` is over 20 days stale. So `RCBugFix` is what you want to test. It consists of the latest release candidate (`RC5`) plus any patches that have been done since `RC5` was tagged. And we've done a lot of patches!

locutus40, 2016-04-23:
ok i confirmed the rcfix works just fine.. g28 does its thing so its touching the bed almost aka papers thin. then straight to a g29 and it lifts and does its thing.. with no crashing into the bed good work guys... thanks...  i love how you brought back where it homes too. it now homes to the far right and back.. 

so if your looking at the machine all the way right and bed slid towards you... perfect dont change a thing on that... where before it was bed all the way back and nozzel all the way right.....

so the fix you implimentd for the crashing head works just fine.. i look forward to rc6 this weekend...

thinkyhead, 2016-04-23:
@locutus40 Good news! Thank you for your report. Would you mind posting your `Configuration.h` file? I just want to make sure that Marlin actually _should_ be homing to the back-right position on your machine based on your settings. (Most machines home to the front-left position, X=0 Y=0.)

Also, for total completeness, would you mind sending us some extra "diagnostic" information? Simply enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware. Then, before you do any homing or leveling, send the command `M111 S35` to enable detailed output for `G28` / `G29`. Perform one or two `G28` / `G29` procedures and send us the output.

Thank you!

Grogyan, 2016-04-30:
Same as #3575

jbrazio, 2016-05-11:
Closing this issue as it is identified as a duplicate of #3575, let's proceed the discussion there.

closed 2016-05-11 by jbrazio