MarlinFirmware/Marlin #3581: Auto Bed Leveling not accurate for Z0 hight

garaguido, 2016-04-21:
I have problems with the Auto Bed Leveling  .

ABL is always working fine by the first print , when I start the printer the first time ( perhaps next day) the first Z layer is always perfect with Z-0.05mm compensation.

If I start a second or more prints , after I give a G28 X0 Y0 , G28 Z0 and G29 (ABL) , there is always a Z layer difference of +0.9 mm , I must then slice again for a good layer hight , with a Z -0.9mm compensation difference .

And NO ,This is not a expansion hot-end problem , this was not the behavier before  without ABL and also not the behavier with the MBL .

My motherboard is a Xinray and compatible with Ramps 1.4
Is this  a bug ?

jbrazio, 2016-04-21:
Did you test your probe for repeatability ?
You're using ABL with how many points ?
Do you have you hotend offset correctly set ? (Is something on your G-Code messing with `M851` ?

apballard, 2016-04-21:
I can reproduce the problem as well.

At first I thought the description given by garaguido was quite strange, with a layer difference of 0.9mm, but the same thing is happening to me.

Been struggling repeatedly to get z height correct.  I would switch on the printer, everything would go fine first test/print then everything goes badly after that (nozzle too close to build plate).  Re-calibrated the z axis, used ABL and so on.  Nothing worked.  Frustrated I switched off the printer, went away came back later to find the first print/test was perfect.  After investigation I found that every subsequent print after the first print, the z axis was out by 0.9mm

Im using a rumba board, with RC5

jbrazio, 2016-04-21:
And the 0.9mm is incremental between prints until you reboot it ?

thinkyhead, 2016-04-21:
Try enabling `DEBUG_LEVELING_FEATURE` and then you can use `M111 S32` to turn on logging for leveling and homing. Do a few tests and post the output here. It should be clear by comparing the tests where the issue is happening, and why it's always 0.9mm.

apballard, 2016-04-22:
Thanks, I'll enable the feature and debug some more.

jbrazio, it only increments once by .9 mm.  But I'm going to run more tests and see.  For example I want to see if there is a difference between printing from SD card or from my computer.  Also just want to double check that I can reproduce the problem again today.  May rollback to rc4 as well to test.

garaguido, 2016-04-22:
jbrazio , I test my probe for repeatability , and still bang on the same number.
I used before 4 points on the bed and after somme reviews on RepRap forum i\I used  9 points but the was differevce was always there .
My hotend offset was correct.

garaguido, 2016-04-22:
apballard , I have hours spending  on finding a solution ( several test prints ) and had to stop this  , I was frustrated. 
Therefor I use MBL now , this has one mayor point , you must not always doing a ABL , the level matrix from MBL is stored in the EEprom and always active.
I found this Github forum and they already solve other problems  and  I think by myself give this a try perhaps they find a solution for it also

jbrazio, 2016-04-22:
ABLx4 should be enough.

So the error only happens with ABL, MBL everything is fine ?

garaguido, 2016-04-22:
Yes , its always via ABL , I use now MBL and there was no problem , even with several prints after one another.

Has this sommeting to do with clearing the Eeprom buffer or a G28 after a sertain command ?

Because its not happening , by the first print , when the power of the printer was out for a will?

jbrazio, 2016-04-22:
> Try enabling DEBUG_LEVELING_FEATURE and then you can use M111 S32

Could you do this ? Compile Marlin with `#define DEBUG_LEVELING_FEATURE ` and them on top of your G-Code put `M111 S32`, then copy all the serial output to here, if possible you should provide one serial ouput *without the error* and another one *with the error*. Thanks.

garaguido, 2016-04-22:
Yes , I will try this week-end . 
I must reconfigurate ABL in Marlin first , now my printer is setup for MBL

apballard, 2016-04-22:
ok, this evening been testing and testing, cannot reproduce the problem. Although I did enable debugging and changed the number of points to scan so it's 9. Before it was just scanning the corners. I found my bed was "bulging" very slightly in the middle. But now that the ABL is testing the center of the bed, it's been perfect, and my first layer is good quality and repeatable.

I must admit though the slight bulge is not .9mm so not sure where that was coming from.

thinkyhead, 2016-04-23:
@garaguido Be sure to download `RCBugFix` code again, as there have been some fixes for MBL. And I'm looking at adding more logging for ABL also, to try and catch an issue that might exist in `G28` with `Z_SAFE_HOMING` (or maybe not).

garaguido, 2016-04-23:
I will try reproducing  the fault (Z0 problem) , if its possible with the latest RCbugfix ,  tomorow  I will send you the logfiles for it !!

hBertin, 2016-04-23:
Hi, what kind of probe are you using for bed leveling ?
In my case I use capacitive and I had to deactivate the bed heating during probing otherwise it would throw off the sensor.
Could this be your issue ?

garaguido, 2016-04-23:
Hi , My sensor is a capacitive one and when I probe my bed the bed temp. setting is not active . 
I set ABL (G29) in the G-code before bed temp. and extr. temp.

When this is hapening by you , maybe tension drop ?

hBertin, 2016-04-23:
OK, in your case the issue shouldn't come from here.

On my side I checked with a scope, there is no voltage drop but the power supply is noisier when the bed is heating up. I will put some caps on the sensor's supply rail to smooth everything out. I will also try to ground the aluminum plate under my glass panel (used as a print surface).

What print surface do you use, do you change it when doing a new print ? I sometimes noticed that the glass plate or the part can have some static charge after removing the printed part.

garaguido, 2016-04-24:
hBertin , I use a heated bed with a mirrorglass plate 200 x 200 x 3mm (from IKEA). 
I also run some test today with Debug active and the newest `RCbugfix software` and I can not reproduce the fault again , I must do some more test after printer is OFF for a will and totaly cooldown , thats where I had the problems with  Z diff , its  always happens on my 1e print .

garaguido, 2016-04-24:
thinkyhead , I also run some test today with Debug active and the newest RCbugfix software and I can not reproduce the fault again . 
Hereby I have for you the logfiles from 6 prints I made.

The first and the second print was without bed correction , I had my bed leveled man corrected for the use of MBL and  therefor was my correction to high ( Z+0.4) these print where bad .
The 3e to 5e was with the extra Z-0.4 correction ( offset was then Z-5.5 ) and these ware bang on verry good.
The 6e was with changing the G-code ( in Slic3r) set the ABL first in the script and after the printer was OFF for a will , these print was also good.

I must do some more test after printer is OFF for a will and totaly cooldown .
But you can already look at these logfiles , succes with it .

Keep you posted.

[Logfile ABL.zip](https://github.com/MarlinFirmware/Marlin/files/233373/Logfile.ABL.zip)

garaguido, 2016-04-24:
thinkyhead , jbrazio ,
Hereby the new logfiles , after I power the printer off and let it cooldown.

I take 3 runs with the same Z-5.50 offset as before , and al these 3 prints (first layer) are verry good and consistend.

Perhaps this  problem is now solved maybe with the new RCbugfix versions ?

Look at these if you wish ?

[Logfile ABL 2.zip](https://github.com/MarlinFirmware/Marlin/files/233412/Logfile.ABL.2.zip)

Thanks for the effort .

jbrazio, 2016-04-24:
Thank you for all the debug, it's great news this problem is fixed. If the issue reappears just (re)open this ticket.

closed 2016-04-24 by jbrazio