MarlinFirmware/Marlin #3582: Z-Probe variable name

sarf2k4, 2016-04-21:
I would like to propose a name change to these `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and `//#define Z_MIN_PROBE_ENDSTOP`, the name itself is quite confusing.

I'm not sure what other names but I also  pointed this out at here https://github.com/MarlinFirmware/MarlinDocumentation/issues/20

thinkyhead, 2016-04-21:
It's true that these options are a bit too "low level" and confusing for the average user. Some of these options might be hidden behind a simpler config system, similar to the way you now choose an LCD controller, and depending on the one you choose various other options are set.

So let's mull over some possible better names.

We might also be able to do something like this perhaps:
```cpp
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN (Z_MIN_PROBE_PIN == Z_MIN_PIN)
```

sarf2k4, 2016-04-22:
I was thinking to go with `z probe as z min` or ` use z-probe only` for `z min probe uses z min endstop pin` while for `z min probe endstop` would be something like this `z probe with z min`

Not to mention since both of these thing are located way above the auto bed leveling section; homing section that is. I also think that these option would be put under auto bed leveling section since the default hardware setup would be all min endstops anyway and be default not to use z probe.

thinkyhead, 2016-04-23:
@sarf2k4 I'm not sure what the reasoning is to use `D32` for the Z probe pin on RAMPS, so I don't want to jump in and alter it just yet. But maybe I will post a PR suggesting it and see if anyone has objections.

It has been suggested by others recently that we ought to move all the PROBE options out of the `AUTO_BED_LEVELING_FEATURE` block, and I think that's a good idea. But there are natural worries that if we introduce these options independent of ABL that it will break something. And we want to get 1.1.0 out as soon as we can, so we have to apply triage to things like this and wait for 1.1.1. I have been holding out for perfection, but I think it's a good idea to get 1.1.0 out, reorganize the code, then focus on 1.2 where we can rework core coe with a little more freedom.

Blue-Marlin, 2016-04-23:
Somehow the knowledge got lost, to see the pins-files as user editable configuration files. Since splitting up the huge pin-file, you don't need witchcraft to edit them to your needs.
Despite having said the above, i guess D1 (or D57) on AUX-1 could be the better choice for a default. There you could use ready made endstop plugs with the same order of pins (signal, GND, 5V)

jbrazio, 2016-05-07:
To be followed up within PR #3672.

closed 2016-05-07 by jbrazio