MarlinFirmware/Marlin #3583: G29 probing

renzaaa, 2016-04-21:
After G28 comand, not properly probing. G28 - home to bed center and after G28 Y axis not move to front bed position. Just moving to bed right, bed left and from bed left to bed back and touch to printer body. Why missed front bed position?

thinkyhead, 2016-04-21:
- What version of Marlin are you running?
- Can you post your configuration files so we can look them over?

renzaaa, 2016-04-21:
Marlin version is 1.1.0 RC5.
--------------------------------------------------------------------
#include <U8glib.h>

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

[... 856 lines cut ...]
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

renzaaa, 2016-04-21:
With Marlin RC3 version everything is fine.

thinkyhead, 2016-04-22:
@renzaaa You should test `RCBugFix` because we might have already fixed this!

renzaaa, 2016-04-22:
I will try.
Thanks.

StrannikZet, 2016-04-25:
I tested the RC Bug Fix, but after the G28 and measuring the center of the table with the G29, the printer considers it Y0. After the ninth point, if you manually send home at X0 Y0, Y - and resets the value becomes -11.5 or more. Example ninth point X = 280 Y = -11.3, but must be X = 280 Y = 280.

thinkyhead, 2016-04-25:
I will look at your configs and see how they must be affecting the homing. I might list out what I _think_ should happen based on your settings, and we can compare notes.

StrannikZet, 2016-04-26:
I miss RCBugFix again and try. If the problem persists, show videos and action logs. If everything is OK, I'll write. Prior to that tested in the firmware download, this Saturday. Maybe there were some changes on the test bench all ok.

StrannikZet, 2016-04-26:
Testbed. As long as the printer will be able to check out the evening. I am confused here these values.
Why is this value:
10:30:57.098 : >>> set_bed_level_equation_lsq > current_position: (275.00, 298.00, 87.95)
Changes to this
10:30:57.117 : <<< set_bed_level_equation_lsq > corrected_position: (276.47, 307.45, 34.68)
[log_debug.txt](https://github.com/MarlinFirmware/Marlin/files/236101/log_debug.txt)
I could be wrong, but as a check on your printer, will show results. It is simply tried to emulate the situation.

StrannikZet, 2016-04-26:
Hmm ... I got a bug fixed renzaaa. Firmware downloaded that morning, kind of like working properly. Sorry and thank you!

closed 2016-04-28 by jbrazio