MarlinFirmware/Marlin #3590: Power input requirement

sarf2k4, 2016-04-22:
This isn't much of an issue though, it's only based on what I experienced.

I have the following setup for my 3d printer
- ramps 1.4
- towerpro sg9s servo for auto bed leveling
- 12v system
- e3d v6 + fan hooked on ramps 1.4
- pwm fan setup
- RRD 2004 lcd
- heated bed

When I was using rc2-rc3, my major issues was to have the printer start to print right after g29. So I did power my mega with additional usb power input from usb charger or my laptop just to get the printer started printing after g29 command.

After upgraded to rc4/rc5 (I think currently rc5 for sure), I noticed that I didn't need the extra usb power input anymore just to get the printer start printing right after g29. My assumption is that some of the codes inside might be messed up that were cleaned somewhere rc4-rc5 to optimize the power requirement for the board.

I thought at first it is a common thing to have such setup that I would need extra power but I was wrong. I wonder if anyone able to confirm this?

jbrazio, 2016-04-22:
Could you please show us a serial log when it refuses to print after a `G29` ?

sarf2k4, 2016-04-22:
There are no logs (as far as I know of), because the printer doesn't want to move. I often do sd card printing. If I use pronterface it means I'm giving extra usb power input to the main board and the printer will move normally the way it should be

jbrazio, 2016-04-22:
You should connect a serial terminal like the Arduino IDE serial monitor and start the SD printing.

Blue-Marlin, 2016-04-22:
Could be caused by a defect D1 diode on the RAMPS.

sarf2k4, 2016-04-22:
I don't know though, but it's okay already since using rc5

closed 2016-05-04 by jbrazio