MarlinFirmware/Marlin #3593: Problem using some letters in folder names when browsing sd card.

lintz, 2016-04-22:
Using Scandinavian letters such as Ä ä makes it impossible to enter a folder. Still showed correctly on display.

Tested on Marlin 1.1.0-RC5 - 01 April 2016

thinkyhead, 2016-04-23:
Thank you for reporting this issue. We haven't heard about this before. Can you issue an `M20` command from Pronterface and post the output of the command here?

lintz, 2016-04-24:
Tested problem still exist on RCBugFix.

Tried with following folder names
1_test_å
2_test_ä
4_test_& (works)
5_test_ñ 
1,2,3 and 5 are impossible to enter.

m20 gives `>>> m20
SENDING:M20
Begin file list
/PRUSAU~1/LCD_BO~1.GCO
/PRUSAU~1/LCD_BO~2.GCO
/PRUSAU~1/LCD_BO~3.GCO
/ELEFANT/LFS_EL~1.GCO
echo:Cannot open subdir
Attempted to write invalid text to console, which could be due to an invalid baudrate
/BESTAL~1/PINE64~1.GCO
/BESTAL~1/PINE64~2.GCO
echo:Cannot open subdir
Attempted to write invalid text to console, which could be due to an invalid baudrate
echo:Cannot open subdir
Attempted to write invalid text to console, which could be due to an invalid baudrate
/MINION/0_2MIN~1.GCO
/MINION/0_06MI~1.GCO
/MINION/MINION~1.GCO
echo:Cannot open subdir
Attempted to write invalid text to console, which could be due to an invalid baudrate
/YI/YI_20_~1.GCO
/YI/MOUNTC~1.GCO
/HAPPYC~1/08MM_H~1.GCO
/RPI/SOAPIB~1.GCO
/RPI/SOAPIB~2.GCO
/FISH/FISH_F~1.GCO
/RC/HMINIH~1/WALLMO~1/BACKIN~1.GCO
/RC/HMINIH~1/WALLMO~1/2XQUAD~1.GCO
/RC/HMINIH~1/ARM31~1.GCO
/RC/HMINIH~1/ARM31X~1.GCO
/RC/HMINIH~1/MAINBO~1.GCO
/RC/HMINIH~1/MAINTO~1.GCO
/RC/HMINIH~1/U_TOP3~1.GCO
/RC/MICROQ~1/FIREFL~1.GCO
/RC/MICROQ~1/FIREFL~2.GCO
/RC/MICROQ~1/SL7_2~1.GCO
/RC/PEON23~1/PEON23~1.GCO
/RC/PEON23~1/PEON23~2.GCO
/RC/PEON23~1/PEON23~3.GCO
/RC/PEON23~1/PEON23~4.GCO
/RC/PEON23~1/PEON23~5.GCO
/RC/PEON23~1/PEON23~6.GCO
/RC/PEON23~1/1XPEON~1.GCO
/RC/TRICOP~1/AOMWAY~2.GCO
/RC/TRICOP~1/LANDIN~1.GCO
/RC/TRICOP~1/AOMWAY~3.GCO
/RC/XT_60_~1.GCO
/RC/305_SQ~1.GCO
/RC/F450/UPMODV~1.GCO
/RC/F450/DOWNMO~1.GCO
/RC/SPACER~1/SPACER~1.GCO
/RC/CHEERS~1.GCO
End file list`

thinkyhead, 2016-04-25:
@lintz As far as I know, all DOS 8.3 filenames are supposed to be plain old ASCII. So there shouldn't be any "invalid characters" in any of the file or folder names.

Can you also try `M111 S1` before you do `M23` or "Print SD File" from the controller? That will show the command that is being sent to start the print. That might also provide a clue.

thinkyhead, 2016-04-25:
Also… Can you use a disk tool to see the formatting of your SD Card? Is should be FAT 16. Windows 8 and later might not format them correctly, [according to this post](https://groups.google.com/d/msg/wanhao-printer-3d/-WSv75lK6xI/8eDN1CwxgQwJ).

thinkyhead, 2016-04-25:
I stand corrected. We should have working FAT-32 at this point. See #2080 for example!

thinkyhead, 2016-04-25:
@lintz Please try [this branch](https://github.com/thinkyhead/Marlin/tree/rc_debug_lsdive_issue_3593) which will generate **a lot** of logging of filename activity. It will be a huge log, but if you can find the interesting / important part and paste it here that will help a lot!

https://github.com/thinkyhead/Marlin/tree/rc_debug_lsdive_issue_3593

Blue-Marlin, 2016-04-28:
Today i tried how folder names with German umlauts behave. I made some folders with Win7 and a card-reader.
I see the same problems. The names show well on the display, but you can't enter the folders. Also M20 can't enter the folders to display their content.
Files with umlauts are well displayed on the lcd but they are not executed. 

On the console window (RH) all the umlauts are displayed as '?'.

Error messages before the related folder-names
```
M20
...
22:56:41.445 : echo:Cannot open subdir
22:56:41.445 : ?TEST
22:56:41.457 : /U-TEST/OKTESTG2.G
22:56:41.457 : echo:Cannot open subdir
22:56:41.458 : M?TEST
22:56:41.462 : echo:Cannot open subdir
22:56:41.462 : M?TEST2
22:56:41.462 : ?123.G
22:56:41.462 : 123?.G
22:56:41.462 : ABC?123.G
22:56:41.467 : End file list
```

Klick on folder returns to top of folder.
```
23:02:36.284 : echo:Cannot enter subdir: M?TERST
```

Klick on file returns to status screen:
```
22:53:18.825 : echo:enqueueing "M23 abc?123.g"
22:53:18.825 : echo:enqueueing "M24"
22:53:18.882 : echo:Now fresh file: abc?123.g
22:53:18.882 : open failed, File: abc?123.g.
```

The file listing in Pronterface looks more like:
```
echo:Cannot open subdir
šTEST
/U-TEST/OKTESTG2.G
echo:Cannot open subdir
MšTEST
echo:Cannot open subdir
MšTERST
š123.G
123š.G
ABCš123.G
End file list
```
where all 'š' in reality are 'ü'.

In Octoprint the files in folders, and the files in the base directory, with umlauts do not appear in the sd-card filelist.
Even on the console they are filtered out - together with the error messages.

@lintz 
I suggest to simply avoid the use of your Scandinavian characters for file or folder names.
For some lucky circumstances Marlin can display the chars correctly, but handling them, for now, is impossible.
Even the hosts all show different (but all wrong) behaviour.

@thinkyhead 
The 8.3 filenames seem to be mangled somewhere when opening folders or files. Reading them seems to be ok.

Blue-Marlin, 2016-04-29:
SdBaseFile.cpp make83Name()
```
if (i > n || c < 0X21 || c > 0X7E)goto fail;
```
As far as i know chars above 0x7f are allowed again.
https://en.wikipedia.org/wiki/8.3_filename
```
Values 128–255 (though if NLS services are active in DOS, some characters interpreted as lowercase are invalid and unavailable)
```

I'll experiment with this tomorrow.

thinkyhead, 2016-04-29:
I look forward to the results. Perhaps the logging branch I linked above can help, as it prints out all the individual bytes.

https://github.com/thinkyhead/Marlin/commit/f575b775944c0b14117aac70828f4d768d8d552a

Blue-Marlin, 2016-04-29:
Sorry - will last a bit longer. Used all the time this morning for the German translation. Later i searched for proper long file name to 8.3 conversions. What's done in http://elm-chan.org/fsw/ff/00index_e.html looks very complete, but maybe a bit to heavy for us. For now i'm just looking for a 8.3 check, not mangling or refusing, Windows created 8.3-filenames. 
```
- if (i > n || c < 0X21 || c > 0X7E)goto fail;
+ if (i > n || c < 0X21 || c = 0X7E)goto fail;
```
seems to be enough for now, but needs some more testing.

thinkyhead, 2016-05-02:
All characters (except DEL) are now good with #3661. Thanks!

closed 2016-05-02 by thinkyhead