MarlinFirmware/Marlin #3595: Bed level probe offsets

Grogyan, 2016-04-22:
I have been using grid mode when bed levelling, 
However with Mesh Levelling, it isn't totally clear that the user also needs to set the offset which is located under the Bed Auto Level.
This should be much more clearly defined, as to avoid confusion.

thinkyhead, 2016-04-23:
@Grogyan Can you be more specific, please? Which "offset which is located under the Bed Auto Level" are you referring to?

Grogyan, 2016-04-23:
This section in configuration.h

  // Z Probe to nozzle (X,Y) offset, relative to (0, 0).
  // X and Y offsets must be integers.
  //
  // In the following example the X and Y offsets are both positive:
  // #define X_PROBE_OFFSET_FROM_EXTRUDER 10
  // #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
  //
  //    +-- BACK ---+
  //    |           |
  //  L |    (+) P  | R <-- probe (20,20)
  //  E |           | I
  //  F | (-) N (+) | G <-- nozzle (10,10)
  //  T |           | H
  //    |    (-)    | T
  //    |           |
  //    O-- FRONT --+
  //  (0,0)
  #define X_PROBE_OFFSET_FROM_EXTRUDER 40  // X offset: -left  +right  [of the nozzle]
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -1.15   // Z offset: -below +above  [the nozzle]

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.

  #define Z_RAISE_BEFORE_PROBING 10   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 10    // How much the Z axis will be raised after the last probing point.

thinkyhead, 2016-04-23:
Actually you don't need any of those settings for Mesh Bed Leveling. Are you getting a compiler error right now due to one or more of these being undefined? I use Mesh Bed Leveling myself, and have not seen any errors related to those defines.

Grogyan, 2016-04-23:
My mistake then. No compile errors.
I had been using grid leveling,  and just tried mesh leveling,  but
couldn't rack my head around the probe offset to the nozzle.  Thought it
was related.

thinkyhead, 2016-04-23:
No worries. Mesh Bed Leveling and Auto Bed Leveling are pretty distinct from one another, but slowly coming closer together.

closed 2016-04-23 by thinkyhead

Pmant, 2016-04-24:
How do you know where it is actually probing if you don't have the offsets?

Blue-Marlin, 2016-04-24:
@Pmant 
In Mesh/Manual bed leveling no probe is involved. No probe -> no probe offset.

Pmant, 2016-04-24:
> In Mesh/Manual bed leveling no probe is involved. No probe -> no probe offset.
So you move Z until the nozzle slightly touches the bed? Sorry I never used MBL.

thinkyhead, 2016-04-25:
> So you move Z until the nozzle slightly touches the bed?

@Pmant Yes. At each point, bring the nozzle to where paper slightly catches under it. You don't even need to look at the printer. It's pretty quick.

thinkyhead, 2016-04-25:
https://github.com/MarlinFirmware/Marlin/wiki/Manual-Mesh-Bed-Leveling