MarlinFirmware/Marlin #3599: Arduino IDE 1.6.x + Marlink Arduino Addons + Gen7

andreiciocanvlad, 2016-04-23:
I can't get to show Gen7 in the board menu with the addon files. How to fix this ?

thinkyhead, 2016-04-24:
I think you have to actually select "Sanguino" as the board. (Someone correct me if I'm wrong.) However the Sanguino files, the package download info file, and  need an update before they will work. I'm working on an update so you'll be able to just update the package from within Arduino IDE. Meanwhile, I will have the patch up shortly at https://github.com/MarlinFirmware/MarlinDev/tree/MarlinPlatform-1.6.7-b if you want to manually replace the package yourself.

thinkyhead, 2016-04-24:
I used the right checksum for the download file (two different tools give the same hash) but Arduino IDE doesn't like it. So, for now just download the .zip and install manually.

closed 2016-05-11 by jbrazio