MarlinFirmware/Marlin #3600: Question about filament change (not but or anything)

CoderSquirrel, 2016-04-23:
I remember, not right where here i've tried to found but i didn't, a topic about changing filament, and there were a discussion about only having it in printing time, about there's no routine about changing filament before, or after printing.
I'd like to know if it still on, or if you guys, or anyone else, have done this routine, because i have done something like it in this past few days  and i'd like to share, but i wanna know first if i has been solved to compare my code.
Thanks

CoderSquirrel, 2016-04-23:
#3264

jbrazio, 2016-04-23:
Please share, we will gladly review it.

CoderSquirrel, 2016-04-23:
Sure @jbrazio, i have done this in RC3, i will now download the RC5 and make it.
Just and observation it only work with 1 extruder. As soon as i finish it i will share. 
What way is best, do a pr agains rcbugfix ou showing the code?

CoderSquirrel, 2016-04-23:
I've made pr #3601

closed 2016-04-23 by CoderSquirrel