MarlinFirmware/Marlin #3603: MKS GEN v1.3  - Z2  pins not defined

jataadroid, 2016-04-24:
Pin_MKS_GEN_13.h redirects to pins_ramps_14_EFB.h which redirects to pins_ramps_14.h 
Here   Z1 is defined as  46,48,62, and E1 (EXTRUDER #2)  as 30,36,34  on MkS GEN v1.3 PCB E1  is labeled Z2 for the second Z axis stepper motor. The second Z axis motor (Z2) has not been defined in the last Marlin release with pins  30,36,34.
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#define X_MIN_PIN           3
#ifndef X_MAX_PIN
  #define X_MAX_PIN         2
#endif

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#define Y_MIN_PIN          14
#define Y_MAX_PIN          15

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_MIN_PIN          18
#define Z_MAX_PIN          19

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30

#define SDPOWER            -1
#define SDSS               53
#define LED_PIN            13

jataadroid, 2016-04-24:
A work around has been located to redifine E1 to Z2 in Marlin.
http://forums.reprap.org/read.php?146,562754,562754#msg-562754

thinkyhead, 2016-04-25:
Did you encounter a compile error while building `RCBugFix`? Because this has already been fixed for a while. It simply gets assigned the first set of unused E pins:
```cpp
//
// Dual Y and Dual Z support
// These options are mutually-exclusive
//

#define __EPIN(p,q) E##p##_##q##_PIN
#define _EPIN(p,q) __EPIN(p,q)

// The Y2 axis, if any, should be the next open extruder port
#ifndef Y2_STEP_PIN
  #define Y2_STEP_PIN   _EPIN(EXTRUDERS, STEP)
  #define Y2_DIR_PIN    _EPIN(EXTRUDERS, DIR)
  #define Y2_ENABLE_PIN _EPIN(EXTRUDERS, ENABLE)
#endif

// The Z2 axis, if any, should be the next open extruder port
#ifndef Z2_STEP_PIN
  #define Z2_STEP_PIN   _EPIN(EXTRUDERS, STEP)
  #define Z2_DIR_PIN    _EPIN(EXTRUDERS, DIR)
  #define Z2_ENABLE_PIN _EPIN(EXTRUDERS, ENABLE)
#endif
```

closed 2016-04-25 by thinkyhead

jataadroid, 2016-04-26:
so if i indicate pins 36,34,30 for Z2 access in pins.h, do i need to modify E1? thank you very much for your attention to my issues

thinkyhead, 2016-04-27:
No. Leave the E1 pins as they are set in your board's pins file. If you have only one extruder, Z2 will automagically get its pins from the pins defined for E1.

thinkyhead, 2016-04-27:
In other words, you don't need to do anything but enable Dual Z Axis. Everything is handled for you.

jataadroid, 2016-04-27:
Ill try to find out how to do that, I havent seen anything yet that directly talked about enabling dual Z or  that about it being that simple. I have my X,Y,Z1,Ext,Z2 plugged in to correct spots, nothings been automatic so far. I hate to ask but where is some material on marlin for situations like this so i dont have to bother the community with beginner stuff. From some further searching I think this is what i need but so fresh at messing with arduino and coding in general( sincerely sorry lol ) im not sure:
// The Z2 axis, if any, should be the next open extruder port
#ifndef Z2_STEP_PIN
  #define Z2_STEP_PIN   _EPIN(EXTRUDERS, STEP)
  #define Z2_DIR_PIN    _EPIN(EXTRUDERS, DIR)
  #define Z2_ENABLE_PIN _EPIN(EXTRUDERS, ENABLE)

and modify it to this:
#ifndef Z2_STEP_PIN
  #define Z2_STEP_PIN      E1_STEP_PIN
  #define Z2_DIR_PIN       E1_DIR_PIN
  #define Z2_ENABLE_PIN    E1_ENABLE_PIN
#endif

thinkyhead, 2016-04-27:
Your proposed modification to `pins.h` should not have any effect. It just does what the code there already does, which is to assign the Z2 pins to the E1 pins (when `EXTRUDERS` == 1).

jataadroid, 2016-04-27:
Ive tested the hardware with no problems. I am unable to grasp the solution if the firmware is already set to recognize the 1 of 2 extruders and mirror Z to that  2nd extruder driver but is not my case.

WZ9V, 2016-04-27:
Did you uncomment this line //#define Z_DUAL_STEPPER_DRIVERS (188) in configuration_adv.h?

jataadroid, 2016-04-27:
you sir have clarified what i needed. Thank you very very very much. Especially for the line number. I knew i had to uncomment something. A perfect direct answer. Thank you again.

WZ9V, 2016-04-27:
I get lucky sometimes.  You just happened to ask a question about something I was already looking into.