MarlinFirmware/Marlin #3617: troubel with different line ending styles in different files

tkurbad, 2016-04-26:
Hi,

while trying to integrate the latest RC-updates into my fork, I realized that some files have Windows style line endings (CR+LF), while others use Unix style (LF only).

Examples for Windows style line endings would be README.md and Marlin/Sd2Card.cpp.
Unix style line endings are used, e.g., in Marlin/Marlin_main.cpp, Marlin/Version.h, and Marlin/Configuration.h

While this seems to be only a "cosmetical" issue, it can produce huge problems when merging upstream changes into a local fork. This is because many editors "guess" the line endings wrong, thus saving the files with line endings different from the original. During a merge then, the whole file can get replaced, which is most probably not what you would expect during a merge from upstream.

I therefore pleed for the line endings to be unified all over the code, preferrably towards the Unix style.

Thank you,
Torsten

bobc, 2016-04-26:
If configured right, I believe git should handle it automatically, although existing files might need cleaning.

https://help.github.com/articles/dealing-with-line-endings/

Roxy-3DPrintBoard, 2016-04-26:
We anticipate going 'Golden' soon.   It might make sense to doing this as a one of the final steps.

If you have strong feelings on the topic, please express them now with your rational.

thinkyhead, 2016-04-27:
The fix for this has been merged, so we should be all set. We'll enforce line endings in the Travis CI test soon.

tkurbad, 2016-04-27:
Thanks for the quick and positive response!

closed 2016-04-27 by tkurbad