MarlinFirmware/Marlin #3628: RC5 Servo Enabled Probe Homing

g2david, 2016-04-27:
1. Z does not raise before servo homing
2. The offset has not applied when enter in LCD

#define X_PROBE_OFFSET_FROM_EXTRUDER +5  // X offset: -left  [of the nozzle] +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -55  // Y offset: -front [of the nozzle] +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -11.9f  // Z offset: -below [the nozzle] (always negative!)

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.

  #define Z_RAISE_BEFORE_PROBING 40   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 15    // How much the Z axis will be raised after the last probing point.

  #define Z_SAFE_HOMING   // Use the z-min-probe for homing to z-min - not the z-min-endstop.
                          // This feature is meant to avoid Z homing with Z probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing.
                          // - Position the Z probe in a defined XY point before Z Homing when homing all axis (G28).
                          // - Block Z homing only when the Z probe is outside bed area.

  #if ENABLED(Z_SAFE_HOMING)

    #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
    #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).

  #endif

Blue-Marlin, 2016-04-27:
We are missing some important information to be able to help you.

What branch/version are you on when you see this? (If not RC6 or RCBugFig please try with one of these first)  (Sorry - title)
Some more info about your system. (DELTA, CoreXY, ... )
What kind of G29 (grid, 3-point, MBL)
Your probe configuration.

The best would be to copy your `Configuration.h` to `Configuration.h.txt` and drag it into your browsers editor window.

thinkyhead, 2016-04-29:
Have you tried setting `MIN_Z_HEIGHT_FOR_HOMING`?

g2david, 2016-05-01:
Hi thinkyhead,

thanks, it works.   When I merged with old RC staging Marlin source, I overlook this new setting.

It works perfectly. Thanks guys

thanks,
David

closed 2016-05-01 by g2david

thinkyhead, 2016-05-02:
I'm still not sure if this represents a bug. When there is **no Z endstop**, but there **is a Z servo probe**, the Z homing procedure should always raise the Z axis (even without `MIN_Z_HEIGHT_FOR_HOMING`) to make room for probe deployment.