MarlinFirmware/Marlin #3629: when is CW = CCW?

ruggb, 2016-04-27:


jbrazio, 2016-04-27:
The logic is simple, if the default is rotating the "wrong direction" enable the reverse feature. It's impossible to guarantee that CW will always be up, that's why the revere function is there.

On the comments we provide the most common hardware behavior but this does not mean all defaults will behave like that. Maybe we should put this in a clear manner on the comments.

ruggb, 2016-04-27:
great idea! On my arduino/ramps1.4/full graphics controller the default CW is UP. (And it has been since the start of time) The comment indication is that that default has changed and CW is now DOWN. The default has not changed, otherwise mine would be going DOWN (since it was previously UP).
So I am not sure your answer passes the smell test. U might want to verify that.

jbrazio, 2016-04-28:
@ruggb From a ux perspective CW should be down, it's more natural. `REVERSE_MENU_DIRECTION` was introduced for the cases where physically the encoder behaves in a different manner and CW is UP as people were working around this by setting `ENCODER_PULSES_PER_STEP` to `-1`.

closed 2016-04-28 by jbrazio

ruggb, 2016-04-28:
I am not sure u understand..............
The comment says:
// This option reverses the encoder direction for navigating LCD menus.
**// By default CLOCKWISE == DOWN. With this enabled CLOCKWISE == UP.
**//

_TO BE CLEAR, the comment would read:_
// This option reverses the encoder direction for navigating LCD menus.
**// If your CLOCKWISE direction moves it up, ENABLE the following to make CW == DOWN
**//
#define REVERSE_MENU_DIRECTION

which I had to do.

jbrazio, 2016-04-28:
:facepunch: Chuck Norris approves.

thinkyhead, 2016-04-29:
Sorry about the comment. I will improve it.