MarlinFirmware/Marlin #3630: I can't make my changes to LCD menu for babystepping

ruggb, 2016-04-27:
with the latest RC6-BF-ad0f70f I cannot make the changes I want to UltraLCD.cpp and have it work.
Previously I moved 2 sections from in and around the TUNE menu to in and around the START menu so that I can have babystepping available all the time instead of waiting until the printer is actually printing. At that point and with the function buried the way it is, it is useless to try to use it.
I would be appreciative if you can tell me how to implement this on this commit.

Again, as I submitted before, this function is useless to me and a lot of other ppl who use it as I do. It needs to be placed on the main menu at the top so that it is easily and quickly accessible before the Print starts since the adjustment needs to be done on the extruder clearing lines. By the time I can access the function, the print has already started on the model first layer.

AnHardt, 2016-04-27:
That wasn't sooo difficult.
https://github.com/AnHardt/Marlin/pull/31

ruggb, 2016-04-28:
Thank U AnHardt
That is pretty much what I had
the only problem is it does not work. (yours or mine)
On the first push of my button it goes to the main menu

Info Screen
Prepare
Control
No SD Card

what I had previously when I modified it in RC5 was

BabyStepping
Info Screen
Prepare
Control
No SD Card

This way it takes 2 sequential button pushes to get to the function

There seems to be a lot of new code. Is something in there stopping it from displaying unless it is printing?

ruggb, 2016-04-28:
@AnHardt 
actually, it was me again. My version worked exactly the way I wanted it to WHEN babystepping was enabled. Somehow I missed the enable. I sure will be happy when I don't have to do that any more.
Your mod puts it below PREPARE, mine puts it at the top so there is no scrolling to get to it. Fastest access that way.
But thank you so much for your response.

closed 2016-04-28 by ruggb

AnHardt, 2016-04-28:
Come on. Shifting a menu item up or down in a menu is trivial. 
This was example code especially for you. I don't intend to have this in the main repository.

ruggb, 2016-04-28:
for an adult, walking is trivial. Ever watch a baby try to do it?
I appreciate your help. thx.