MarlinFirmware/Marlin #3635: Updating to latest version

jonnyk2, 2016-04-28:
So I am not sure what I am missing? But I uploaded the latest version of Marlin because I wanted the enhanced auto leveling . When I run a g28 it goes nuts and Rams the head into the bed. I have used a text comparison site to check the differences in my working configuration.h as well as the .adv files. The only differences are where I need to change some of the measurements and offsets etc. is there another file that I need to compare as well?

Blue-Marlin, 2016-04-28:
We are missing some important information to be able to help you.

What branch/version are you on when you see this? (If not RC6 or RCBugFig please try with one of these first)
Some more info about your system. (DELTA, CoreXY, ... )
Your probe/endstop configuration.

Info where/what endstop/probe you really connected.

The best would be to copy your Configuration.h to Configuration.h.txt and drag it into your browsers editor window.

jbrazio, 2016-04-28:
@jonnyk2 you should check `Configuration.h` and `Configuration_adv.h`.
Depending on how old you previous configuration is some of the `#define` options have changed their names, please double check that.

jonnyk2, 2016-04-28:
[latest config_h.txt](https://github.com/MarlinFirmware/Marlin/files/241413/latest.config_h.txt)
[latest config_adv.txt](https://github.com/MarlinFirmware/Marlin/files/241414/latest.config_adv.txt)
[older configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/241415/older.configuration_adv.txt)

jonnyk2, 2016-04-28:
Missed one sorry

[older configuration_h.txt](https://github.com/MarlinFirmware/Marlin/files/241417/older.configuration_h.txt)

jonnyk2, 2016-04-28:
There are some differences in the offsets and rod measurements ,but thats because I re-uploaded the older firmware and I am using that. I would change those parameters before uploading the new firmware. Something else was causing the issues though.

jonnyk2, 2016-04-28:
Jbrazio, I checked again in those areas that you recommended , but I don't see anything missing

jbrazio, 2016-04-28:
I'm not an expert on DELTA but diff the files I see the following:
`(-) old config, (+) new config`

```cpp
-#define INVERT_X_DIR true    // for Mendel set to false, for Orca set to true
-#define INVERT_Y_DIR true    // for Mendel set to true, for Orca set to false
-#define INVERT_Z_DIR true    // for Mendel set to false, for Orca set to true
+#define INVERT_X_DIR false    // for Mendel set to false, for Orca set to true
+#define INVERT_Y_DIR false    // for Mendel set to true, for Orca set to false
+#define INVERT_Z_DIR false    // for Mendel set to false, for Orca set to true
```

If you printer is a Kossel, why don't you start with one of the `example config` files for Kossel ?

esenapaj, 2016-04-29:
>[latest config_h.txt](https://github.com/MarlinFirmware/Marlin/files/241413/latest.config_h.txt)

It looks like that it's similar to [configuraton file](https://github.com/RichCattell/Marlin/blob/master/Configuration.h) of [customized version of the Marlin](https://github.com/RichCattell/Marlin)...

thinkyhead, 2016-04-29:
Is this an issue of older Configuration files causing mayhem? I'm liking the configuration versioning idea more.

jonnyk2, 2016-04-29:
Thanks for the all the suggestions folks. I am a little confused on some of the advice because I am brand new at this. From the feedback it looks like I need to start off with a different download of the latest Marlin build, one that is specifically for a kossel type 3d printer? Or like jbrazio said start with an example configuration of a Kossel. I am not seeing where the example config files are located.

Blue-Marlin, 2016-04-29:
https://github.com/MarlinFirmware/Marlin/tree/RC/Marlin/example_configurations/delta

jbrazio, 2016-04-29:
@jonnyk2 do you need more help ?

jonnyk2, 2016-04-29:
I think I'm good for now thanks for your help

closed 2016-04-30 by thinkyhead