MarlinFirmware/Marlin #3636: Making forks for machines manageable

Alex9779, 2016-04-28:
Hey dear Dev team,
I have been thinking about this for a few days now and maybe I am missing something but I try to describe what I am looking for.

I forked the repo to make a version of Marlin for my machine.
Then I changed the files with the configuration to fit my machine.
I changed Configuration.h, Configuration_adv.h, dogm_bitmaps.h, the PINs file for my board and Version.h.
Then I tried to keep up with the Marlin development in RC and RCBugFix branch, you may have noticed, I opened some issues that you fixed very quick. Many thanks for that. I like to be at the bleeding edge, solve problems, help to find bugs and stuff, to improve the stuff I use. Either by real contributions or by testing and reporting.
But every now and then I was not able to merge the changes cleanly into my repo because of changes in the files, in lines, I had to change for my configuration.

I am not a Git star, I am using it and I think I cover the basics so maybe I am not aware of some magic option or workflow which could solve my problems. So correct me if I am wrong...

What I want to propose is that the base Marlin repo, which is not really functional for any machine, should move the files you typically have to change to configure Marlin to some other location, maybe a "Default configuration" folder or something.
So when loading the repo the firmware would not compile because necessary files are missing, but this could be covered in some documentation: "To start copy file x, y, z, to the root directory and modify them."

This way the changes to configure Marlin for a machine are always in files which are not in the original repo. So they will be never changed by the Devs and merges will not end in conflicts.
To keep the "user" safe from using deprecated options in his config file you could use the already installed sanity checks. For example if an option is removed or renamed the check could add compiler warnings or errors just like you do now.

The result will be a fork that can easily keep up with changes in the main repo without having to solve conflicts because only one line of comment or something minor or unimportant for your specific config changed.

I thought about other options, writing an own config file and include it, using undef so I have not to change the original file but there are borders I cannot cross, like the const for the bitmaps, or nested defines in if clauses.

So I wrote this to get some feedback if I do something wrong or if I understand something wrong. And if not if it would be possible to change something to make life easier...

Kind reagards,
Alexander

thinkyhead, 2016-04-29:
Are you keeping your custom configurations in a branch separate from our evolving branches? In other words, did you do something like this?
```
$ git checkout 1.1.0-RC6 -b alex_robot_configured
$ git push --set-upstream origin alex_robot_configured
. . . or . . .
$ git checkout upstream/RCBugFix -b alex_robot_rcbugfix
$ git push --set-upstream origin alex_robot_rcbugfix
```
Because then you can use `git rebase` to keep up with the changes. Usually easier than `git merge`.
```
$ git fetch upstream
$ git rebase upstream/RCBugFix alex_robot_rcbugfix
$ git push -f
```
There may still be "merge conflicts" but I find them easier to clean up working on top of the latest code (as with `rebase`) rather than trying to `merge` the latest code onto my custom branch.

Alex9779, 2016-04-29:
Yes I have separate branches.

After your comment I read again the rebase documentation and some tutorial. Until now I didn't use rebase because a normal merge was enough for me. Currently all my other work with Git is mostly for me alone so merging is no problem as there are not a lot of branches and changes in a lot of branches. So there was no need for me rebase and I have to admit I didn't understand it completely.
Now I think I know what it is about.

But what I read everywhere in the guides this is not working if you work in public.

So your suggestion works if I clone to my local repo, then branch maybe RC, do my changes and compile. You change something, I rebase on my branch and so on.
I still works if I push my repo to my private Bitbucket server to backup.

But I have concerns publishing a configured firmware repo on Github for others to see. They can then fork or clone the repo and here comes the fun. You change something, I rebase and have to force push and all the others have the mess...

I know that if you do what I proposed I will have the configuration in my repo and forks of my repo will have the problem I described.
But the problem at this stage may not be so present like it is now as "Marlin" is just like a base not working without a proper configuration. Sure you can build the repo as it is but is there actually a machine the base config is for without any change in the configuration files?

Alex9779, 2016-04-29:
To add to my last comment if someone fork my repo with the config he can easily change his upstream to Marlin itself if he does changes to my config.
And then he will have the advantage that his config, though based on mine, is not an altered version of some config in the Marlin repo and will not be change with commit of you.

Alex9779, 2016-04-29:
Thinking about the options and the possible workflows I am still convinced it would make life a lot easier if the relevant files for configuring Marlin wouldn't be present at the locations where they are needed to compile.

Let setup another possible situation: I fork, make my config, someone clones it and makes change to my config. How would it be possible for me and him to make changes to the codebase for testing or to add special features?
I have in mind that this could also be easily done if I just make such changes in separate branches with come from base Marlin without my config. I have that additional feature branch which I can merge easily into my configuration branch because the config branch does not have changes in the code base and the creature branch has not changes in the configurations because they are not there.
And so does the one who has forked my configuration branch can easily integrate the changes from the feature branch...

thinkyhead, 2016-04-29:
Go through this once or twice before you decide how to proceed:

http://pcottle.github.io/learnGitBranching/

Alex9779, 2016-05-06:
Thanks, I really did it.
But I still don't get what you wanna tell me. Since this is a beginners guide you are pointing me at, I assume I am too dumb to get it.

I though about all the alternatives again and again.
Merging: produces more conflicts and I have to take care that my altered config values are not overwritten again.
Rebasing: produces less conflicts because git tries to do my commit on top of the new changes
Cherry picking: is cool, I used it before but I have to select what I want.

I had a look at the Dev repo, there you change the config structure to something that may be what I request now. Didn't dig into it very deep but it looks like...

My goal is to do an automated nightly build of the RCBugFix branch but this does not work out and maybe will not work in the current development state.
There are several prerequisites, at first the config has to be independent from the code base and the changes made may not include changes to configuration options, deprecate options or introduce new ones.
Looking back at what happend this is not guaranteed at the moment, RCBugFix does not only fix bugs but also reorganizes the code...

So I am going to wait and do it with merging as I do not want to rewrite the history in a public repo every time with rebasing...

thinkyhead, 2016-05-08:
> So I am going to wait and do it with merging as I do not want to rewrite the history in a public repo every time with rebasing...

That's not how it works. You would only be rewriting the history in your own branch, and consequently any PR based on that branch. It has no effect on Marlin. When we merge, we get the commits that you have at the time. We never see the ones you redid, threw away, etc.

Alex9779, 2016-05-08:
Ok then I think I am going to reopen something config related.m if there is no hope for MarlinDev config.

I know it will work for my branches and that you do not see them but I think of the people forking MY repo. With the public repo I was speaking of I meant mine not the base Marlin.

So I have no problem but it would be worst practice for me to publish a repo I rewrite history every week or two...

Alex9779, 2016-05-08:
So to pick this up again, I thought about this and all I would request is an additional folder "default_configuration" where the "Configuration.h" and the "Configuration_adv.h" are located.
That would be enough for the moment to fit my needs. I changed the boot up logo too and something in the PINs for the setup we have on the BigBox but that might need some additional changes to extract that setting for the bitmaps from the file and to change the logic of the PIN files if they would not all be present where they are now.

To get a working firmware you have to copy those two files to the root of Marlin or just make your own, alter if you want, and then compile.
Again, I think the default configuration is not for any available box so nearly everyone will change that.

I just wanna be as close to you as possible so I don't wanna just copy your code and make a repo and then from time to time pick fixes or new features into my "fork".
I want to publish the repo to others, so rebasing is no option for me, I'd rather go the hard merge way if this leads to nothing.
I know I am repeating myself, but this "little" change, making Marlin just a not working "base", would help me, maybe others, to just fork, add a config and merge changes of you later without the hassle of merge conflicts.

There would be nothing else to change for you because you are doing the sanity checks already for new or changed configuration options.
And to be honest what are you doing in your Travis? You copy all the test configs and build them, you are not building the default config...
So what would be the problem or the concerns of moving them out of the way?

thinkyhead, 2016-05-09:
> what would be the problem or the concerns of moving them out of the way?

You will need to alter your `.travis.yml` so it copies the "clean configs" initially from the moved location instead of the default one.
```yml
  #
  # Backup Configuration.h, Configuration_adv.h, and pins_RAMPS_14.h
  #
  - cp Marlin/default_configuration/Configuration.h Marlin/Configuration.h.backup
  - cp Marlin/default_configuration/Configuration_adv.h Marlin/Configuration_adv.h.backup
```
This will allow the `restore_configs` helper script to work without changes

> I think of the people forking MY repo

The more your fork diverges from any particular release of Marlin, the harder it will be to integrate new changes, or to contribute changes from your fork. So it is recommended not to mess with the file structure too much if you want to share code both ways.

As I understand it, people can only make a fork of top-level repositories. So you would need to make a new repository and copy the Marlin files into it rather than using the "fork" button on Github.

Alex9779, 2016-05-09:
Ok think we are talking at cross purposes...

I checked your Travis file again, and I was wrong you actually compile with the default configs. Maybe the default config is for a particular box but that does not matter to me...

> The more your fork diverges from any particular release of Marlin, the harder it will be to integrate new changes, or to contribute changes from your fork.
That is what do NOT want. I just want to have a fork with a configuration for an other box. I want to stay as close at possible with your code so I can upgrade my machine with the newest changes to help testing and improving and to find possible problems, maybe bugs.

I don't wanna move directories or files, I want YOU to move the two files but not change ANYTHING else. So all other files require Configuration.h and Configuration_adv.h at the location where they are now. But if you just clone the repo, open in Arduino IDE you will not be able to compile because of the missing configuration.
So the way to set up a Marlin for a new box is to either copy those files from they new position and change them to what you need or just to recreated them your self. For example all the comments could be omitted.

> As I understand it, people can only make a fork of top-level repositories. So you would need to make a new repository and copy the Marlin files into it rather than using the "fork" button on Github.
That is what I was talking about.
But since I consider Marlin as a base which is copied/forked by most of the people and then they change the configuration, YOU could make the first iteration of forks more manageable by moving those files.

That way I have my own configuration files at the location where all other files need them BUT since they are not at those location in your repo you won't change them ever (can't change something which is not there). So they never will appear in a merge. Only the files normally not altered to configure Marlin for a box will be changed then by you and those should merge flawlessly.
If I didn't make changes to those ones too which I normally don't or I would do it in a way to make that also more manageable for example I make new special function only for my box. I could create additional files and just include them. Those include lines also are no problem when merging your changes again into my fork because those are new lines. The conflicts only occur when trying to merge changes to same lines.

Again my goal is to just provide a source repo with an initial configuration for the box. No changes to code.
If I wanna contribute I will make an other branch from RCBugFix or RC, make the changes and open a PR from that branch. If I have made the changes in a branch I qactually use for compiling a FW for my box then I can cherry pick the changes to such a branch, squash them and make the PR.
The workflow would be so much easier...

thinkyhead, 2016-05-09:
We might very well make the change you request. But I don't think we will be doing this for 1.1.0. It's too late in the process.

For now, please just contribute your example configs and we will put them into the `example_configurations` along with the other well-specified machines.

jbrazio, 2016-05-09:
TL;DR

I keep a copy of my config files somewhere on my PC, when I want to upgrade I do:
```
git fetch --all
git reset --hard HEAD
git rebase
```

Then I have a quick look if something changed drastically by diff my config backups with the standard configs an finally I copy over my config files.

And trust me, as a developer you do this multiple times.. :-)

Alex9779, 2016-05-12:
@jbrazio yeah sure that is fine if you only manage configs for your own machines but not if you intend to publish them as I described above.

@thinkyhead sorry I know I am too verbose :)
Though I don't think you want the config I currently have because the BigBox offers too many setups even though it is not very old...

How shall I continue? Close and reopen as a feature request here or in MarlinDev?

thinkyhead, 2016-05-12:
@Alex9779 Supply your configs in a PR against `RCBugFix`, and this will give us a chance to see how they fit. Then we can decide how to proceed as far as the main Marlin project goes.

closed 2016-05-24 by jbrazio