MarlinFirmware/Marlin #3639: Compilation on board without SD card reader fails

dethegeek, 2016-04-28:
Hi

I'm trying to compile Marlin for a Minitronics without SD card reader.

I left the following commented out in Configuration.h
```
//#define SDSUPPORT // Enable SD Card Support in Hardware Console
```
and the source failed to compile because card is not defined.

After searching, in found the following : 
```
#ifdef SDSUPPORT
CardReader card;
#endif
```

but in several places in the same file, all functions (or function calls) related to SD card reader are not conditionnaly compiled.

I added the missing #ifdef / #endif blocs to compile successfully, but not yet tested

jbrazio, 2016-04-28:
Why don't you try a newer version like [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) ?

dethegeek, 2016-04-29:
Hi

I tried to find the latest stable version. I guessed all tags containing RC are release candidates.

Do you recommend I switch to RCBugfix ?

thinkyhead, 2016-04-29:
I recommend that you use `1.1.0-RC6` for now, or `RCBugFix` which has a few patches since then. We know all about that bug, and we fixed it a long time ago.

J-Dunn, 2016-04-30:
> Why don't you try a newer version like RCBugfix ?

Why would anyone try an unstable release? Is that the normal situation with Marlin? 

> We know all about that bug, and we fixed it a long time ago.

That sounds rather off-hand. If it was fixed a long time ago why is it not included in the current stable release?

This is my first look at Marlin, the attitude displayed is not encouraging.

jbrazio, 2016-04-30:
Do you know the definition of RC and bug fix ? The unstable branch of Marlin is MarlinDev.

dethegeek, 2016-05-01:
Hi

I'll switch to RCBugfix. After 6 revisions I guess it should be useable in production use with a standard hardware.

Thanks

closed 2016-05-01 by dethegeek

thinkyhead, 2016-05-02:
> Why would anyone try an unstable release? Is that the normal situation with Marlin?

@J-Dunn No one is fixing bugs in older versions of Marlin at this time. If you would like to volunteer to maintain and patch version 1.0.2 and provide future updates to this branch, we would be grateful for the help. This is a 100% volunteer-run project.

Meanwhile, we recommend that when users discover bugs in older versions of Marlin that they confirm for us whether or not the bug has been fixed since then. If users find that the latest code (1.1.0-RCBugFix) works for them, and they encounter no other bugs, then we recommend that they stick with it until the 1.1.0 official release.

Version 1.1.0 is getting very close. As soon as it is released we will recommend it over versions 1.0.2-1 and earlier for most users.

> That sounds rather off-hand. If it was fixed a long time ago why is it not included in the current stable release?

Because no one is patching older releases right now. We are short on help. You seem quite passionate about the situation. If you have any programming skills and would like to offer your help to overcome our labor shortage, we would be most grateful!

WZ9V, 2016-05-02:
I fixed most of my problems with my Robo R1+ by moving to 1.1.0 RC6.  Before I moved to 1.1.0 it was really hard to get SD prints to start with Pronterface.

thinkyhead, 2016-05-02:
Is anyone else seeing what I am? The first time I connect Pronterface, I get the `M115` report, but when I try to send a command Pronterface says it isn't connected. The second time I connect it works fine.