MarlinFirmware/Marlin #3640: HELP CHANGE D9 and D8 in ramps 1.4

NervenCid, 2016-04-29:
Hello:

My D8 transistor is damaged so i want to change it with D9 instead... but i don't know how to switch those pines, i cannot find the output pin for the heated bed

Thank you

Blue-Marlin, 2016-04-29:
Why not?
Search for `HEATER_BED_PIN`
The other is, depending on your configuration, `HEATER_1_PIN` or `FAN_PIN`
(pins_RAMPS_14.h)

NervenCid, 2016-04-29:
Thank you for your answer:

in the current version i find that the fan is 9 but i don't find the HEATER_BED_PIN for the EFB configuration

Thankyou

Blue-Marlin, 2016-04-29:
If you cant find it literally it's the else
```
#if MB(RAMPS_14_EFF) || MB(RAMPS_14_EEF) || MB(RAMPS_14_SF) || MB(RAMPS_13_EFF) || MB(RAMPS_13_EEF) || MB(RAMPS_13_SF)
  #define HEATER_BED_PIN   -1    // NO BED
#else
  #define HEATER_BED_PIN    8    // BED //  !!!!!!!!!!!!!!!!!!!!!!!!!!! HERE !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
#endif```

NervenCid, 2016-04-29:
Hello again:

Thank you for your answer

I'ts worked :) Thank you very much

I was using v 1.0.8 but i don't know if i can combine this file with this file

But thank yu so much :)

closed 2016-04-29 by NervenCid