MarlinFirmware/Marlin #3645: Temperature reading update

polo78, 2016-04-29:
Hi,
I would like to know how frequently the temperature is read by marlin.
Thanks

thinkyhead, 2016-04-30:
I believe the temperature ISR `ISR(TIMER0_COMPB_vect)` runs at ~125Hz because:
```cpp
  OCR0B = 128;
  SBI(TIMSK0, OCIE0B);
```

(I may be wrong about that – the documentation on this is hard to follow. So someone may correct me.)

One temperature sensor is read on every other interrupt, in order. So if you have 2 temperature sensors, they will each be read every 4 interrupts, or about 30 times per second. For each ADC sensor you add, it delays reading by 2 more interrupts.

Blue-Marlin, 2016-04-30:
Then when OVERSAMPLENR (16) samples are made current_temperature is updated -> ~2 times / second.

closed 2016-05-01 by jbrazio