MarlinFirmware/Marlin #3646: Scrolling text

sarf2k4, 2016-04-29:
I would like to suggest a scrolling text feature. Unsure if this can be done for displaying text manually via M117(?) for that is longer than the lcd capacity.

Primarily I would like to suggest for scrolling text for the filenames when reading the sdcard files.

Is this possible?

jbrazio, 2016-04-29:
Thank you for your interest making Marlin better.
You should re-open this topic on [MarlinFirmware/MarlinDev](https://github.com/MarlinFirmware/MarlinDev) for proper followup from the development team. We suggest you to use the `MarlinFirmware/Marlin#<topic id>` markdown to link back to this original topic.

closed 2016-04-29 by jbrazio

thinkyhead, 2016-04-30:
It is possible, but it will require some extra code in the LCD routine which handles extra long strings in the message area by redrawing them periodically.