MarlinFirmware/Marlin #3649: multiple enquecommands do not work

karabas, 2016-04-30:
I tried to add sounds and other gcodes to some places but found that only 4 enquecommands work in  marlin ( and maybe 6 in newest). F.e. I added menu function item:
   static void load_filament_test(){
    enquecommand("G91");
    enquecommand("G92 E0");
    enquecommand("G1 E50 F200");
    enquecommand("G1 E700 F1000");
    enquecommand("G1 E100 F200");
    enquecommand("G90");
    enquecommand("G92 E0");
  }
only first four work.
If I use  enquecommands_P with one bif string some commands are missed.
What I do wrong?

Roxy-3DPrintBoard, 2016-05-01:
So, this isn't simple.   And we certainly don't want to mess with this for the Release Candidates....  But I think I would argue if we are going to allow enquecommand()'s, we should allow an arbitrary amount of them.   The problem is if the enquecommand() fills up the command buffer, we need to give up control so the buffer can drain and we are not setup to multi-process commands.   We can't finish the current command without doing the remaining enquecommand()'s.   And we can't do those without finishing the current command.   So we end up in a Catch-22.   

@karabas    Right now the command buffer is not deep enough to allow that many separate commands.   It would be good for you to figure out a way to use less buffer space.   One option might be for us to allow a command that expands into multiple sub-commands.   For example, 
~~~
G999 G91!G92 E0! G1 E50 F200
G999 G1 E700 F1000!G1 E100 F200
G999 G90! G92 E0
~~~
I'm sure the people that really understand the command parsing and dispatching will cringe at this line of thought.  But we do need to find a way to let a function enquecommand()'s and much greater depth than what we tolerate right now.  

Long term, we should be looking at non-blocking command processing and maybe the ability to fetch commands from multiple buffer areas.   We do have some RAM available.   It may be some of that can be used to buffer up enquecommand()'s for currently executing functions???

karabas, 2016-05-01:
Ok, but when we know that printer is in idle state and no commands in buffer, we can feed a command and wait for it's finishing then feed next and so on. But when I try to insert delays(seconds) I found that commands are executed but still only first four of them! So how to force executing command immediatly after enquecommand, assuming that buffer is empty or how force inbuffer commands to be executed?

Blue-Marlin, 2016-05-01:
If you know the complete sequence you can use:
```
enqueue_and_echo_commands_P(PSTR("G91\nG92 E0\nG1 E50 F200\nG1 E700 F1000\nG1 E100 F200\nG90\nG92 E0"));
```
Then the string is the queue.

karabas, 2016-05-01:
New command? Excellent! It works. Thanx!

closed 2016-05-01 by karabas