MarlinFirmware/Marlin #3652: How to modify last G28?

karabas, 2016-05-01:
On deltas after print finishing endscript sends G28. I want to modify G28 exactly for this condition. What can be used as condition for IF statement? End_of _file?

thinkyhead, 2016-05-02:
@karabas What do you want to do instead of `G28`? Why not remove `G28` from the end of the gcode and put in the GCode you would prefer?

If you want to intercept the end-of-file and do something extra, you will need to add code to the `get_sdcard_commands` function in `Marlin_main.cpp`.

karabas, 2016-05-03:
Some people use  many slicers simultaniusly and wont edit end-Gcode every time. For big deltas parking of head after finishing is important to prevent drop down to model. Yes End-Gcode scrypt is most logical and easy solution.

thinkyhead, 2016-05-03:
@karabas The idea of Marlin is not to be too smart, but to faithfully execute commands as-given. If you can do something with the existing commands, even if it requires editing your slicer's end-gcode, that is preferable to modifying Marlin itself. Of course, most of us don't put `G28` at the end of our GCode because on Cartesian machines the head will crash into the print.

WZ9V, 2016-05-04:
@karabas - The dropping is due to turning off the motors in the end G-Code.  If you remove the command to power down the steppers which almost every slicer does by default then G28 will hold at the top.

closed 2016-05-11 by jbrazio