MarlinFirmware/Marlin #3656: marlin is growing.... why?

boelle, 2016-05-01:
why is it that marlin with only sd card enabled does not fit a 644p any longer? it did so when we took over from Erik Zalm's own private git repro

thinkyhead, 2016-05-01:
It often happens with development that elaboration happens first, and only afterward optimization. We could use help in the optimization part.

boelle, 2016-05-01:
:-/ i wish i was a code gury

thinkyhead, 2016-05-02:
Unfortunately we haven't heard anything but a few whispers for months about the size of the binary, so no one has been keeping an eye on this.

Some options are now enabled by default which probably weren't before. Try turning off every option you can find, including thermal protection, watchdog, etc. Make sure no bed leveling options are turned on.

Once you have the smallest build you can get, if it still doesn't fit, please report the size of the build and how many bytes we need to shave off to get it to fit and function on the 644p.

Roxy-3DPrintBoard, 2016-05-02:
@boelle   Hi!   How have you been?   Its been a long time since you have been around.   You need to stop in more often!!!!

In regards to your question, I haven't noticed any code size problem.   Marlin still fits into the 128 KB available on my PrinteRboard   It hasn't expanded enough to cause any problems.

Blue-Marlin, 2016-05-02:
Dear @Roxy-3DPrintBoard 
@boelle is talking aboute 64 KB !

Roxy-3DPrintBoard, 2016-05-02:
Oh!   And I thought I had a low capacity board!   This thinking might not be popular, but we need to be moving to higher performance (and larger capacity) processors and not spend huge amounts of time trying to squeeze new code into a miniature foot print.  

> why is it that marlin with only sd card enabled does not fit a 644p any longer? it did so when we took over from Erik Zalm's own private git repro

Well...  Remember the SD-Card file system has been upgraded to handle long file names and such.   I bet it got bigger.

Blue-Marlin, 2016-05-02:
RCBugFix 2016/5/2 (values only valid in this sequence)
```
Plain                            -> 52.220 / 2.993 Byte
+ SDSUPPORT                      -> 68.282 / 4.201 Byte +16.062 /+1208 Byte
- THERMAL_PROTECTION_BED         -> 67.984 / 4.195 Byte    -298 /   -6 Byte
- THERMAL_PROTECTION_HOTENDS     -> 66.906 / 4.157 Byte  -1.078 /  -38 Byte -> 26 / 50% still to big
- PIDTEMP                        -> 61.924 / 3.820 Byte  -4.982 / -337 Byte -> 24 / 46%
- TEMP_RESIDENCY_TIME 0          -> 61.708 / 3.820 Byte    -216 /  +-0 Byte -> 24 / 46%
- TEMP_BED_RESIDENCY_TIME 0      -> 61.708 / 3.820 Byte     +-0 /  +-0 Byte -> 24 / 46%
- Z_SAFE_HOMING                  -> 62.216 / 3.820 Byte    +508 /  +-0 Byte -> 24 / 46%
+ DISABLE_HOST_KEEPALIVE         -> 61.654 / 3.814 Byte    -562 /   -6 Byte -> 24 / 46%
- AUTOTEMP                       -> 60.718 / 3.797 Byte    -936 /  -17 Byte -> 23 / 46%
- ENDSTOPS_ONLY_FOR_HOMING       -> does not compile // on again
- SLOWDOWN                       -> 60.580 / 3.797 Byte    -138 /  +-0 Byte -> 23 / 46%
- ENCODER_RATE_MULTIPLIER        -> 60.580 / 3.797 Byte     +-0 /  +-0 Byte -> 23 / 46%
- SD_DETECT_INVERTED             -> 60.580 / 3.797 Byte     +-0 /  +-0 Byte -> 23 / 46%
- SDCARD_RATHERRECENTFIRST       -> 60.580 / 3.797 Byte     +-0 /  +-0 Byte -> 23 / 46%
- USE_WATCHDOG                   -> 60.548 / 3.797 Byte     -32 /  +-0 Byte -> 23 / 46%
- BLOCK_BUFFER_SIZE 8            -> 60.548 / 3.181 Byte     +-0 / -616 Byte -> 23 / 38%
- BUFSIZE 2                      -> 60.540 / 2.987 Byte      -8 / -191 Byte -> 23 / 36%
```
Nothing left to deactivate or decrease.

Roxy-3DPrintBoard, 2016-05-02:
@boelle   How much too big was it when you compiled the sketch?  It should tell you how much space the code requires and how much is needed.    The reason I'm asking is there are a lot of M and G commands that the Slicers do not use.   It might make sense to add some #ifdef's around them and make them go away...

If I'm reading the above post correctly, the SD Memory card stuff grew by 1200 bytes.   It should not be hard to reclaim that much code space if you want to do a one-off version.    I'll help you make a hacked up version if you want to do that.   Kind of like the one-off we did for your printer to get the Auto Bed Leveling going.

jbrazio, 2016-05-02:
We should ask ourselves if this is really a valid question.. what I mean is, should we continue to support old boards which are no longer the market standard ? This was exactly the same issue we had with old Arduino IDE/cores.

Roxy-3DPrintBoard, 2016-05-02:
> We should ask ourselves if this is really a valid question.. what I mean is, should we continue to support old boards which are no longer the market standard ? 

Agreed!   But Boelle gets a pass!  If he wants my help to keep his old board limping along, I want to give it to him!   :)

thinkyhead, 2016-05-02:
I think it's probably possible to get the code back down to a size that will fit onto a Very Small Board. We could make a special language file with shorter strings, find and eliminate redundant code (though code is usually the smaller part), and see if there are any static buffers that we can combine or place on the stack…

We simply haven't done all the optimization steps yet – not in any thoroughgoing way.

Meanwhile, another possibility for users of small boards is to stick with 1.0.2-1, assign someone to keep that patched and release a version 1.0.3 that still supports smaller boards but has all the bug-fixes that we've applied for 1.1.0. All someone has to do is go through the PRs one-by-one since 1.0.2-1 and apply every patch that is possible to apply.

jbrazio, 2016-05-03:
We got to start organizing ourselves in some way..
- We have language files who aren't maintained anymore (#3360, #3357, #3355, #3354, #3349, #3348, #3347)
- We have multiple building methods for Marlin which aren't maintained or are redundant (IDE, Makefile, platformio, IDE lib)
- We have compilation bugs on old stable releases we should at least fix #3639
- We should plan a target date for releasing the 1.1.0 GA
- We need to change get maintainers for each one of the example_configs we have on Marlin
- We have too many old issues (I've been closing some of them with no negative feedback so if no one is against I will start closing old "Feature Requests" and asking people to reopen them on MarlinDev)
- We need to decide if we want to keep the current dev method of having MarlinDev completely split from Marlin or should we change to a more "rolling release cycle"
- We need more than 24h in a day :-)

@thinkyhead @Roxy-3DPrintBoard @AnHardt @CONSULitAS @Blue-Marlin @esenapaj fire at will !

thinkyhead, 2016-05-03:
> decide if we want to keep the current dev method of having MarlinDev completely split from Marlin

I think we have a good plan for the next stage of MarlinDev, building on the 1.1.0 release and reorganizing into folders similar to MarlinKimbra. I think it's probably fine to keep it as a separate repository. After all, we can still `git add remote` from the release repository, and move branches back and forth between them. Are there other complications I'm forgetting…?

jbrazio, 2016-05-03:
That's clear !

I will try to explain myself, I was speaking more into the direction of how can we handle two distinct repositories and avoid what happened with MarlinDev where we reached a position that is über difficult to merge both repos to prepare the next dev cycle.

My definition of rolling release cycle is something like stable tagged releases in the branch "stable", then a "unstable" intermediate branch (currently RCBugfix) and a "development" branch.

We can tag the unstable branch with things like 1.1.0-RC5 etc.

The natural cycle is development > unstable > stable by always making sure they stay in sync. 

We can keep both repositories as this process will work with as many repos we want. I just find difficult from a user's perspective to correctly open issues and post PR when we have the source code spread.

Don't know, I'm just thinking about the problem.

Roxy-3DPrintBoard, 2016-05-03:
This may not be the right way to think about it...  But I kind of think the stable branches on the Marlin side just get bug fixes.   They should get more and more stable.    But on the development side when a branch with enough good stuff gets stable enough, it gets promoted to a 'Release Candidate' status and moved over to the Marlin side.

There may be problems with that thinking, but that is how I understood the plan when the two sides were formed.

jbrazio, 2016-05-03:
There are no right or wrong ways to think about this. :-)

IMO the problem is that is not **the** feature which is getting stable it's the entire code base.

If we did development by branches where each feature lives in an independent branch until it gets mature.. then it's the feature's code going stable, but this method is much more helpful with large dev teams with distinct responsibilities.

Roxy-3DPrintBoard, 2016-05-03:
That does sound like a better way to do things.   And once enough stable new features get merged together, promote that to be a Release Candidate over on the Marlin side of the fence.

jbrazio, 2016-05-03:
> If we did development by branches where each feature lives in an independent branch until it gets mature.. then it's the feature's code going stable, but this method is much more helpful with large dev teams with distinct responsibilities.

Just to be clear: I do not agree with "dev by branches by feature" for Marlin. I was giving an example of something that would not work for  us.

thinkyhead, 2016-05-03:
> avoid what happened with MarlinDev

Well the problem with MarlinDev was that it jumped the gun and tried to do a complete reorganization when we barely begun to work out the bugs and issues in the 1.1.x pre-release. Version 1.1.x should have been prioritized, and `MarlinDev` should have had a couple of experimental branches to play with reorganization and configurations while we worked on 1.1.x. The experimental branches would have helped us to shake out the best ideas, which we could then apply to 1.1.x to form the basis of 1.2.x. But instead of being a laboratory for ideas it became a point of divergence.

Impatience with the process, failure to consider the collaborative nature of Marlin, and a healthy dose of hubris led to that mess.

Fortunately we still have all the good ideas and lessons learned from `MarlinDev` — only now we have a better starting-point. And I think we have a much more collaborative crew.

boelle, 2016-05-04:
back.... my board is the OMC or OMCA

it has the 644p chip from atmel and a sd card on board.

if i remember with SD card en EEprom enabled it was 105%... will have to recheck

boelle, 2016-05-04:
about the mess.... the reason for that is that there are just to many roasters in the hen house... one is more busy to show off that he has be bigger and better tool than the other... so a lot of time is wasted on endless debates that leed to nothing and the dev work is all over the place.

that was the main reason why i left... and the list of issues has not shrunk in size either

jbrazio, 2016-05-04:
You should have a look at the "pulse" tab. Despite being a collaborative project someone must take the lead.

boelle, 2016-05-04:
well that is what i kind of tried.... i made milestones with about 5 issues in each.. the idea was to work on those 5 alone and forget about everything else...

once they where confirmed fixed i would batch of the next 5...

but again people where just picking them and it all got out of hand.... 

i can take this role again.... but i will not do it if i just waste my time

of course we should all agree what issues comes first etc etc

boelle, 2016-05-04:
but my first comment would be... why the heck a different git repro... is a branch for each different feature not enough... if the feature is voted a no no we just bring out the saw and cut the branch... just like when a tree gets sick... you trimm of the bad part

jbrazio, 2016-05-04:
I've been deleting those milestones lately as they had no more significance.

Multiple "parallel" milestones which are not minor version objectives are a mess.

Although I agree with milestones like 1.1.1, 1.1.2, 1.1.3 etc having each one of them assigned the bug fixes of the previous milestone. But we have to wait for 1.1.0 GA to "reboot" all management system and implement something as light as possible.

boelle, 2016-05-04:
what i had in mind..

get rid of DEV repro... lets focus on one tree and make that a good one :-D
go through issue list and close stuff that are not bug etc
cut any branches we dont need

just so we have a reasonble starting point

boelle, 2016-05-04:
btw... the sizes... 

if i just download the default RC branch like everyone else would..
set it to omca board, its 79% codespace

CONSULitAS, 2016-05-04:
@jbrazio 
>We got to start organizing ourselves in some way..

Right.

> * We have language files who aren't maintained anymore (#3360, #3357, #3355, #3354, #3349, #3348, #3347)

Perhaps lets start with finding maintainers? They should be at least collaborators. I could adopt the german one (with others), if needed. Look at the K8200 example. In the header i tried to document, that i try to be the caretaker for it. Update coming soon.

* We have multiple building methods for Marlin which aren't maintained or are redundant (IDE, Makefile, platformio, IDE lib)

We only need one: IDE
Because: Users (not nerds!) will not learn the other ones to update their printer.

* We have compilation bugs on old stable releases we should at least fix #3639

Perhaps not: As @thinkyhead states we are just few. Go on and don't look back.

* We should plan a target date for releasing the 1.1.0 GA

RIGHT! We will find bugs if we look for them. But if we merge, we will have more bugs. Endless story.

Proposal: Give Bugs a impact factor: critical / major / moderate / minor / cosmetic (like http://istqbexamcertification.com/what-is-the-difference-between-severity-and-priority/)

If there is no critical or major bug known lets release on 15.05.2016.

* We need to change get maintainers for each one of the example_configs we have on Marlin

Lets sort the configs in configs_maintained and configs_example. Short read me with people (care takers) to ask about the maintained ones and "at your own risk" for the example ones.

* We have too many old issues (I've been closing some of them with no negative feedback so if no one is against I will start closing old "Feature Requests" and asking people to reopen them on MarlinDev)

Right way.

But: Perhaps save the good ideas (only) on a documentation page with links to the issues.

* We need to decide if we want to keep the current dev method of having MarlinDev completely split from Marlin or should we change to a more "rolling release cycle"

Kill MarlinDev. To complicated. Nobody (except Thinky 😄) understands how to do it with github.

* We need more than 24h in a day :-)

No problem: There is a night with 12h more! 😁

boelle, 2016-05-04:
@CONSULitAS 

i think we think a like here.... we need to cut off what keeping us back

there will be bugs... but that should not hold us back from doing a release

if we fix releases to every 2 months we will just fix the bugs we can in that time... if we only manage to fix one bug then that is what we fix

boelle, 2016-05-04:
with eeprom the codespace is now 83%

boelle, 2016-05-04:
wooow

only 4 people left? ie that is what i can see here: https://github.com/orgs/MarlinFirmware/people

boelle, 2016-05-04:
my thoughts on new features.... we should judge every new one.. ie how many does it benefit... is it really needed etc...

and new features should not be our job collabs.... ie they should come as a PR 

we will have plenty of work on our hands allready

CONSULitAS, 2016-05-04:
@boelle 
Do you remember that i proposed (see https://github.com/MarlinFirmware/Marlin/issues/1184#issuecomment-77519521) a versioning like **2016-05**?

I would prefer a release every 2 months and thats it. **Marlin patch day**s:
* 15.01.
* 15.03.
* 15.05. --> go for it
* 15.07.
* 15.09.
* 15.11.

The next release could have the name 1.1.0 **and** 2016-05, like `1.1.0-2016-05` to get people used to this.

BTW: Welcome back! You where the one who brought me to Marlin. I saw your work for housekeeping around christmas 2014 and decided to be a part of it.

emartinez167, 2016-05-04:
Most of us left or at least stopped contributing/testing or participating due to the bad nature of the guy who had hijacked this project. Happy he is gone and there is a healthier environment now.

Regards,

Ernesto

boelle, 2016-05-04:
can you give a small hint on who?

boelle, 2016-05-04:
@CONSULitAS i dont remember that much.... but the idea is good...

if i should be a part again i need some access rights to create and delete stuff

but my first vote would be to get rid of the DEV repro...

boelle, 2016-05-04:
just enabled SD card... 109% code space

boelle, 2016-05-04:
with bed level it jumps to 146%, but i only did enable that for the fun of it

CONSULitAS, 2016-05-04:
perhaps people like @nophead @alexborro @fmalpartida @Wurstnase @MagoKimbra @foosel 

Just a short look at the PRs of the last 2 years...

CONSULitAS, 2016-05-04:
> if i should be a part again i need some access rights to create and delete stuff
but my first vote would be to get rid of the DEV repro...

@MarlinFirmware/owners @Roxy-3DPrintBoard @ErikZalm @daid 
I think @boelle asks to be a member of @MarlinFirmware/collaborators again.

👍 from me.

boelle, 2016-05-04:
hmmm i agree that a few of those could be "bad apples"

but i will be out today at swimhall.... take a vote among the 4 left if i can be trused with access rights

boelle, 2016-05-04:
not sure what level i need to create milestones etc... but like last time i would rather walk than destroy.

emartinez167, 2016-05-04:
I agree on the list. I can’t remember who specifically stated the same reasons, but the important thing is that there is now a chance to rebuild the team.

boelle, 2016-05-04:
and if we let new people in we should "have the balls" to get rid of bad elements so they dont destroy the environment again

and yes i speak in plain terms.... for the same reasons i dont eat candy still in the wrapping paper

emartinez167, 2016-05-04:
Could not agree more. Don’t let a rotten apple spoil the entire basket.

boelle, 2016-05-04:
:-) but can we use this thread to come up with a list of actions?

1. Delete DEV repro - Reason: We dont need it, new features could be a branch - if the branch is a no good one we do like garden people, trim it off

2. Close all feature requests with a remark that people should submit a PR with the feature and that we simply dont have the time to create features

please add your thoughts and i will update the list

emartinez167, 2016-05-04:
Before DEV is deleted, can we make sure the good stuff that has already been developed there is back ported into the main repo? Not talking about the “Marlin as a library” stuff, but some new functionality like the ADVANCE logic, for example. Not sure if that is already in the current main branch.

boelle, 2016-05-04:
of course... we pick the tree for the best fruits first :-)

jbrazio, 2016-05-04:
I will close Feature Requests older than 2 months in a couple of hours as this was already on my todo list.

Killing MarlinDev.. I agree but we should listen @thinkyhead opinion about this first.

Branch dev I don't like because it will mean a lot of people need write access. We should align "veto" power with write access and do not start giving it away (in the past a lot of issues happen by people deleting branches as far as I can tell).

boelle, 2016-05-04:
about write access.... i will not agree here.

people can submit PR's and we can either merge or reject.

boelle, 2016-05-04:
what we have ahead of us can be very well illustrated with the apple tree's outside my flat.

they have been allowed to grow out of control and the care taker have not trimmed them.

result is that the apples do not grow to a size where you can eat them as the tree have to feed to many... so when they finally do drop they are half rotten allready

we need to cut the number of repro's and trim the remaing repro to a size we can manage with the number of people we have left.

our advantage is that we live on different places so we can get benefits from the time zones. one can take over code work while the other sleeps

jbrazio, 2016-05-04:
Wait.. write access is different from a PR, write access means direct merge on the repo.

boelle, 2016-05-04:
ahh... agree.... direct merge should not happen anyway if it can be avoided... if i remeber correct a PR is more easy undone

boelle, 2016-05-04:
so far i have added 5 points to my list... 

have i missed something important?

CONSULitAS, 2016-05-04:
> pick the best fruits

agreed

> Close all feature requests

agreed, and:
```
6. Mark all issues with last comment older than 2 weeks with label "Status: Waiting for feedback" in red and tell author to test with RCBugFix.
Close issues with no feedback within 1 week.
Similar process with new tickets if running version is in question.
```

boelle, 2016-05-04:
well i'm off for a few hours :-D

we should also have a permanent google hangout we can use for dev contact so er dont spam with issues

boelle, 2016-05-04:
@CONSULitAS 

your point 6 is added to my list as point 7

boelle, 2016-05-04:
o&o

CONSULitAS, 2016-05-04:
Ok its topic 7. not 6.

Hangout: The old one? https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia

CONSULitAS, 2016-05-04:
last Hangout issue: https://github.com/MarlinFirmware/Marlin/issues/2583

[57 more replies]

closed 2016-05-05 by boelle