MarlinFirmware/Marlin #3663: babysteps

MoonshineSG, 2016-05-03:
after the "rc_singletons" update in RCBugFix Babysteps need a change as well

``` 
#if ENABLED(BABYSTEPPING)
    for (uint8_t axis = X_AXIS; axis <= Z_AXIS; axis++) {
      int curTodo = babystepsTodo[axis]; //get rid of volatile for performance

      if (curTodo > 0) {
        babystep(axis,/*fwd*/true);
        babystepsTodo[axis]--; //fewer to do next time
      }
      else if (curTodo < 0) {
        babystep(axis,/*fwd*/false);
        babystepsTodo[axis]++; //fewer to do next time
      }
    }
  #endif //BABYSTEPPING
```

`babystep(axis,/*fwd*/true);` should be `stepper.babystep(axis,/*fwd*/true);`

MoonshineSG, 2016-05-03:
btw, why is BABYSTEPS in Temperature ?

thinkyhead, 2016-05-03:
@MoonshineSG This is historical, but there are probably good technical reasons for it. The `temperature.cpp` file has its own ISR for reading sensors that is separate from the stepper ISR. It was picked as the best place to handle baby-stepping some time ago. We should take a look at that point in history and see what the reasoning was. Perhaps it's the same way in Sprinter or some other firmwares?

MoonshineSG, 2016-05-04:
This seams fix now. Thanks.

closed 2016-05-04 by MoonshineSG