MarlinFirmware/Marlin #3675: Nozzle crashes into bed after autobedleveling with RC6

a-sprenger, 2016-05-04:
Hi,

I used RC5 the last time. It work very fine with my Prusa (Ramps 1.4 EMF / 2 TMC2100 / 2 DRV8825
3 min endstopps, 1 inductive sensor at Pin32, Octoprint)
Autobedleveling was active. Not ever but it worked fine

Now, with RC6, the autobedleveling process works fine, but in the end of that procedure the z-axis don't stop moving down.

When moving down the z-axis manually nothing happens when triggering the endstop or the sensor. It moves down and down!

I' attach my two configuration.h files. 
Where is the mistake, or is it a bug?
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/248949/configuration.zip)

The console shows me with M119 that all endstops and the sensor as triggered when pressed

jbrazio, 2016-05-04:
Duplicate of #3575

a-sprenger, 2016-05-04:
I don't think so.
#3575 has been opened for more than 14 Days. At this time there was no RC6!
With RC5 I have no problems like that!

jbrazio, 2016-05-04:
Don't worry, I was just referencing the other one and kept this one open for evaluation.

Blue-Marlin, 2016-05-04:
What is you setting for ENDSTOPS_ONLY_FOR_HOMING (Configuration_adv.h)?
If that is active, together with your `min_software_endstops false` there is no mechanism to protect in z-min direction.
Since some time you should not need ENDSTOPS_ONLY_FOR_HOMING for a inductive probe any more.

Blue-Marlin, 2016-05-04:
What do you mean with
> the autobedleveling process works fine, but in the end of that procedure the z-axis don't stop moving down.

When exactly? Is the last probe made correct? Is it during the 'virtual' stowing of the FIX_MOUNTED_PROBE ? Is it a move downwards after G29?

Blue-Marlin, 2016-05-04:
For what reason is Z_PROBE_END_SCRIPT activated? It's an example for stowing an Allen Key Probe. Your defined max feedrate is 1,500 the ordered move in the script has F12000.

Blue-Marlin, 2016-05-04:
> Z_PROBE_OFFSET_FROM_EXTRUDER -9.93

What kind/brand/type of inductive sensor do you have? Most of them do work up to ~4mm. 1cm is enormous.

a-sprenger, 2016-05-04:
ENDSTOPS_ONLY_FOR_HOMING is active (it's activated in RC5, too; there it works)

Well, the last probe is correct; After that the carriage drives to the left side, and goes down. exactly then!

Z_PROBE_END_SCRIPT is used, because I use clambs! Oh, the F12000 will be changed, but the printer don't come to that point.

a capacitive pnp sensor is correct. 9.93 is a changed value. 
I know 7.93 is a high value, but I changed it after changing to RC5. If something happens I have a little bit more time to react!

Grogyan, 2016-05-04:
@a-sprenger can you try repeating a G28 Z0?
Does it ever go into a slow feedrate mode when doing the initial probing?

Blue-Marlin, 2016-05-04:
Please try G29 an other time. Keep an eye on, if the probing moves are stopped by the probe (LED) or the z-min-endstop. If they are stopped by the endstopp the correction G29 makes, will allow to move z 9.93mm below the bed. (keep your finger on the reset button)
Z_PROBE_OFFSET_FROM_EXTRUDER is not the distance between the nozzles tip and the (yellow?) bottom surface of the probe, but the distance from the bed to the bottom surface, when the probe triggers, minus the distance from the nozzle to the bottom surface. That normally should be about a mm. A cm is a extremely unlikely value.

Pardon? What do you mean with 'clambs'? And what has the Z_PROBE_END_SCRIPT to do with it/them?

scruffy4386, 2016-05-07:
Could the Negative 53 mm Y offset of the z probe sensor also be a typo leading the bed leveling off from what it should be? Also in Homing at line 649 the bed center is defined however you have not defined Manual Home Position above it. And lastly, what I learned through experience is that setting the Z offset far larger than what it truly is will end up with the adjusted Z zero being below the true zero leading to a collision.

thinkyhead, 2016-05-08:
Looks like a configuration issue. If we get no feedback soon, this can be closed.

Roxy-3DPrintBoard, 2016-05-08:
I wonder if this is related to: #3700

thinkyhead, 2016-05-09:
@a-sprenger Try the latest `RCBugFix` and see if your configuration triggers any errors in `SanityCheck.h`.

jbrazio, 2016-05-10:
Closing this issue as it is identified as a duplicate of #3700, let's proceed the discussion there.

closed 2016-05-10 by jbrazio