MarlinFirmware/Marlin #3677: Current rcbugfix (5/5/16) won't compile with //#define ENDSTOPS_ONLY_FOR_HOMING

samwyze, 2016-05-04:
I'm finding if #define ENDSTOPS_ONLY_FOR_HOMING is commented out compiler throws:

```
> Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> In file included from sketch\Configuration.h:42:0,
> 
>                  from sketch\Marlin.h:44,
> 
>                  from sketch\endstops.cpp:27:
> 
> sketch\endstops.cpp: In constructor 'Endstops::Endstops()':
> 
> macros.h:51: error: 'SWITCH_ENABLED_ENDSTOPS_ONLY_FOR_HOMING' was not declared in this scope
> 
>  #define ENABLED(b) _CAT(SWITCH_ENABLED_, b)
> 
>                          ^
> 
> sketch\macros.h:45:22: note: in definition of macro '_CAT'
> 
>  #define _CAT(a, ...) a ## __VA_ARGS__
> 
>                       ^
> 
> sketch\endstops.cpp:40:19: note: in expansion of macro 'ENABLED'
> 
>    enable_globally(ENABLED(ENDSTOPS_ONLY_FOR_HOMING));
> 
>                    ^
> 
> exit status 1
> 'SWITCH_ENABLED_ENDSTOPS_ONLY_FOR_HOMING' was not declared in this scope
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.
> 
```

Grogyan, 2016-05-04:
Please include your configuration.h

samwyze, 2016-05-04:
Here you go.
NB...In practice I don't comment that line so it is highly likely I have set an incompatible option somewhere! 
Thanks
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/249686/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/249687/Configuration_adv.h.txt)

thinkyhead, 2016-05-04:
Sorry about that. It should be…
```cpp
Endstops::Endstops() {
  enable_globally(
    #if ENABLED(ENDSTOPS_ONLY_FOR_HOMING)
      true
    #else
      false
    #endif
  );
  enable(true);
  #if ENABLED(HAS_Z_MIN_PROBE)
    enable_z_probe(false);
  #endif
} // Endstops::Endstops
```

closed 2016-05-04 by samwyze