MarlinFirmware/Marlin #3680: Serial communication errors: checksum and line number not +1

Sebastianv650, 2016-05-05:
As mentioned in https://github.com/MarlinFirmware/MarlinDev/issues/402 the advance thread, I'm getting communication errors when printing over USB. The error messages are:

```
Resend: 380
Error:checksum mismatch, Last Line: 385
[ERROR] Error:checksum mismatch, Last Line: 385
```
and
```
Resend: 386
Error:Line Number is not Last Line Number+1, Last Line: 412
[ERROR] Error:Line Number is not Last Line Number+1, Last Line: 412
```
Printer: Lulzbot TAZ 5 (Rambo board). I'm also writing for others from the Lulzbot forum, where we have threads from time to time that are also based on communication problems.

What I know through tests:
.) This is not related to my advance patch, but enabling it makes the problem worse as the main isr takes more time.
.) It's not the USB cable or the USB port.
.) It scales with print speed (higher ISR rate): At normal speeds (say up to 50-60mm/s) there is nearly no error, maybe one in some print hours. At high speed, especialy high speed combined with high needed command transfer rate due to short line segments, the error rate increases.
.) It was better in RC4, I get significant more errors in current RCBugFix.

I have not the skill to track this down in detail without help. I guess the main ISR can interrupt something in the USB communication, if that happens the transfer gets corrupted. There are more errors as the ISR rate or duration is increased because the probability increases that an ISR happens at the time where something with the communication is done.

If there are already ideas to test in some branches or forks, I would be happy to test them.

Blue-Marlin, 2016-05-05:
PR "Encapsulate Stepper, Planner, Endstops in singleton classes" #3631 can potentially affect the ISR-runtime.

AnHardt, 2016-05-05:
@Sebastianv650 
What host program do you use? What BAUDRATE do you use? Is the error rate significantly dropped when using a lower BAUDRATE?

If you are using RepetierHost in not ping-pong-mode, try ping-pong, or reducing 'Receive Cache Size' to about 75%.

If you have the suspicion your LIN_ADVANCE code takes much extra time like when the 'normal' stepperISR goes in dubblestep/quadstep-mode, try to put in:
```
      #ifndef USBCON
        customizedSerial.checkRx(); // Check for serial chars.
      #endif
```

Sebastianv650, 2016-05-05:
I'm using pronterface. I'm using 250000 baud, changed it to 115200 with no sucess.
I'm wan't using speed ranges where double stepping would occur in the last days, I went up to 80% of single step mode.

thinkyhead, 2016-05-05:
> #3631 can potentially affect the ISR-runtime

Not if the compiler is smart. Firstly, using `myObject.functionCall()` is not any more expensive than `functionCall()`. So that alone should not add any cycles. In "Embedded C++" we assiduously avoid `virtual` to ensure flatter compilation.

The compiler should also be smart about this:
```cpp
ISR(TIMER1_COMPA_vect) { stepper.isr(); }
```
It should recognize that it should not do this…
```
ISR_HANDLER:
  JSR function_stepper_isr
  RET
```
…but that it should do this…
```
ISR_HANDLER:
  JMP function_stepper_isr
```
…and, frankly, it should recognize that:
```
ISR_HANDLER = function_stepper_isr
```

jbrazio, 2016-05-05:
I just don't agree with `this->method()` being less efficient than `method()` when inside the class, the compiler will throw away the `this` reference and do a `JMP func()` in both cases, the advantage of using `this->` is increased readability.

Blue-Marlin, 2016-05-06:
> I'm using pronterface with 250000 baud, changed it to 115200 with no sucess.
I'm not using speed ranges where double stepping would occur in the last days, I went up to 80% of single step mode.

Using Promterface translates to me as using ping-pong, That makes overfilling the RX-buffer unlikely.
No improvement with 115200bd means doublet time between arriving chars. That makes missing to fetch a char from the register unlikely.
Errors on the USB between host and AT32U2 (used as the USB chip) is unlikely because of the USB block checksum.
Errors on the serial line between AT32U2 and AT2560 (hardware problem). That's unlikely to change with release versions. But more likely with increased frequency of high current lines near the serial lines.
RX-buffer corruption. Is unlikely with 8bit buffer indices and interrupt protection. Likelihood increases with the amount of buffer content - unlikely with ping-pong. 'Strange' pointer from somewhere? Wrong array indices in a near by array?

Hard to say what is going on.
Maybe @AnHardt's little input debug helpers can give a hint.
https://github.com/AnHardt/Marlin/pull/32
RX-buffer corruption?

jbrazio, 2016-05-06:
> I'm using pronterface with 250000 baud, changed it to 115200 with no sucess.

This for me somehow clears the ISR delay errors.. otherwise we would need to be really screwing it up to still affect 115.2bps.

Are we positive this is now hardware ? Could it be noise on the serial line ? Noise could increase with the higher stepping on the motors at higher speeds. Do you have a scope ?

Sebastianv650, 2016-05-06:
Sorry, no scope available. I can't imagine it's hardware. I'm in the process of printing 70 identical parts, so all tests are done with the exact same gcode. RC4 runs "smooth" even with advance (maybe 4-5 errors per hour), RCBugFix on the other side is flooding the pronterface status window with error messages and 1-2 times per hour the printer is even doing wrong "print" moves to one edge of the print bed (only limited by software end limits I guess).

> This for me somehow clears the ISR delay errors.. otherwise we would need to be really screwing it up to still affect 115.2bps.

I might be wrong, but do you think so? If we need to transfer 10 commands for a print, say each one takes 0.5s (I know, only for easy calculation) at 115200 baud and 0.25 at 250000 baud and the ISR is running at a fixed rate. Than it should be more likely that a ISR is "hitting" the transfer of the 0.5s long slow 115200 baud than it's hitting a only 0.25s long transfer?
For me as a hobby-programmer I would say higher rate should be better to not influence with stepper ISR.

I will build AnHardt#32 into my code and run some tests with it, but it will take some time to get results as my printer is blocked for the next 9h.

jbrazio, 2016-05-06:
Yes but lowering the baud rate you'd expect to see a difference in the number of errors, if no difference then one of two things could be true:
 - it's not the ISR
 - it's the ISR with a deep screw up

But the ISR changes were minimal.

So last good version was RC4 right ?

Sebastianv650, 2016-05-06:
OK to be honest I can't say for sure if lowering the baud rate made it worse. In both cases, the errors come one after the other. If we want to know that precise, I have to do several runs with both baud rates and count an average value.

Yes I'm seeing errors also in RC4, but without advance maybe 1 in 10h print. Don't nail be down on the number, but it's a rate where I'm not worried about.

thinkyhead, 2016-05-06:
> RCBugFix on the other side is flooding the pronterface status window with error messages

@Sebastianv650 Are you talking about "host keepalive" messages like "busy:processing", or line number / checksum errors?

Sebastianv650, 2016-05-07:
All versions of error messages: Mostly checksum, some line number mismatch, some "no checksum with line number".

Sebastianv650, 2016-05-08:
I did @AnHardt debug changes, here is an example:

SENT: N803 G1 X145.808 Y146.311 E0.14351*107
RECV: ok N802 P0 B2 R128 "N802 G1 X147.472 Y147.976 E0.11133"
SENT: N804 G1 X144.915 Y146.058 E0.15620*108
RECV: ok N803 P0 B2 R128 "N803 G1 X145.808 Y146.311 E0.14351"
SENT: N805 G1 X147.099 Y148.241 E0.19841*98
RECV: ok N804 P0 B2 R128 "N804 G1 X144.915 Y146.058 E0.15620"
SENT: N806 G1 X146.764 Y148.544 E0.20460*98
RECV: ok N805 P0 B2 R128 "N805 G1 X147.099 Y148.241 E0.19841"
SENT: N807 G1 X143.891 Y145.671 E0.26015*107
RECV: Error:checksum mismatch, Last Line: 806 "N803 G1 X145.808 Y146.311 E0.14351"
[ERROR] Error:checksum mismatch, Last Line: 806 "N803 G1 X145.808 Y146.311 E0.14351"

Somebody has an idea? The line looks OK to me..

bobc, 2016-05-08:
Can you post that formatted as code or something? It's too garbled to make sense of.

Sebastianv650, 2016-05-08:
Done, sorry.

thinkyhead, 2016-05-08:
@bobc I have edited the post to make it prettier.

thinkyhead, 2016-05-08:
> The line looks OK to me..

Looks bizarre to me. It's telling us that when it tries to interpret line "806" it is seeing the text of line "803."

Did I already ask which version of Arduino you are using to build with?

Sebastianv650, 2016-05-08:
I'm using 1.6.8.
I did a test, adding customizedSerial.checkRx(); also in the extruder ISR. This reduced the error rate significantly, now it's ~4 for a 6min print, without this second check its >> 12.

AnHardt, 2016-05-09:
Sorry. 
https://github.com/AnHardt/Marlin/pull/32 is debug code that used to work before `serial_line_buffer[]` was introduced in `get_serial_commands()`.  Now when the errors happen the relevant string is still in `serial_line_buffer[]` and not copied to  `command_queue[cmd_queue_index_w]` yet. 
The wrong line is printed.
Sorry.

thinkyhead, 2016-05-09:
@Sebastianv650 Well, that is interesting. I cannot (quite) imagine what would make the stepper isr _significantly_ slower between RC4 and RC6, but it sure seems to be the case. I'll do more research on optimizing C++ code and see if there's a way to keep the encapsulated `Stepper::isr` method but still maintain top performance.

thinkyhead, 2016-05-09:
@AnHardt If the `gcode_line_error` function is altered to take a string pointer, then you can pass `command` to it from within `get_serial_commands` and print that string instead.

Sebastianv650, 2016-05-09:
Thanks for the hints, @AnHardt, @thinkyhead ! I did the modification and it's now printing the corrupted line, here one of the many ones I get:

```
SENT: N967 G1 X134.346 Y136.203 E2.96084*105
RECV: ok N967 P0 B3 R128 "N967 G1 X134.346 Y136.203 E2.96084"
SENT: N968 G1 X134.403 Y135.652 E2.96970*97
RECV: ok N968 P0 B3 R128 "N968 G1 X134.403 Y135.652 E2.96970"
SENT: N969 G1 X139.652 Y140.901 E3.08866*96
RECV: Error:checksum mismatch, Last Line: 968 "N969 G1 X19.652 Y140.901 E3.08866*96"
[ERROR] Error:checksum mismatch, Last Line: 968 "N969 G1 X19.652 Y140.901 E3.08866*96"
```

It's missing single chars in commands, in this example the "3" in the X coordinate. Now makes also sense that my change yesterday (putting customizedSerial.checkRx() also in the stepper ISR) made it less worse: The "input" is checked more frequently and missing chars are not happening that often any more.

But I'm not smart enough to know the real answer: Why it is missing chars when the main stepper ISR takes more time?

Sebastianv650, 2016-05-09:
After thinking about what @AnHardt wrote:

> Missing chars -> an ISR takes too long, or a CRITICAL_SECTION is not closed properly. The serials RX-interrupt is executed too late - the RX-registers content was replaced by the next char. 

and looking how Marlin handles available chars, I would sum up as followed. Please correct me if I'm wrong:
.) At a baud rate of 250000, there is a new char every 40µs. If the char is not read, it gets replaced by the next one sent by the host.
.) There are two ways of storing a new char:
The interrupt triggered when a new char is recieved - but this one has nearly the lowest possible priority available on the Atmega so it may be "postponed" by timer0 interrupts used by the steppers, it get's even worse with the timer1 used by the advance stepper ISR.
That may be the reason why there is another checkpoint inside the stepper ISR, which tries to store a char at every loop. But if this loop takes more than 40µs, the char is gone - my error messages are triggered now.

So the answer why I get much more errors with enabled advance is clear: I'm doing a few more calculations inside stepper ISR. This may take the ISR close to the 40µs.
That there are even more errors with RC6 means that the stepper ISR is a tad slower than in RC4. Maybe not much, but 40µs are not much too..

To proof this, I set the baud rate to 115200 again. I did this already as I saw the errors when I started the advance coding, without success. But I optimized the code since theese day. With 115200 baud the Atmega has 2x the time to get a char from the buffer. With the latest advance code, this solved 100% of the error messages! :-)

This leads two following options:
.) Live with "only" 115200 baud. Was sombody testing if this ever has an impact to printing? Maybe the impact due to the fast char impact at 250000 baud is worse than 115200 baud "slow" transfer rates?
.) Find some ways to improve the stepper ISR so it takes less cycles.
.) Test if it helps when the serial receive thing has it's one ISR, fired at a fixed rate depending on the selected baud rate. This way, no char should be lost. In worst case, this will brake down the rest of the code due to the high check rate, but in this case the current implementation shouldn't be better as than you get the missed chars..

Your opinion?

thinkyhead, 2016-05-09:
I'm considering another option, which is to revert all the Singletons back to (still better-encapsulated) flat C code with `static` linkage. But before I do that, I'm looking to exhaust every other option, for example finding out of the C++ compiler can be "smarter" and produce machine code that is just as efficient as the old code.

thinkyhead, 2016-05-09:
Something worth reading…

http://www.drdobbs.com/implementing-interrupt-service-routines/184401485

andrebstv, 2016-05-09:
It really points to starvation, that is the processor is running out of time and pretty much only working on the ISR. I`m not sure if what is being generated with c++ coding is slower than C, though I must say the probability of that is quite high. 

It does depends on the compiler flags/settings. Note that arduino IDE uses a quite optimized version of these flags. It won`t get (much) better than that.

AnHardt, 2016-05-09:
I love it, when a theory can be confirmed by a test.

250000 <-> 115200 baud. Receiving with 115200bd should be ok. Sending with 115200bd may be a problem. Marlin is busy waiting while writing. (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/MarlinSerial.h#L122  https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/MarlinSerial.h#L152 ++)
Writing for 80 chars means about 3.2ms at 250000bd, more than 6ms at 115200bd. That could make a difference as soon we are not only sending 'ok's but lots of debug output. However the ISRs can interrup the busy waiting.

>  serial receive thing has it's own ISR
It already has. But no (software) interrupt can interrupt an other interrupt. The interrupt priority only schedules, who is the next, if more than one interrupt request is in the register, after the current interrupt has finished.

thinkyhead, 2016-05-10:
> I love it, when a theory can be confirmed by a test.

Thank you testers, feedbackers! Without you we who know code well, but not the platform or the compiler, often feel a little lost in the dark. But through this kind of experience, much is learned. Deep gratitude to all of you!

Roxy-3DPrintBoard, 2016-05-10:
>  Sending with 115200bd may be a problem. Marlin is busy waiting while writing. 

This is kind of a brute force way to work around it...  But this would work for either 115,000 or 250000 baud rate.   And there isn't much penalty because we are busy waiting anyway.   As long as we are spinning waiting to see if we can transmit the character, we probably should be checking to see if a character came in the receive side of the UART.   We would just add a little bit of jitter to when we return:
~~~~
    FORCE_INLINE void write(uint8_t c) {
      while (!TEST(M_UCSRxA, M_UDREx))
        checkRx();
      M_UDRx = c;
    }
~~~~

andrebstv, 2016-05-10:
Or you could put transmition on a TxInterrupt routine....

thinkyhead, 2016-05-10:
"transmition" ?

Transmission? Like on a car?

Roxy-3DPrintBoard, 2016-05-10:
> Or you could put transmition a Tx-Interrupt routine....

Yes.   And if that happened, it might make sense to check the Rx status while in the Interrupt handler.  Adding a few lines of code to it:
~~~~
      if (TEST(M_UCSRxA, M_RXCx)) 
          checkRx();
~~~~

jbrazio, 2016-05-14:
I've always printed at 250kbps, I can confirm that with RC6 I'm also seeing random checksum errors.

```
Resend: 9399
Error:checksum mismatch, Last Line: 11315
[ERROR] Error:checksum mismatch, Last Line: 11315

Resend: 11316
Error:checksum mismatch, Last Line: 12547
[ERROR] Error:checksum mismatch, Last Line: 12547

Resend: 12548
Error:checksum mismatch, Last Line: 12580
[ERROR] Error:checksum mismatch, Last Line: 12580

Resend: 12581
Error:checksum mismatch, Last Line: 13270
[ERROR] Error:checksum mismatch, Last Line: 13270

Resend: 13271
Error:checksum mismatch, Last Line: 14503
[ERROR] Error:checksum mismatch, Last Line: 14503
```

What is the current status of this thread ?

Roxy-3DPrintBoard, 2016-05-14:
> I've always printed at 250kbps, I can confirm that with RC6 I'm also seeing random checksum errors.

I think AnHardt already did this, but it would be good to print out the line that has a checksum error.  Most likely, some characters got lost.    If that is the case, we would at least know that we need to do a better job keeping up with the incoming character stream.

jbrazio, 2016-05-14:
OK I've been doing some testing.. 
- Prointerface: Random checksum errors during a 5m print but prints successfully
- Repetier Host (Pin-ping option enabled): Random checksum errors during a 5m print but prints successfully
- Repetier Host (Pin-ping option disabled): Printing stutters, unable to finish printing due to **massive** checksum errors

All tests have been done with `DISABLE_HOST_KEEPALIVE`.

Blue-Marlin, 2016-05-16:
@jbrazio 
Could you please try
* Repetier Host (Pin-ping option disabled) Buffersize decreased to 50% (64)

Sebastianv650, 2016-05-16:
I bet you can also see missing chars with AnHardts mods..
By the way, I did some time measurements that proofes my assumption about the runtime of the main stepper ISR:

ISR runtime with enabled Advance, but K=0: 81µs
ISR runtime with enabled Advance with K<>0: 95µs
So it's clear it can't take 250000 baud with the current way of handling incomming chars..

But a calculation of a new segment (add it to the buffer and recalc all the existing segments) takes **3.15ms**, so I would say it different: Running the serial connection at 250000 baud is useless (except for sending things) as the FW can't calculate all the new commands in time:
If you asume 30chars per command, you can send about 384 commands per second at 115200 baud. But with a planner needing 3.15ms per new line, you can handle only up to 317 lines (commands) per second. If we take into account that the FW has more things to do than only recalculating incomming line segments, figures get even worse..

I think there is a lot of potential for improvements in the planner calculation speed, I will have a look at them in the next days.

Blue-Marlin, 2016-05-16:
Ahh. Numbers. :-)
How deep is your planner buffer?

jbrazio, 2016-05-16:
> Could you please try
Repetier Host (Pin-ping option disabled) Buffersize decreased to 50% (64)

I'm currently SD printing, this evening I'll try and report back to you.

> If you asume 30chars per command, you can send about 384 commands per second at 115200 baud. But with a planner needing 3.15ms per new line, you can handle only up to 317 lines (commands) per second

This is valid if you assume a sync state, that's why we have buffers.

Sebastianv650, 2016-05-16:
I have not changed the buffers size, so it should be whatever it is inside the default RCBugFix code.
@jbrazio , thats true, it's a worst case calculation. But it's true even with buffer: If the buffer is full, you need no baud rate at all (no space to calculate new lines). If it is empty, more than 317 lines can't be handled / s, so 115200 is the max. usable baud rate. If the code could get all chars at 250000 baud, of course I would see no disadvantage of using 250000!

thinkyhead, 2016-05-16:
The ultimate fix (or at least improvement) for comm errors is going to be re-writing Stepper (and Planner, Temperature, Endstops) to use `static` data and methods, because currently all these non-static method calls have the additional parameter `this`, and elements of the stepper ISR are slowed just enough to make it run too long. Any other clever optimizations we can think of will also help!

jbrazio, 2016-05-16:
Or the AVR gcc behaves like a retard or I'm not understanding why "this->" is a hogger when compared with a static var.

Static keyword hints the compiler that a variable is a class variable thus shared among all instances of a class, in the other hand a "standard" class var is only accessible by the instance itself (or other instances of the same class but that's another story).

"this" is nothing more than a pointer to the class instance.

AVR-gcc should optimize and do a direct address "jump" either is accessing a static var or a var being called like "this->foo".

Blue-Marlin, 2016-05-16:
If you want to investigate the results in depth - https://github.com/MarlinFirmware/Marlin/issues/3212

andrebstv, 2016-05-17:
I do believe static vars will be otimized away in interrupts. Unless they
are declared as volatile

thinkyhead, 2016-05-17:
@jbrazio The C++ compiler has no clue that when calling stepper.my_instance_method() that there is only ever going to be a single instance of that class. So, there is always a hidden `this` variable passed into the method, on the stack, as the first argument. This is just the normal calling convention for instance methods. To get rid of the `this` hidden argument, you use `static`, and the method is now a class method. And of course you have no `this`, but you can synthesize a `me` variable, for example, (a fairly common practice with singletons) to address the instance.

Sorry if this is all just a C++ review.

When it comes to the ISR, the ISR itself is (normally) a global static C-style function. For the Stepper ISR it currently calls `stepper.isr()`. This passes `&stepper` on the stack to the `Stepper::isr()` method, which receives it as the "`this`" variable. (`Stepper::isr` is not a `FORCE_INLINE`, but I suppose it could be, by moving the method body into the `stepper.h` header. That will likely save the one call to `stepper.isr()` while also not passing `stepper` as `this` on the stack, but it doesn't do anything for calls to the other singletons, all of whose _instance methods_ are taking that extra `this` parameter.

To get leaner C-function-style calling conventions and save over the instance method calls, the only option is to declare class data and methods as `static`. They can still be `volatile` or `inline` or even `FORCE_INLINE` with the `static` keyword.

Portions of a class that aren't performance-sensitive can remain instance data and methods, if that makes sense, but most of the class needs to be "flattened" into a single static "class instance" to get maximum performance via this compiler.

I have tried the `-fwhole-program` flag too, which can do extra magic static optimizations, but I have not been able to successfully compile with it. I was curious to see if it would affect performance, if it automated some of this stuff. But it looks like I have to do more "by hand" and sort out the compile errors.

jbrazio, 2016-05-19:
So I followed the `ASM` path and learned new things.

The test subject.
```cpp
class aClass {
  public:
    char i;
    static char j;

  void foo() {
    this->i = 5;
  }

  static void bar() {
    aClass::j = 10;
  }
};

int main (void)
{
  aClass a;
  a.foo();
  a.bar();
}
```

Disassembly code for `a.foo();`.
```asm
void aClass::foo() { ; 34 clk, 2.125 uS
 132: cf 93         push  r28              ; 2 clk
 134: df 93         push  r29              ; 2 clk
 136: 1f 92         push  r1               ; 2 clk
 138: 1f 92         push  r1               ; 2 clk
 13a: cd b7         in  r28, 0x3d ; 61     ; 1 clk
 13c: de b7         in  r29, 0x3e ; 62     ; 1 clk
 13e: 9a 83         std Y+2, r25  ; 0x02   ; 2 clk
 140: 89 83         std Y+1, r24  ; 0x01   ; 2 clk
  this->i = 5;
 142: 89 81         ldd r24, Y+1  ; 0x01   ; 2 clk
 144: 9a 81         ldd r25, Y+2  ; 0x02   ; 2 clk
 146: 25 e0         ldi r18, 0x05 ; 5      ; 1 clk
 148: fc 01         movw  r30, r24         ; 1 clk
 14a: 20 83         st  Z, r18             ; 2 clk
}
 14c: 0f 90         pop r0                 ; 2 clk
 14e: 0f 90         pop r0                 ; 2 clk
 150: df 91         pop r29                ; 2 clk
 152: cf 91         pop r28                ; 2 clk
 154: 08 95         ret                    ; 4 clk
```

Disassembly code for `a.bar();`.
```asm
void aClass::bar() { ; 17 clk, 1.062 uS
 156: cf 93         push  r28             ; 2 clk
 158: df 93         push  r29             ; 2 clk
 15a: cd b7         in  r28, 0x3d ; 61    ; 1 clk
 15c: de b7         in  r29, 0x3e ; 62    ; 1 clk
  aClass::j = 10;
 15e: 8a e0         ldi r24, 0x0A ; 10    ; 1 clk
 160: 80 93 00 02   sts 0x0200, r24       ; 2 clk
}
 164: df 91         pop r29               ; 2 clk
 166: cf 91         pop r28               ; 2 clk
 168: 08 95         ret                   ; 4 clk
```

**Conclusion: to update a class variable takes 2.125 uS (34 clock cycles), to update a static class variable takes 1.062 uS (17 clock cycles) which is exactly double the time.**

--

The test subject.
```cpp
class aClass {
  public:
    char i;

  void foo() {
    this->i = 5;
  }

  void baz() {
    i = 15;
  }
};

int main (void)
{
  aClass a;
  a.foo();
  a.baz();
}
```
Disassembly code for `a.foo();`.
```asm
void aClass::foo() {
 12c:	cf 93       	push	r28
 12e:	df 93       	push	r29
 130:	1f 92       	push	r1
 132:	1f 92       	push	r1
 134:	cd b7       	in	r28, 0x3d	; 61
 136:	de b7       	in	r29, 0x3e	; 62
 138:	9a 83       	std	Y+2, r25	; 0x02
 13a:	89 83       	std	Y+1, r24	; 0x01
  this->i = 5;
 13c:	89 81       	ldd	r24, Y+1	; 0x01
 13e:	9a 81       	ldd	r25, Y+2	; 0x02
 140:	25 e0       	ldi	r18, 0x05	; 5
 142:	fc 01       	movw	r30, r24
 144:	20 83       	st	Z, r18
}
 146:	0f 90       	pop	r0
 148:	0f 90       	pop	r0
 14a:	df 91       	pop	r29
 14c:	cf 91       	pop	r28
 14e:	08 95       	ret
```

Disassembly code for `a.baz();`.
```cpp
void aClass::baz() {
 150:	cf 93       	push	r28
 152:	df 93       	push	r29
 154:	1f 92       	push	r1
 156:	1f 92       	push	r1
 158:	cd b7       	in	r28, 0x3d	; 61
 15a:	de b7       	in	r29, 0x3e	; 62
 15c:	9a 83       	std	Y+2, r25	; 0x02
 15e:	89 83       	std	Y+1, r24	; 0x01
  i = 15;
 160:	89 81       	ldd	r24, Y+1	; 0x01
 162:	9a 81       	ldd	r25, Y+2	; 0x02
 164:	2f e0       	ldi	r18, 0x0F	; 15
 166:	fc 01       	movw	r30, r24
 168:	20 83       	st	Z, r18
}
 16a:	0f 90       	pop	r0
 16c:	0f 90       	pop	r0
 16e:	df 91       	pop	r29
 170:	cf 91       	pop	r28
 172:	08 95       	ret
```

**Conclusion: using `this` to increase the overall class readeability is irrelevant from a performance perspective because the compiler create exactly the same ASM code.**

thinkyhead, 2016-05-19:
Yep, just as expected. Using `this->` inside of _class methods_ is perfectly harmless, but we take a hit by calling _instance methods_ from anywhere else. So, `static` methods are going to be our friend, especially where we're currently calling _instance methods_ from within ISRs.

andrebstv, 2016-05-19:
Could you please tell how you managed to disassemble with the C++
references?
Good job btw

jbrazio, 2016-05-19:
Where would you use `this` without being inside the instance ?..

You know to use static class variables is basically the same thing as using global vars ? (Rhetoric question I know you know)

To put things into perspective if inside a ISR we update four class non static vars we are still inside the **resolution margin** of the `micros()` function on a 16Mhz board.

So.. there **are** some places and **some** very special objects where static var class **maybe** make sense do we can squeeze **one** micro seconds.

--

The dump was done with `avr-objdump -F -S`, when the binary is unstripped and you do it inside the source code folder it seems to inline the thing.

thinkyhead, 2016-05-19:
> Where would you use this without being inside the instance?

I never intended to bring up "`this`" as a topic. It's fine.

thinkyhead, 2016-05-19:
> some places and some very special objects where static var

Improving performance by replacing **instance methods** with **`static` methods** is the main task at hand.

thinkyhead, 2016-05-19:
I just want to make sure that especially the stepper ISR is as fast as it can possibly be. But still keep the class structure, namespace, and access privileges.

CONSULitAS, 2016-05-19:
@jbrazio 
> You know to use static class variables is basically the same thing as using global vars ? (Rhetoric question I know you know)

In ASM yes, but the main advantage over global variables is encapsulation and the namespace context! So the class structure is indeed the right way...

CONSULitAS, 2016-05-19:
@jbrazio 
BTW: Great analysis. So now we know what we speak about!

jbrazio, 2016-05-19:
> Improving performance by replacing instance methods with static class methods is the main task at hand.

Let's measure the impact then.. I will reuse the previous assembly code because we can perfectly see the callings, by starting with the biggest asm block which is the code for `a.foo();`, the *non-static method* we do the diff to the `a.bar();` asm block, the *static method*, the ending result is the total overhead.

```asm
void aClass::foo() {
 (...)
 136: 1f 92         push  r1               ; 2 clk
 138: 1f 92         push  r1               ; 2 clk
 (...)
 13e: 9a 83         std Y+2, r25  ; 0x02   ; 2 clk
 140: 89 83         std Y+1, r24  ; 0x01   ; 2 clk
 [ignoring here the specific var type IO]
}
 (...)
```

This leaves us with an overhead of **8 clock cycles** meaning **0.5 uS** (yes.. **half a micro second**).

jbrazio, 2016-05-19:
Some empirical data..

It's infeasible to try to measure the parts of the code we have been discussing [method calls/overhead], so what I have done is to measure the execution time of a whole method for a static and a non-static. This will not give us real numbers but at least will allows use to see the difference between one call type and another in a **comparison**. It's also important to note that with Arduino the minimum time difference between two `micro()` is *4 microseconds*, even if less time has passed between the them, trying to even this out.. I call each method 10000 so I can calculate an average.

The text subject.
```cpp
class aClass {
  public:

  void foo() {
    return;
  }

  static void bar() {
    return;
  }
};

void setup() {
  Serial.begin(115200);

  aClass a;
  uint32_t s, e, j;

  const int32_t cycles = 10000;

  for (uint8_t i = 0; i < 5; i++) {
    s = micros();

    for (j = 0; j < cycles; j++) {
      a.foo();
    }

    e = micros();
    Serial.print("a.foo(): ");
    Serial.print((float) (e-s) /j);
    Serial.println("uS");

    s = micros();

    for (j = 0; j < cycles; j++) {
      a.bar();
    }

    e = micros();
    Serial.print("a.bar(): ");
    Serial.print((float) (e-s) /j);
    Serial.println("uS");

    Serial.println();
  }
}

void loop() {}
```

Which outputs:
```
Welcome to minicom 2.7

OPTIONS: I18n 
Compiled on Feb  7 2016, 13:37:27.
Port /dev/ttyACM0, 17:36:00

Press CTRL-A Z for help on special keys

a.foo(): 4.42uS
a.bar(): 3.48uS

a.foo(): 4.43uS
a.bar(): 3.48uS

a.foo(): 4.43uS
a.bar(): 3.48uS

a.foo(): 4.43uS
a.bar(): 3.48uS

a.foo(): 4.43uS
a.bar(): 3.48uS
```

**Conclusion: the empirical difference per method call is ~0.95 uS**

ruggb, 2016-05-22:
I had a similar checksum error @ 60mm/sec and above. @ 50mm/sec is was almost gone. @ 40mm/sec it was under control. changing the baud rate made no diff.

Grogyan, 2016-06-24:
I want to put my 2 cents in, with what I had suspected, confirmed by someone else with regards to the random over temp problem I had been seeing with using the MAX31855 and MAX6675 thermocouples.

#https://github.com/Smoothieware/Smoothieware/issues/894

Problem could quite possibly be down to the high EMI from the stepper motors impacting the SPI lines.

Which would be nearly impossible to fix in firmware, as it is a hardware grounding/shielding issue.

Sebastianv650, 2016-06-24:
@Grogyan this problem here is definitly not EMI, it's

thinkyhead, 2016-06-24:
@Sebastianv650 I wonder if inserting the following into the `advance_isr` steps loop would make serial connection more reliable… or if it would only slow other things…
```cpp
#ifndef USBCON
  customizedSerial.checkRx(); // Check for serial chars.
#endif
```

[51 more replies]

closed 2016-08-11 by thinkyhead