MarlinFirmware/Marlin #3683: Marlin 1.1.0-RC4 in Printrboard rev D

Joda89, 2016-05-05:
I would like to update the firmware of my Printrboard Rev d so I downloaded the version marlin rc4 then I compiled with Arduino 1.0.5.

When compiling I have no error and the bin is 58 KB.
when I upload the bin in my card I have no problem.
when I restart the card I have nothing in the console.
as if the setup method was not called.

  Can you help me ?

thinkyhead, 2016-05-05:
First, switch to `RCBugFix` and work with that. RC4 is pretty old now, and we've fixed some things.

If you still have trouble, then you should post your `Configuration.h` for examination.

Joda89, 2016-05-05:
Hello ,
thank you for your quick reply.
the RCBugFix not run with 1.0.5 arduino.
so I download arduino 1.6.8 + teensyduino 
build ok but impossible to upload the card .

"Opening Teensy Loader...

Le croquis utilise 53 820 octets (41%) de l'espace de stockage de programmes. Le maximum est de 130 048 octets.
Les variables globales utilisent 2 850 octets (34%) de mémoire dynamique, ce qui laisse 5 342 octets pour les variables locales. Le maximum est de 8 192 octets.
Teensy did not respond to a USB-based request to automatically reboot.
Please press the PROGRAM MODE BUTTON on your Teensy to upload your sketch."

do you have an idea ?

tiagom62, 2016-05-05:
Doesn't sound like you have the board correctly set to upload firmware. Take a look here http://reprap.org/wiki/Printrboard

Roxy-3DPrintBoard, 2016-05-06:
I have a PrintRboard.  On that board you have to short a special pin when you do a reset to get it into the programming mode.   That is kind of what it sounds like you need to do on the Teensy.   But I'm just guessing.

closed 2016-05-12 by jbrazio