MarlinFirmware/Marlin #3684: [RC3] X axis inverted.

CoderSquirrel, 2016-05-05:
Hi i'm using the marlin RC 3, and i realizes that the x is inverted, like is mirrored, does it was a bug that was fixed or may be any trouble here?
I used Cura as slicer, my `INVERT_X_DIR` is false and `X_HOME_DIR -1`.
I've tried to change its value and nothing happend. At the slicer the picure is ok, but when print is mirrored.
Any suggestions?

jbrazio, 2016-05-05:
You mean the part when printed is mirrored ?

CoderSquirrel, 2016-05-05:
@jbrazio i'm trying to get a good print to show u, but to be more precise, it looks like the x is rotated.

CoderSquirrel, 2016-05-05:
OMG, i was so dump. X rotate is actually the Y inverted. LOL
So i had changed Y inverted and inverted the home dir, so whats happend was that the bed is now ignoring my end stop.
[image]

jbrazio, 2016-05-05:
`INVERT_Y_DIR` is your friend here.

CoderSquirrel, 2016-05-05:
@jbrazio if i used the `INVERT_Y_DIR` and...
My printer endstop(in front) min, it is being triggered when i report it via `M119`.
After i changed`INVERT_Y_DIR` to true, i keep  `HOME_DIR` as -1, once i knew my endstop was reported as min. Result: my printer "home" was in fact going to the back, the max, so i changed my `HOME_DIR` to 1, cuz if the -1 that supposed to be min and was being max, the 1 that supposed to be max would be min(?), but this time its behavior was normal, going to back that was 1, max.
I'm pretty confused now :S

jbrazio, 2016-05-05:
I'm also confused.. lol
So what are the settings which worked for you ?

CoderSquirrel, 2016-05-06:
@jbrazio without the `INVERT_Y_DIR` setted as true

jbrazio, 2016-05-06:
Well.. my Prusa i3 has the following:

```
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1
```

CoderSquirrel, 2016-05-06:
No, i mean, i never used `#define INVERT_Y_DIR true`, i'm only using now because i sundelly realize that the Y is invented, i was used to print simetric stuff so i never print anything that realise that it was inverted, just today. And to be fairly i almost not see it printing, i'm intern and in my intern i make some custom modifications at marlin, and i basically see the mods i do, only recentely i'm seeing the printer full routin

Blue-Marlin, 2016-05-06:
If you have to set `INVERT_?_DIR` is a matter about how you plugged your stepper.
If you have to set ?_HOME_DIR to +1 or -1 is a matter of if your are homing towards a min or max -endstop/homing -switch.
If you change the `INVERT_?_DIR` a min switch converts to a max switch.
You always want to have a right handed coordinate system - else you are printing mirrored.

CoderSquirrel, 2016-05-06:
@Blue-Marlin  u said:
> If you change the INVERT_?_DIR a min switch converts to a max switch.

So, my endstop that is min when my `#define INVERT_Y_DIR false` would became max right?
Well, i'm afraid that there might be a problem, because when i run M119 with `#define INVERT_Y_DIR true` and `#define Y_HOME_DIR 1` it keeps saying 

> < y_min: TRIGGERED
> < y_max: open

My bed is going to the max, but the endstop is not being triggered as max, instead it has been triggered as min.

CoderSquirrel, 2016-05-06:
I've tried to "debug" the code.
The line where the ` Y_HOME_DIR` is analized to change the min and max is:
```
#ifdef Y_STOP_PIN
 SERIAL_PROTOCOLLN("------ Definido Y Stop");
  #if Y_HOME_DIR < 0
    #define Y_MIN_PIN Y_STOP_PIN
    #define Y_MAX_PIN -1
  #else
    SERIAL_PROTOCOL("------ max pin ");
    SERIAL_PROTOCOLLN(Y_STOP_PIN);
    #define Y_MIN_PIN -1
    #define Y_MAX_PIN Y_STOP_PIN
  #endif
#endif
```

at `pins.h` but my serial_protocolln never appear at the serial monitor, maybe this is a part of code that is no being reach?

thinkyhead, 2016-05-06:
Unplug the endstop, plug it in one over. Max endstops must be connected to the next set of pins on your board. On a Cartesian machine with coordinates that increase to the right, the max endstop switch will be mounted on the right end of the X axis, and attached to the XMAX pins on your board. On a Cartesian machine the Y axis coordinates should increase towards the back, and a Y max endstop will be mounted on the front of the machine, plugged into the YMAX pins on the board.

CoderSquirrel, 2016-05-06:
@thinkyhead thanks, i inverted the pins 14 and 15 in `Y_MIN_PIN` and `Y_MAX_PIN` and it's working, i'm just wondering if, there is an definition for invertion, this wouldn't supposed to make this invertion, like a said ealier, in here
```
#ifdef Y_STOP_PIN
 SERIAL_PROTOCOLLN("------ Definido Y Stop");
  #if Y_HOME_DIR < 0
    #define Y_MIN_PIN Y_STOP_PIN
    #define Y_MAX_PIN -1
  #else
    SERIAL_PROTOCOL("------ max pin ");
    SERIAL_PROTOCOLLN(Y_STOP_PIN);
    #define Y_MIN_PIN -1
    #define Y_MAX_PIN Y_STOP_PIN
  #endif
#endif
```

?

thinkyhead, 2016-05-06:
No. Sorry, but that's a terrible idea.

thinkyhead, 2016-05-06:
Those are only for cases, as I understand it, where a board only has 3 endstop plugs instead of 6.

bobc, 2016-05-07:
It's actually pretty impossible for the firmware to flip an axis like that.

In all cases of flipped prints that I have seen, it is due to a misconfiguration. It usually happens with a moving bed, the endstop is placed at the wrong end and set to a min endstop.

The fix is very simple, move the endstop to the other end of the axis, or configure as a max endstop. Often the hard part is convincing people their config is wrong...

CoderSquirrel, 2016-05-07:
@thinkyhead is a cartesian model, printerbot like.
Yes in my case i made a change in the `Y_MIN_PIN` and` Y_MAX_PIN` that are in the `pins_RAMPS_13.h` to change the min and max, and inverting them, what was 14 i put 15, and what was 15 i put 14, and it was solved.
But considering what @Blue-Marlin  said about:

> If you change the INVERT_?_DIR a min switch converts to a max switch.
 
it should automatically convert(invert/change) this pins one in another's place?

Blue-Marlin, 2016-05-07:
@CoderSquirrel 
> If you change the INVERT_?_DIR a min switch converts to a max switch.

Purely logically - when it does not change it's position.

What you have to do, is to put the new fact into the config, or connect it to the other pin.

thinkyhead, 2016-05-08:
> Purely logically - when it does not change its position.

Let me clarify.

On a board that has only 3 endstop plugs, the `X_STOP_PIN` pin will be assigned to `X_MIN_PIN` when a min endstop is configured, and it will be assigned to `X_MAX_PIN` for a max endstop.

On a board with 6 endstop plugs, you must physically connect the min endstop to a min pin and a max endstop to a max pin.

Blue-Marlin, 2016-05-08:
Lets suppose a working system - working with a left handed coordinate system - so all parts are mirrored.
The small values of an axis are on the left side - the large values are on the right side. 
The min-endstop is on the left side - the max-endstop on the right.

Now we try to make the coordinate system right handed by reversing an axis by setting `INVERT_?_DIR false` to `INVERT_?_DIR true`.
Now the small values are on the right and the large values are on the left.
Now the former min-endstop on the left side is hit when the values grow - so it is now a max-endstop.
Now the former max-endstop on the right side is hit when the values shrink - so it is now a min-endstop.
That is the purely logical conversion.

Usually we have only one endstop for an axis.
We could mount a the right endstop to the left and the left to the right.
Or we could swap the cables, if we have 6 endstop pins.
Or we could swap the numbers in the pin-file for the max/min-pins, if we have 6 endstop pins.
Or if we have only 3 endstop pins, we have to tell the config, that now the other type of endstop is connected.

Additionally we have to think about the homing direction.

thinkyhead, 2016-05-09:
Should Marlin support coordinates that increase from right-to-left? I believe that all slicers assume the front-left corner (of the object on the plate) has smaller XY coordinates than the back-right corner (of the object on the plate). I have no problem with rotating the coordinate system, but I don't see any reason to support a mirrored-only X or Y axis.

Blue-Marlin, 2016-05-09:
> Should Marlin support coordinates that increase from right-to-left?

It already does. Just look from the backside of your printer.
No. We should not support left handed coordinate systems.
Exactly _because_ we do not support left handed systems an axis has to be reversed here to make it right handed.

The main source of errors when setting up the machine seems to be a wrong definition of the coordinates in the uses heads, as soon as the bed is moving and not the nozzle.
The user has to see the difference from [0,0,0] to the nozzle - not the direction the bed is moving.

Roxy-3DPrintBoard, 2016-05-09:
@JBrazio 

> The main source of errors when setting up the machine seems to be a wrong definition of the coordinates in the uses heads, as soon as the bed is moving and not the nozzle.  The user has to see the difference from [0,0,0] to the nozzle - not the direction the bed is moving.

Valuable tip for better documentation!   (I know from personal experience this comment is true!!!)

jbrazio, 2016-05-09:
Note taken.

closed 2016-05-15 by jbrazio