MarlinFirmware/Marlin #3688: Babystepper Axis Issue

daedalus14850, 2016-05-06:
Hello all, novice here, I'd appreciate your help in resolving an issue.

I've set up babystepper for my 3d printer and have hooked up a dedicated lcd to it. Currently, when I start the print and move the dial, the extruder only steps up (not down), and only does so when the printer is printing in the negative y direction.

I have played around with making the multiplicator a negative value, but it didn't seem to help. Regardless of whether I turn the dial left or right (i.e. positive or negative), it makes no difference. The extruder only steps (i.e. moves) up. Never down (which is what I primarily want it to do). And, like I said before, it only does this when it is moving in the -y direction. +y, +x, and -x do not register any input from the dial. Any help will be greatly appreciated. Thank you!

thinkyhead, 2016-05-06:
What version of Marlin are you using?

daedalus14850, 2016-05-06:
We Are currently using a modified version of Marlin_RAMPS-Epcos_i38 with Arduino 1.6.7.

thinkyhead, 2016-05-06:
Is there something in your modified Marlin that you cannot do without, and which is not in the main branch? We've advanced the firmware a lot, and chances are issues in a custom fork are long fixed in the main one. i would recommend you upgrade to the latest code, rather than spend too much time beating that horse,

closed 2016-05-30 by jbrazio