MarlinFirmware/Marlin #3690: G3 negative I problem

mirecta, 2016-05-06:
Hi i build laser engraver ramps + marlin everything works but i engraving something from inkscape with plugin which use G2 G3 commands even with negative I or J 

if i send to machine 
G2 X100 Y100 I1.2208 J16.3622 - machine do it (CW)
G3 X100 Y100 I1.2208 J16.3622 - OK too (CCW)
and 
G2 X100 Y100 I-1.2208 J16.3622 - machine do it with CCW arc
but 
G3 X100 Y100 I-1.2208 J16.3622 - machine do nothing !!!!

jbrazio, 2016-05-06:
@mirecta you will have to contact directly the maintainer of the Marlin fork you're using, this is the repo for the official Marlin which currently does not support laser engraving.

mirecta, 2016-05-06:
i use this marlin and patch it for laser , i add three new commands M160 M161 M162 but i didnt touch anything other so i use master from this repo

mirecta, 2016-05-06:
so u can fix this

mirecta, 2016-05-06:
G3 command is indipendent to tool which i use (3d printer, mill , laser etc)

jbrazio, 2016-05-06:
Indeed, but we don't know if the bug is coming for the additional code base on top of Marlin.
We have no reports of CCW arc being broken.

Which is the base version of Marlin you're using ?

mirecta, 2016-05-06:
my revision f0203728f30d7809dfc74c880f1eea2df87be4d1
Fri Apr 15 16:15:43 2016 -0700

i wrote tat G3 works but only with positive parameters I and J

jbrazio, 2016-05-06:
After that ref id two PR were merged regarding arcs: #3502 #3514.
I would suggest to test the latest RCBugfix branch.

thinkyhead, 2016-05-06:
I recently worked on the arc code, and found it to function really well on my Cartesian machine. I haven't tested it on COREXY or DELTA, so if you have one of those, and you still have a problem, I would value your feedback on the current RCBugFix. This is one feature that should "just work."

closed 2016-05-10 by jbrazio