MarlinFirmware/Marlin #3691: Irregular movement on a delta with 'normal'  DELTA_SEGMENTS_PER_SECOND

nomukaiki, 2016-05-06:
I couldn't find a similar issue so here it goes.

I am using the version from RCBugFix branch.

From previous released versions I had `#define DELTA_SEGMENTS_PER_SECOND 160` and it was working very well. 
With smooth movements during print or travel at any slow of very fast speeds.

When I started using the RC versions I noticed that, specially and maybe only the 1st layer of printing, the head moves very irregularly. Almost like if the nozzle is getting friction on the bed (but it doesn't).
This is happening at any speed, on a AtMega2560.

I even replaced the stepper drivers because I thought they were the problem. They are running at 1/16.

I then tested reducing the delta segments and it is now working smoothly again.
`#define DELTA_SEGMENTS_PER_SECOND 80`

This is probably irrelevant when we can reduce this setting, but maybe the new version should prevent or warn that this might be required?

Does anyone have the same behaviour?

[image]

clefranc, 2016-05-07:
@nomukaiki Are you using a graphical LCD?

https://github.com/MarlinFirmware/Marlin/issues/1404

thinkyhead, 2016-05-07:
We haven't really done performance testing between versions to see how many cycles various functions are eating up. Something we should figure out a way to do soon.

clefranc, 2016-05-07:
@thinkyhead A workaround here: https://github.com/clefranc/Marlin4Delta/pull/7/commits/6ae457b74b2c20f97a03ee8d6f856e2802d83ced

The idea is to prevent the menu system to automatically go back to the status screen, This is a temporary solution, until a complete rewrite of the delta kinematics (lot of works) is done.

thinkyhead, 2016-05-07:
Rewrite of kinematics would be cool. The need for `sqrt` and floating point maths is the biggest constraint. Maybe there is some kind of clever way to get around it, using approximations and smoothing, or pre-generating the `sqrt` values and storing in RAM on boards with the extra space. I will look into this kind of optimization via The Prestigious Internet.

thinkyhead, 2016-05-07:
There may also be ways to save time in the LCD update, perhaps, by staggering the screen updates so only half is done, then there is a tiny delay, and then the other half is done. Currently both halves are updated concurrently.

thinkyhead, 2016-05-07:
The same trick could be used to help optimize the delta calculations, by only doing one `sqrt` calculation per loop instead of all three concurrently. This alone could restore performance - and still maintain accuracy if done with some careful forethought.

nomukaiki, 2016-05-07:
@clefranc thank you for pointing out the LCD ticket and your code.
I will disable my LCD today and see if that makes any difference.
If it does I will definitely bring your branch into my forked version.

nomukaiki, 2016-05-07:
Some more insights about my tests today.

Because I use the LCD always on the status screen, the solution suggested by @clefranc didn't help me.
What I did instead was reducing the LCD refresh rate in general. That value is by default 10 times per second. https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.h#L62
Is this high refresh rate really necessary? 
I changed mine to be 2 times per second (every 500ms) and is more than enough.
This definitely removed completely that irregular movement.
https://github.com/nomukaiki/Marlin/commit/af00786d9b39ca87b06e50015561bc8c03b7522a#diff-f7f60d8a724b8fc428f6990f3ff78245

Thanks

clefranc, 2016-05-07:
@nomukaiki What's your print speed (mm/sec)?

nomukaiki, 2016-05-07:
I tested with slow speeds (10 mm/s) and fast (60 mm/s) and it is the same.
What is more strange is that it happens only during the 1st layer.

Could it be the additional calculations of the bed levelling?

Roxy-3DPrintBoard, 2016-05-07:
> Could it be the additional calculations of the bed levelling?

If auto bed leveling is enabled and built into the firmware, it will do the same calculations whether you do a G29 or not.   If you don't do a G29, you will be multiplying by a unity matrix instead of one that corrects for a slanted bed.

Building your firmware without Auto Bed Leveling is going to eliminate some floating point math and speed things up a little bit.

Blue-Marlin, 2016-05-07:
So the question is.
What is different in the first layer?
Other temperature? Other extrusion rate (fatter/thinner lines)? Other ... ?
What we see in the picture, is that the outer perimeter or a brim?

nomukaiki, 2016-05-08:
The print in the picture had the following settings:
- 1st layer at 60% speed (rest of the layers at 30mm/s, so this one is 18mm/s)
- 110% height and 110% extrusion (this is a 0.8 nozzle at 0.32 layer height)
- 2 outer perimeters and 100% infill
- the same temperature for all print (110C bed and 230C extruder)

I disabled the ABL and without it the print is again perfect at 160 segments.
My ABL was also giving more uneven 1st layer than without it. Maybe is related?

clefranc, 2016-05-08:
@nomukaiki 30mm/sec is considered slow for a delta, mine is running around **80mm/sec** (200 for non-print moves), **120** segment per second (sps), at these speeds you need to remove all graphical LCD status update, slowing it's refresh rate is not enough. Unlike cartisian ones, using Marlin for deltas are full of compromise. I even came up to remove calls in code and increase the optimization level of the compiler.

nomukaiki, 2016-05-08:
Thank you @clefranc.
I will definitely follow your guidelines and start cutting out unnecessary features from the firmware to get the best out of my delta.

What I see as a conclusion of this issue is that the new 1.1.0 should:
- Document better the constraints of speed for a delta, maybe next to the `DELTA_SEGMENTS_PER_SECOND `?
- Having its default setting not 160 but a bit lower? I found 80 more than enough for printing circles without seeing any corners with 0.4 nozzle.
- Bring the LCD status refresh or enabling it at all during print to a configuration level?

Maybe after the 1.1.0 a rewrite of kinematics suggested by @thinkyhead can be done?

nomukaiki, 2016-05-08:
I've found a pattern that is easy to reproduce.
It happens when the printing speed is really slow. Thats why it was doing for me only on the 1st layer.
For example this snippet from the gcode with F around 500 creates that irregular pattern.
`G1 X-8.711 Y3.308 E0.0449 F540`
Exactly the same firmware settings but if instead I make the print faster (F > 1000), the problem goes away.
Maybe the problem is not about processing speed?

If no one else can reproduce this could it be my stepper drivers?

thinkyhead, 2016-05-09:
> …a rewrite of kinematics…

That will be fun. But in the meantime the optimizations described are probably the best ones at the moment. Reducing the screen refresh rate to 5Hz is fine. There isn't anything that needs to update on the screen more often than that. Reducing segments-per-second works, but it can make the movement a little jagged for long-and-fast moves, so you don't want to go too low.

> not about processing high speed but lower speeds?

More experimentation and observation will help to make more sense of this, I hope!

karabas, 2016-05-10:
On my deltas if I go higher than 100-110 segments, printer stutters. I use only nongraphic displays and old RichieCattell's Marlin version (because of autocalibration). First layer speed 20-25

tkurbad, 2016-05-12:
@AnHardt has an old lcd_speedup branch in his fork, I've been able to adapt to latest RC. His method **drastically** reduces LCD related stuttering, blobs and the like.

My Deltaprintr works smoothly with `DELTA_SEGMENTS_PER_SECOND` set to 200 and speeds beyond 80mm/s.

Unfortunately, AnHardt states that he doesn't have the time to properly adapt his branch to newer Marlin versions atm... So, I'm trying in my RC_Deltaprintr branch.

tkurbad, 2016-05-12:
Regarding rewriting the kinematics, this could be of help:

https://en.wikipedia.org/wiki/CORDIC

AnHardt, 2016-05-12:
@tkurbad 
Too many and too drastically changes for an RC. Refreshing https://github.com/AnHardt/Marlin/pull/13 now, means doing it twice. (And i already did a couple of times)

tkurbad, 2016-05-12:
@AnHardt No problem,I fully understand you. Just wanted to hint people who happen to switch off the status screen during print (as I did before) that there's a much nicer solution around that just needs some work.
Btw., my RC_Deltaprintr branch is just an enriched RC6 that can be used on any printer besides Deltaprintr (care has to be taken by Azteeg_X3 board users, however, since Deltaprintr uses a board wth modified pinout). All the other specifics of the Deltaprintr (like FSR bed leveling) are enabled by additional #defines in the Configuration.h. Take them out and you have RC6 with a super-fast LCD... It's now in use by at least 4 Deltaprintr owners - and they all run it successfully.

closed 2016-05-30 by jbrazio