MarlinFirmware/Marlin #3693: Why autobedleveling not work in big areas?

fbarcenas, 2016-05-07:
Hi,
my machine have a big area of 1200x900mm,
If i test in 200x200 or 300x300 the autobedleveling do a perfect job, but if i test with all area the system not work well... When must up the nozzle, down...

I read some about this in another post, i don't know if this can be a enhancement or that.
What is the opinion?

boelle, 2016-05-07:
how many points are you using for abl ?

fbarcenas, 2016-05-07:
I test with 3, 4 and 5.
And always fail.

fbarcenas, 2016-05-07:
Always fail in the big dimensions,
In low dimension always is perfect

Blue-Marlin, 2016-05-07:
Because most machines do not fulfil the assumptions the algorithms make about the machine. 
Because on big machines it's even more difficult to keep the errors small.

3-point- and grid- ABL are based on the assumptions:
* The machine is infinity stiff - no sag of the axes.
* The machine has perfect geometry - no twist in the axes, straight axes.
* the bed is perfectly flat.
* the one and only _allowed_ error is a slightly tilt bed. That's what it tries to correct.

Mesh- ABL makes less assumptions, but suffers from the limited number of mesh points we can take. On big machines the distances between the mesh-ponts grow - so the errors. 

The big advantage of MBL, in the original implementation, is probing directly with the nozzle. That means no probe_offsets. Probe offsets can amplify some of the geometry errors.

boelle, 2016-05-07:
try with 3 point only, not grid or mesh

fbarcenas, 2016-05-07:
I test and nothing...

Actually i use auto bed leveling not for leveling the bed, i use for than the Z0 is always ok, because use the probe sensor!!

Roxy-3DPrintBoard, 2016-05-07:
Something else to check is how good your Z-Probe measurements are.  If you don't have enough repeatable precision on your Z-Probe your Auto Bed Leveling experience is not going to be good.

Do an M48 in the center of the bed and you should get a Standard Deviation of less than .025 mm if you are using .3mm layers.  For thinner layers, or bigger beds you are going to need to be under .005 mm to be happy.

fbarcenas, 2016-05-07:
The precision of the z problem is really good, i do more than 10 times to test and always do the same distance +/-0.01

boelle, 2016-05-08:
that is bad, you need to get it to 0.005

if your probe is an inductive one then its not sensitive enough

jbrazio, 2016-05-08:
Not optimal, but you may try to compensate on the slicer by setting the first layer extrusion width to 120%.

thinkyhead, 2016-05-09:
Automated Mesh Leveling is still a work-in-progress, but in the meantime you can try Manual (Mesh) Bed Leveling and see how well it works. Once we automate the process it will be quicker, but doing it manually should give you very good measurement accuracy.

jbrazio, 2016-05-16:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-16 by jbrazio

fbarcenas, 2016-05-16:
I will test the actual RCBUGFIX and if is necessary i will reopen this.
Thanks for everything!