MarlinFirmware/Marlin #3694: Marlin 1.0.2.13 doesn't complie if change Board

djibb, 2016-05-07:
Arduino 1.6.5
Marlin 1.0.2.13
First : Compile Marlin as it. -> works
Change :   //#define MOTHERBOARD BOARD_RAMBO
  #define MOTHERBOARD 33
Try to compile : 
pins.h:2968: error: 'Z_PROBE_PIN' was not declared in this scope

Any idea ?

jbrazio, 2016-05-07:
Please test with the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch.
I've tested by taking a vanilla `Configuration.h` and change:
```cpp
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMBO
#endif
```

And it compiles:
```
jbrazio@dagobah:~/Documents/src/Marlin/Marlin (RCBugFix) $ arduino --verify Marlin.ino 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...

Sketch uses 54,376 bytes (21%) of program storage space. Maximum is 253,952 bytes.
Global variables use 2,966 bytes (36%) of dynamic memory, leaving 5,226 bytes for local variables. Maximum is 8,192 bytes.
```

Roxy-3DPrintBoard, 2016-05-08:
> Please test with the latest RCBugfix branch.
> I've tested by taking a vanilla Configuration.h and change:

@jbrazio   This is a problem associated with Auto Bed Leveling.   Try taking the vanilla Configuration.h and turning Auto Bed Leveling on.   That might make it happen.

jbrazio, 2016-05-08:
@Roxy-3DPrintBoard 

Before writing my answer I did the following, `Z_PROBE_PIN` seems to be not used anywhere on the latest `RCBugfix`.
```
jbrazio@dagobah:~/Documents/src/Marlin/Marlin (RCBugFix) $ grep Z_PROBE_PIN * -R
jbrazio@dagobah:~/Documents/src/Marlin/Marlin (RCBugFix) $
```

Nevertheless I tried the following config.

```cpp
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMBO
#endif

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)

// A fix mounted probe, like the normal inductive probe, must be deactivated to go below Z_PROBE_OFFSET_FROM_EXTRUDER
// when the hardware endstops are active.
#define FIX_MOUNTED_PROBE
```

```
jbrazio@dagobah:~/Documents/src/Marlin/Marlin (RCBugFix) $ arduino --verify Marlin.ino 
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...

Sketch uses 78,814 bytes (31%) of program storage space. Maximum is 253,952 bytes.
Global variables use 3,111 bytes (37%) of dynamic memory, leaving 5,081 bytes for local variables. Maximum is 8,192 bytes.
```

Roxy-3DPrintBoard, 2016-05-08:
So Z_PROBE_PIN doesn't exist any more?   I did a grep too just to double check.  It didn't show up.  

@djibb you need to figure out where that is coming from.   It isn't in RCBugFix!

Blue-Marlin, 2016-05-08:
`Z_MIN_PROBE_PIN` in RCBugFix not `Z_PROBE_PIN`

thinkyhead, 2016-05-08:
@djibb If you don't "need" to use 1.0.2, then I suggest trying `RCBugFix` to see how it does. We're tracking a possible communication bug, but otherwise it has a lot of fixes over 1.0.2.

closed 2016-05-15 by jbrazio