MarlinFirmware/Marlin #3695: Losing connection when servo is put down

haondt, 2016-05-07:
Recently upgraded to RCBugFix and added a Z probe to my prusa i3. When I tried doing a G28, the Printer would home x and and y and deploy the z probe and then freeze. Upon further investigation, I found out even Just deploying the probe, i.e M401 causes the printer to freeze. It deploys the probe, then instantly moves it back up then stops communcating with the host.

jbrazio, 2016-05-07:
Which type of probe are you deploying ?
Could you please attach (by drag and drop here) a copy of you config files ?

haondt, 2016-05-07:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/253683/Configuration.zip)

Servo with a mechanical endstop on it

Blue-Marlin, 2016-05-08:
Servo moves -> draws to much current -> voltage drops -> reset?

haondt, 2016-05-08:
i'd be surprised, considering this happens even when the motors are not moving, yet it can run 5 steppers at once

Blue-Marlin, 2016-05-08:
The steppers are connected to at least 12V i hope.
Depending on the servo, feeding them with 5V from the Arduino, may work or not. A separate 5V power supply is on the safe side.

haondt, 2016-05-08:
This seems to have fixed the issue! fantastic! However, this is not the first time I've attempted to get the auto level working on my printer, I've had the servo working properly before without having the need for a second power supply. Any suggestions on getting this working without an auxiliary power supply?

Blue-Marlin, 2016-05-08:
Was the servo moving free, or did it hit the bed. If hitting the bed increase `Z_RAISE_BEFORE_PROBING` (`Z_RAISE_AFTER_PROBING` should be the same). If the servo can move, it draws less current.

haondt, 2016-05-08:
Its completely free, it doesnt run into anything sitting at or moving between its two angles

Blue-Marlin, 2016-05-08:
Same board? Same servo? Other added 5V devices?
Sorry, no further ideas.
My servos just work.

haondt, 2016-05-08:
Same board, same servos, everything was the same. The only difference was that between then and now i cleaned up all of the wiring but all the wires are exactly where they used to be. Nonetheless, now that I know exactly what is causing the problem, i can come up with some sort of workaround. Thank you so much!

CONSULitAS, 2016-05-08:
@haondt 
Try connecting a electrolytic capacitor (at least 470 uF) to the servo (from GND to 5V). Take care to connect it right (- to ground).

The servo draws current only for a short time. So a capacitor could help stabilizing the voltage.

nomukaiki, 2016-05-08:
I had the same issue using a [BLTouch](https://plus.google.com/u/0/113792662927481823969), but even worse because it was draining too much current every time the sensor touched the bed, making the processor hang and eventually crash.
Completely fixed it by adding an additional 5V power supply.

The suggestion of @CONSULitAS is definitely a great idea and I will try it out!

Grogyan, 2016-05-08:
My Bltouch probe works fine without an additional power supply

nomukaiki, 2016-05-08:
My BLTouch probe also works fine and was not the problem, sorry if I wasn't clear about that.
Because my 5V circuit is already too saturated, the 300mA of the BLTouch was already enough to drain power from the processor and reset it. I don't know how much amps can the Arduino provide in the 5V circuit.
That why the suggestion of using a power supply only for the servos is a good idea.

jbrazio, 2016-05-08:
Arduino pins should not be used to power anything with more current draw than s LED.. this is generally a good role of thumb. :-)

If you want a quick hack, go to the three dollar store and buy one of those car lighter-plug USB chargers, most of them they are 500mA 12V to 5V switching converters.

Roxy-3DPrintBoard, 2016-05-08:
> Servo moves -> draws to much current -> voltage drops -> reset?

This is a well know problem.   For a lot of boards the +5 Volt line supplied by the board is not enough for the initial surge current that the servo produces.   You don't say what kind of board you have but RAMPS boards in particular have this problem.   Most people take a LM7805 and feed the +12 Volts into it to get a very strong (and isolated) +5 Volt supply for the servo.  

You can get 10 of these for $1.55 on eBay  (or 1 for $.99)   
http://www.ebay.com/itm/NEW-10pcs-L7805CV-LM7805-7805-Voltage-Regulator-ST-TO-220-ZH175-/201576577347?hash=item2eeee67943:g:4XoAAOSw5HJXKUWC

haondt, 2016-05-08:
It's a RAMPS 1.4 Board on an Arduino Mega. I like the idea @jbrazio  suggested of connecting the servo to the power supply with a cheap dollar store car lighter usb chargers. Will definitely give that a try!

jbrazio, 2016-05-08:
As this is not directly related with Marlin I will now close the issue.
You may open it at any time if you require more help.

closed 2016-05-08 by jbrazio