MarlinFirmware/Marlin #3696: Thermal Runaway debugging code doesn't compile

Grogyan, 2016-05-08:
Using RAMPS 1.4 
RCBugFx 8-5-2016

In temperature.cpp
Firmware does not compile with Arduino 1.6.8 with this uncommented
``` SERIAL_ECHO_START;
        SERIAL_ECHOPGM("Thermal Thermal Runaway Running. Heater ID: ");
        if (heater_id < 0) SERIAL_ECHOPGM("bed"); else SERIAL_ECHOPGM(heater_id);
        SERIAL_ECHOPGM(" ;  State:");
        SERIAL_ECHOPGM(*state);
        SERIAL_ECHOPGM(" ;  Timer:");
        SERIAL_ECHOPGM(*timer);
        SERIAL_ECHOPGM(" ;  Temperature:");
        SERIAL_ECHOPGM(temperature);
        SERIAL_ECHOPGM(" ;  Target Temp:");
        SERIAL_ECHOPGM(target_temperature);
        SERIAL_EOL;```

esenapaj, 2016-05-08:
When I change the code to follows, it seems that it works fine.
But I can't be determined that it's impeccable.

```
        SERIAL_ECHO_START;
        SERIAL_ECHOPGM("Thermal Thermal Runaway Running. Heater ID: ");
        if (heater_id < 0) SERIAL_ECHO("bed"); else SERIAL_ECHO(heater_id);
        SERIAL_ECHOPGM(" ;  State:");
        SERIAL_ECHO(*state);
        SERIAL_ECHOPGM(" ;  Timer:");
        SERIAL_ECHO(*timer);
        SERIAL_ECHOPGM(" ;  Temperature:");
        SERIAL_ECHO(temperature);
        SERIAL_ECHOPGM(" ;  Target Temp:");
        SERIAL_ECHO(target_temperature);
        SERIAL_EOL;
```

Blue-Marlin, 2016-05-08:
@esenapaj 
Looks good for me and compiles.

Variable can not reside in the program space. They do always reside in RAM. So SERIAL_ECHOPGM is wrong.
An alternative would be to use `SERIAL_ECHOPAIR(name,value) (serial_echopair_P(PSTR(name),(value)))`

esenapaj, 2016-05-08:
I gratitude for your explanation.
I feel that ```SERIAL_ECHOPAIR``` is smarter than ```SERIAL_ECHOPGM``` + ```SERIAL_ECHO```.
Thus I've rewritten the above code that it used ```SERIAL_ECHOPAIR```,
and have submitted it as PR #3701.

closed 2016-05-08 by thinkyhead