MarlinFirmware/Marlin #3697: LCD Z offset limits

A-Nesh, 2016-05-08:
Where do i find and change limits for Z OFFSET for the LCD? 

I would like to change Z offset to -0.2 but u hit the limit in (-)000.50 in the LCD and it wont go below that.
I can set the limit by sending M851 and it works great. The new value shows on the LCD but it jumps up to 000.50 when i'm trying to change it.

```
  #define CUSTOM_M_CODE_SET_Z_PROBE_OFFSET 851
  #define Z_PROBE_OFFSET_RANGE_MIN -15
  #define Z_PROBE_OFFSET_RANGE_MAX 15
```

A-Nesh, 2016-05-08:
Found it!
in ultralcd.cpp, line 847

`    MENU_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, 0.5, 50);`

closed 2016-05-08 by A-Nesh

Blue-Marlin, 2016-05-08:
In the _newer_ Marlins this is already looking like https://github.com/MarlinFirmware/Marlin/blob/dee6322bf012c117f4fbb434aaf775ddaffc3334/Marlin/ultralcd.cpp#L1630
```
MENU_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX);
```