MarlinFirmware/Marlin #3698: Where is the option to home in a single axis?

Drmaestro, 2016-05-08:
I had this option in my older Marlin version, where I could choose an axis to home (Home X, Home Y and Home Z). Now in RC6 I can only see Auto home and I haven't been able to find the option in config.h. Is it somewhere else? Is it removed?

Thanks.

Blue-Marlin, 2016-05-08:
I'v never seen this in MarlinFirmware/Marlin or ErikZalm/Marlin. So likely you used an other fork before.

Drmaestro, 2016-05-08:
This could be a feature request then? I find it very useful the option to home the Z axis anywhere over the bed. This helps me find if there is a difference of height between two Z elevators (if they are level). Auto home goes to a very specific location, so that's not useful for this purpose.

boelle, 2016-05-08:
G28 Z 

??

Drmaestro, 2016-05-08:
Of course, but I'd like to have this option on the LCD menu. I don't want to use a USB connection.

jbrazio, 2016-05-08:
There is an option which allows you to "select" the order from which the axis are homed, but I confirm @Blue-Marlin, I don't remember seeing this as a specific LCD menu on Marlin.

thinkyhead, 2016-05-09:
If you want to just toss it in yourself…
```diff
   //
   // Auto Home
   //
   MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
+  MENU_ITEM(gcode, "Home Z", PSTR("G28 Z"));
```

Drmaestro, 2016-05-09:
Thanks thinkyhead, that's what I was looking for.

CONSULitAS, 2016-05-09:
@Drmaestro The answer of @thinkyhead seems to have solved your issue.

So i am closing this for house keeping.

If your issue persists, feel free to reopen it.

closed 2016-05-09 by CONSULitAS

gddeen, 2016-05-10:
Just to point out, I think executing that can turn off mesh bed leveling... (G29) and it has to be
turned back on... or releveled.

thinkyhead, 2016-05-10:
@gddeen With mesh leveling, `G28` actually remembers whether you had it on and re-enables it when it's done. Not so with Auto Bed Leveling, however.