MarlinFirmware/Marlin #3700: Mesh bed levelling now crashes nozzle into bed after homing

brainscan, 2016-05-08:
Hi just tried out rcbugfix, after using RC3 without any problems, and now when I home z it bumps the endstop but rather than backing off again it now drives the nozzle down into the bed. I've been using mesh bed levelling, again without problems, and the behaviour now seems totally different to what I was expecting. Before it would home all axis then move inward by 10 mm for both X and Y and Z would rise away from bed to a safe position. Now X and Y don't move and Z just drives nozzle into bed. I'm any help would be much appreciated.

tiagom62, 2016-05-08:
Please post your configuration.

brainscan, 2016-05-08:
Same settings I've been using with RC3
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/254087/Configuration.txt)

epatel, 2016-05-08:
Maybe you need to enable `MIN_Z_HEIGHT_FOR_HOMING`

See https://github.com/MarlinFirmware/Marlin/pull/3488

tiagom62, 2016-05-08:
Do you have a z endstop and a z probe?

brainscan, 2016-05-08:
@epatel Thank you I will try that and see if it makes things better. It does not work as it previously did and I thought that those settings were only linked to the other bed levelling routines.

@tiagom62 no just endstop but it seems that MBL has been changed since my version and using the same settings causes unwanted behavior

tiagom62, 2016-05-08:
Possibly Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled is the cause?

brainscan, 2016-05-08:
Unfortunately no I changed both of those and it's made no difference. Still bumps end stop then drives past it into the bed. Like I said it used to home and then place the nozzle in a good position to heat up, now not so much.

Roxy-3DPrintBoard, 2016-05-08:
Is this related to #3675

thinkyhead, 2016-05-09:
@brainscan Try the latest `RCBugFix` and see if your configuration triggers any errors in `SanityCheck.h`.

brainscan, 2016-05-10:
Hi, I just downloaded the latest RCBugFix and still have the same issue. Mesh used to signal that it was active at the end of G28 by moving X and Y to the first probe point but kept Z where it found the end stop. This keeps the nozzle clear of any problems and was set up using the original documentation, I had no problems getting it working when first introduced. Now when I send a G28 it homes X and Y then Z comes down hits the end stops, pushes past it and down into bed. Sorry if I missed something but I haven't spotted any notes to say the behaviour would change and potentially break stuff.

thinkyhead, 2016-05-11:
> Sorry if I missed something

@brainscan Could be a bug. I wish I had more time to do testing, troubleshooting, narrow it down, and figure it out. But it will have to wait some days, I'm afraid. Maybe someone else has time to play with this…?

epatel, 2016-05-14:
Yes, that behaviour has changed. It should not _crash_ into the bed though. It should go to position `0,0,0` 
I added this in this commit https://github.com/MarlinFirmware/Marlin/commit/14afe1a017610e1530185b40bd2676e45695cbb7
This was added after a request long ago, someone thought the original behaviour was strange. Still, it should not _crash_ into the bed.

brainscan, 2016-05-14:
> I added this in this commit 14afe1a

Ah yes that will be it, unfortunately I set my printer based on the original configuration of mbl, so everything was safe when homing. Now the behaviour has changed it smashed into the mounting clip. Sorry I didn't notice any new documentation stating this had changed but if that's how it is now I'll just go back to my old firmware. This used to be so easy to set up.

brainscan, 2016-05-20:
Fixed with #3802 thanks everyone

closed 2016-05-20 by brainscan