MarlinFirmware/Marlin #3703: False thermal runaway detect

MikeInMinnesota, 2016-05-08:
On both the RC6 and RCBugFix branches, I'm seeing false thermal runaway detection during initial startup of a print.  The cause is the timer isn't initialized when going from thermal monitor state TRFirstHeating to TRStable.  If the extruder is preheated before printing, or printing is restarted then there is a tendency to go from a temperature above the target to below the target minus hysteresis and the timeout gets immediately triggered rather than waiting the required timeout.  The extruder temperature drops fairly rapidly when not driven.  That happens on my printer because the extruder gets shutoff at the start of a print and doesn't get heated until after the bed reaches its target temperature.  This issue is especially apparent if the bed starts at the target temperature because there is a small delay between recognizing the bed is at the desired temperature and starting the extruder heating.

I've tested a one-line fix that initializes the timer in the TRFirstHeating state.  I can submit that if desired.

thinkyhead, 2016-05-08:
That would be great. You should get the credit for this find!

thinkyhead, 2016-05-08:
I'm trying to just comprehend what the bug might be, just scanning the code, and so far I can't see the logical hole.

At the beginning of the function, we check to see if the target temperature has been changed at all, and go into a reset state if it has:
```cpp
// If the target temperature changes, restart
if (tr_target_temperature[heater_index] != target_temperature)
  *state = TRReset;
```
This causes the timer to be cleared (set to zero) and the state to immediately change to `TRInactive`. It falls through to do the first `TRInactive` check, where if the temperature is anything over zero, it saves the target temperature (to be able to tell when the target temp is changed), goes into `TRFirstHeating` stage, and then breaks out of the `switch` block.
```cpp
case TRReset:
  *timer = 0;
  *state = TRInactive;
// Inactive state waits for a target temperature to be set
case TRInactive:
  if (target_temperature > 0) {
    tr_target_temperature[heater_index] = target_temperature;
    *state = TRFirstHeating;
  }
  break;
```
On the next invocation, assuming the temperature doesn't change, it will be in `TRFirstHeating` state, where it waits for the target to be reached and sets `TRStable` state. It then exits.
```cpp
// When first heating, wait for the temperature to be reached then go to Stable state
case TRFirstHeating:
  if (temperature >= tr_target_temperature[heater_index]) *state = TRStable;
  break;
```
As you point out the `timer` value hasn't been set at this point. But according to my reading of the code, the next invocation should _necessarily_ set the timer because the condition it tests for is actually looser than the condition that it had to pass to get into `TRStable` state in the first place:
```cpp
// While the temperature is stable watch for a bad temperature
case TRStable:
  // If the temperature is over the target (-hysteresis) restart the timer
  if (temperature >= tr_target_temperature[heater_index] - hysteresis_degc)
    *timer = millis();
  // If the timer goes too long without a reset, trigger shutdown
  else if (ELAPSED(millis(), *timer + period_seconds * 1000UL))
    *state = TRRunaway;
  break;
```
Are we mistaken in assuming that these two tests can actually have different results? :
```cpp
temperature >= tr_target_temperature[heater_index]
temperature >= tr_target_temperature[heater_index] - hysteresis_degc
```

thinkyhead, 2016-05-08:
Rather than just a one-line change to initialize the timer, I would propose instead to make the switch block a bit "smarter," basically by having it fall-through every time a state change occurs, instead of waiting for the next invocation. Like this:
```cpp
switch (*state) {
  case TRReset:
    *timer = 0;
    *state = TRInactive;
  // Inactive state waits for a target temperature to be set
  case TRInactive:
    if (target_temperature <= 0) break;
    tr_target_temperature[heater_index] = target_temperature;
    *state = TRFirstHeating;
  // When first heating, wait for the temperature to be reached then go to Stable state
  case TRFirstHeating:
    if (temperature < tr_target_temperature[heater_index]) break;
    *state = TRStable;
  // While the temperature is stable watch for a bad temperature
  case TRStable:
    if (temperature < tr_target_temperature[heater_index] - hysteresis_degc && ELAPSED(millis(), *timer)) {
      *state = TRRunaway;
    }
    else {
      *timer = millis() + period_seconds * 1000UL;
      break;
    }
  case TRRunaway:
    _temp_error(heater_id, PSTR(MSG_T_THERMAL_RUNAWAY), PSTR(MSG_THERMAL_RUNAWAY));
}
```
This should force `*timer` to be initialized, because it falls-through to the temperature hysteresis check right away, giving no chance for `temperature` to change, as it could in-between invocations.

MikeInMinnesota, 2016-05-09:
Yes, the problem is the first test of:
if (temperature >= tr_target_temperature[heater_index]) *state = TRStable;
is true, but the 2nd test of:

                    if (temperature >= tr_target_temperature[heater_index] - hysteresis_degc)
is false.  This occurs because the actual extruder temperature is rapidly falling and has dropped below the hysteresis level.  The heater control hasn’t had time to bring the temperature back up to the set level.

My fix was:

case TRFirstHeating:

        if (temperature >= tr_target_temperature[heater_index]) *state = TRStable;

        *timer = millis();

        break;
Resetting the timer before the TRStable state insures we will get the full period_seconds timeout.

As an aside, the trigger for this condition is related to the fact that we turn off the extruder/bed and then check the bed temperature.  I’m using Slic3r 1.2.9, and it generates an M190 for the bed, followed by an M104 for the extruder.  I suspect this is another variant of the old M109 blocking condition until the temperature is reached.

Regards,
Mike

Note from PRGA.COM: This message has been scanned for SPAM.

Click here to mark email as junk<https://mail.quarantine.prga.com:8444/FrontController?operation=mbeu&f=00001_49_20160508_4254677.eml&chkBayesian=1&pr=1&mt=1&ma=s>.

-----------------------------------------------------------

thinkyhead, 2016-05-09:
Thanks for the extra feedback. That makes perfect sense. The PR #3705 should have the same effect, by falling through and doing the test on the same invocation. Naturally I found other ways to reduce the code a little more, as it was a little "over-designed."

jbrazio, 2016-05-10:
Closed with #3705.

closed 2016-05-10 by jbrazio