MarlinFirmware/Marlin #3704: Compile error for SAV MK1

boelle, 2016-05-08:
in ArduinoAddons/Arduino_1.6.x i found that there are support for my board

i copy the marlin folder from there to my hardware folder as i would with any board.

when i compile i get:

WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
sketch\libraries\twi.c:1:17: fatal error: twi.c: No such file or directory

 #include <twi.c>

                 ^

compilation terminated.

exit status 1
Error compiling for board SAV MkI (CDC).

boelle, 2016-05-08:
upps... IDE had repetier firmware open as default, but its more or less the same with marlin:

WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from sketch\pins.h:105:0,

                 from sketch\Conditionals.h:270,

                 from sketch\Configuration_adv.h:688,

                 from sketch\Configuration.h:1129,

                 from C:\Users\Bo\Desktop\Marlin\Marlin.ino:37:

pins_SAV_MKI.h:31: error: #error Oops! Make sure you have 'Teensy++ 2.0' selected from the 'Tools -> Boards' menu.

   #error Oops!  Make sure you have 'Teensy++ 2.0' selected from the 'Tools -> Boards' menu.

    ^

In file included from sketch\Marlin.h:43:0,

                 from sketch\thermistortables.h:26,

                 from sketch\Configuration.h:1130,

                 from C:\Users\Bo\Desktop\Marlin\Marlin.ino:37:

fastio.h:4041: error: #error pins for this chip not defined in arduino.h! If you write an appropriate pin definition and have this firmware work on your chip, please submit a pull request

   #error pins for this chip not defined in arduino.h! If you write an appropriate pin definition and have this firmware work on your chip, please submit a pull request

    ^

exit status 1
#error Oops! Make sure you have 'Teensy++ 2.0' selected from the 'Tools -> Boards' menu.

boelle, 2016-05-08:
hmm... need to figure why it wants Teensy++ 2.0 and not SAV MK1

Roxy-3DPrintBoard, 2016-05-08:
I forget what the reason is...   But the PrintRboard requires you select Teensy in order to compile and flash the board.   I bet the reason your board is saying this is the same reason the PrintRboard says that.   If you learn more about what is going on, please post it!

boelle, 2016-05-08:
in hardware\marlin\avr\variants\at90usb\pins_arduino.h

__AVR_ATmega32U4__ is the only one.... could this be it?

boelle, 2016-05-08:
yep.... if i select the teensy++ 2.0 and not SAV MK1 it compiles and opens the teensy loader

so that makes the addons a bit not needed for now....

boelle, 2016-05-08:
damm.... 

tired so off to bed with me, more time tomorrow to poke arround in this

thinkyhead, 2016-05-09:
@boelle [Three back-ticks](https://guides.github.com/features/mastering-markdown/) for multi-line code quoting, please!

boelle, 2016-05-09:
well i can live with selecting teensy++ 2.0 if that makes it work

thinkyhead, 2016-05-09:
@jbrazio We should make sure this is covered in the official documentation. We should just do a page for each of the boards we support – as specified in the `boards.h` file.

emartinez167, 2016-05-09:
@jbrazio, I am looking after the Rigidbot in this forum and we have a wiki we have been maintaining to keep anything related to those machines. It would be great for me to be able to contribute the specifics for that machine to the official documentation. Please let me know how to proceed.

Regards,

Ernesto

jbrazio, 2016-05-10:
@emartinez167 Please have a look at [MarlinFirmware/MarlinDocumentation](https://github.com/MarlinFirmware/MarlinDocumentation) if you require assistence please open a issue request on that repo.

@thinkyhead This is the [current list](http://www.marlinfw.org/articles/hardware/boards.html) of supported boards, would you like to have specific configuration options for each group of them ? I require to know in fact which boards do we *actually* support.

boelle, 2016-05-10:
@Roxy-3DPrintBoard... so you compiled and flashed from IDE 1.6.8 ?

boelle, 2016-05-12:
@Roxy-3DPrintBoard alive?

Roxy-3DPrintBoard, 2016-05-12:
I have not ventured into the IDE 1.6.8 world yet with my PrintRboard.    I've been super busy printing Glowing Skull Night Lights!   

[image]

closed 2016-05-24 by jbrazio