MarlinFirmware/Marlin #3709: 1.0.x stable gives an error when compiling with Arduino 1.6.8

agorsak, 2016-05-09:
The stable branch gives the following error when compiling with latest stable Arduino 1.6.8:

sketch\Marlin_main.cpp:2668:36: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

         LCD_MESSAGEPGM(MACHINE_NAME" "MSG_OFF".");

                                                                      ^

Blue-Marlin, 2016-05-09:
We already know about that. In the current code it looks like
```
LCD_MESSAGEPGM(MACHINE_NAME " " MSG_OFF ".");
```
What you get is a warning, not an error. Ignore it or correct it like above.

CONSULitAS, 2016-05-09:
@agorsak Your issue should be solved by the feedback of @Blue-Marlin.

I close it for housekeeping.

BTW: We would recommend to use our latest RCBugFix branch, because there are a lot of other bugs fixed, too.

And if you don't mind waiting: There should be a 1.1.0 release rather soon.

closed 2016-05-09 by CONSULitAS

agorsak, 2016-05-10:
Hello guys,

thanks for responses and pointing out I should use RCBugFix branch, but I do wonder what is the point of having a "stable" branch which is not completely stable and is in effect stale and fixing the bugs in a branch that is being developed. 
If you prefer to work this way you should at least point this out in the README, so that users who want a stable version know what to expect when using either 1.0.x or RCBugFix.

BR, Andraž

jbrazio, 2016-05-10:
Andraž, the current stable Marlin 1.0.x will soon be replaced by RCBugfix which is now in the 6th RC cycle, we do consider it very stable. As soon as we finish this cycle and publish 1.1.0 we will stop supporting completely the 1.0.x.

Roxy-3DPrintBoard, 2016-05-10:
>  but I do wonder what is the point of having a "stable" branch which is not completely stable and is in effect stale and fixing the bugs in a branch that is being developed. 

Andraž, the problem is you are missing some of the history.   Marlin never has had a Stable and Development branch of the tree.   This Release Candidate Cycle (which we are very far into) is an attempt to get to that mode.   But because we are making changes to the Release Candidate to fix bugs (or mostly issues at this point), it is a way to provide some stability in the interim.   

Soon...   There will be a Stable Release that embodies everything you expect of something with that name.   But we are not quite there yet.

thinkyhead, 2016-05-10:
The term "stable" means the features are unchanging, and that the configuration is well-defined. It is not a synonym for "bug-free."

agorsak, 2016-05-10:
Hey guys,

again thanks for the responses and clarifications. I do value them. 
Keep up the good work.

BR, Andraž