MarlinFirmware/Marlin #3714: Home without Zeroing

chevellebro1, 2016-05-09:
I would like to run the homing sequence without zeroing the machine. I can't find in the code where there is a command to zero when an endstop is hit. Can someone shed some light on this?

```
#define HOMEAXIS_DO_LIDAR(LETTER) \
  ((LETTER##_MIN_PIN > -1 && LETTER##_HOME_DIR==-1) || (LETTER##_MAX_PIN > -1 && LETTER##_HOME_DIR==1))

  if (axis==X_AXIS ? HOMEAXIS_DO_LIDAR(X) :
      axis==Y_AXIS ? HOMEAXIS_DO_LIDAR(Y) :
      axis==Z_AXIS ? HOMEAXIS_DO_LIDAR(Z) :
      0) {

    int axis_home_dir = home_dir(axis) * -1;

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);

    destination[axis] = 1.5 * max_length(axis) * axis_home_dir;
    feedrate = 500;

    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
    destination[axis] = -home_retract_mm(axis) * axis_home_dir;
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    destination[axis] = 2*home_retract_mm(X_AXIS) * axis_home_dir;
    feedrate = 100;

    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    axis_is_at_home(axis);
    destination[axis] = current_position[axis];
    feedrate = 0.0;
    endstops_hit_on_purpose();
    axis_known_position[axis] = true;

    }
}

Blue-Marlin, 2016-05-09:
That's not code from MarlinFirmware/Marlin. We do not have a macro named `HOMEAXIS_DO_LIDAR`.
```
current_position[axis] = 0;
plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
```

chevellebro1, 2016-05-09:
Sorry I copied the wrong set of code. 

```
static void homeaxis(int axis) {
#define HOMEAXIS_DO(LETTER) \
  ((LETTER##_MIN_PIN > -1 && LETTER##_HOME_DIR==-1) || (LETTER##_MAX_PIN > -1 && LETTER##_HOME_DIR==1))

  if (axis==X_AXIS ? HOMEAXIS_DO(X) :
      axis==Y_AXIS ? HOMEAXIS_DO(Y) :
      axis==Z_AXIS ? HOMEAXIS_DO(Z) :
      0) {
    int axis_home_dir = home_dir(axis);
#ifdef DUAL_X_CARRIAGE
    if (axis == X_AXIS)
      axis_home_dir = x_home_dir(active_extruder);
#endif

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);

#ifndef Z_PROBE_SLED
    // Engage Servo endstop if enabled
    #ifdef SERVO_ENDSTOPS
      #if defined (ENABLE_AUTO_BED_LEVELING) && (PROBE_SERVO_DEACTIVATION_DELAY > 0)
        if (axis==Z_AXIS) {
          engage_z_probe();
        }
	    else
      #endif
      if (servo_endstops[axis] > -1) {
        servos[servo_endstops[axis]].write(servo_endstop_angles[axis * 2]);
      }
    #endif
#endif // Z_PROBE_SLED
    destination[axis] = 1.5 * max_length(axis) * axis_home_dir;
    feedrate = homing_feedrate[axis];
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
    destination[axis] = -home_retract_mm(axis) * axis_home_dir;
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    destination[axis] = 2*home_retract_mm(axis) * axis_home_dir;
#ifdef DELTA
    feedrate = homing_feedrate[axis]/10;
#else
    feedrate = homing_feedrate[axis]/2 ;
#endif
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();
#ifdef DELTA
    // retrace by the amount specified in endstop_adj
    if (endstop_adj[axis] * axis_home_dir < 0) {
      plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
      destination[axis] = endstop_adj[axis];
      plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
      st_synchronize();
    }
#endif
    axis_is_at_home(axis);
    destination[axis] = current_position[axis];
    feedrate = 0.0;
    endstops_hit_on_purpose();
    axis_known_position[axis] = true;

    // Retract Servo endstop if enabled
    #ifdef SERVO_ENDSTOPS
      if (servo_endstops[axis] > -1) {
        servos[servo_endstops[axis]].write(servo_endstop_angles[axis * 2 + 1]);
      }
    #endif
#if defined (ENABLE_AUTO_BED_LEVELING) && (PROBE_SERVO_DEACTIVATION_DELAY > 0)
  #ifndef Z_PROBE_SLED
    if (axis==Z_AXIS) retract_z_probe();
  #endif
#endif

  }
}
#define HOMEAXIS(LETTER) homeaxis(LETTER##_AXIS)

jbrazio, 2016-05-10:
I don't understand what do you mean by homing without zeroing, could you please explain it with some context ? Thanks.

chevellebro1, 2016-05-10:
I am using the machine for CNC machining. I have the zero set at the corner of the stock. I would like to reference the home position without reseting the zero to the home switches. This way I can reference home and then return to the zero based on the distance from the home position.

brainscan, 2016-05-10:
Do you just need homing offsets then? Sorry I'm trying to work out why you would want to home but not let the machine know that's where it is.

Grogyan, 2016-05-10:
A G28 -> axis homes and zeros
A G92 -> references the axis by a certain value as the new value for which serves as home
[RepRap Gcodes](http://reprap.org/wiki/G-code)

boelle, 2016-05-10:
hmmm if i understand this correct this could also be used to have the machine tell how long and axis really is.... ie if you have a top mounted z limit switch and you have the nozzle just about right over the bed for 1st layer but you want to know how high Z is then you will just home Z and it will stop at the top and tell how far it did go

jbrazio, 2016-05-10:
@chevellebro1 I believe I already discussed this topic with you a couple of weeks ago (you're the LIDAR guy :-). In a standard CNC system, homing will always create a (0, 0) reference this is known as the "machine zero", when you need to change the (0, 0) reference to your workpiece you should use "offsets" which basically set a new "virtual" (0, 0) in the position you want. On a CNC this is commonly archived by using `G52, G54` and in Marlin try to experiment with [G91](http://reprap.org/wiki/G-code#G91:_Set_to_Relative_Positioning) and [G92](http://reprap.org/wiki/G-code#G92:_Set_Position).

chevellebro1, 2016-05-11:
Can someone explain the difference between absolute and relative positioning? I tried to use this when experimenting. When I change it in the command line I don't notice a difference in the position readout. When does changing this have an effect and how can I see it?

brainscan, 2016-05-11:
Absolute positioning gives coordinates based on your home or zero position. Relative gives coordinates based on the current position. 
I use relative positioning in my end gcode to lift the nozzle away from the print as I only need to tell it to go 5mm higher than its current position, with absolute I'd need to change the code for every print.

jbrazio, 2016-05-11:
Quote the link I sent you:
> A G92 without coordinates will reset all axes to zero.

This means that the current (X,Y) where the nozzle is will become the new (0,0).

closed 2016-05-16 by jbrazio