MarlinFirmware/Marlin #3717: Z-Probe failed to retract! on Endstop it

MartinPeartree, 2016-05-10:
On a Delta with Allen Key Probe. While stowing the probe, as soon as the Z_MIN Endstop get hit the head stop going down and go back up in position failing to stow the key.  It's look like the process of stowing is interpreted as if the probe as hit the bed and cannot go down further. The Z minmax get adjusted and the head cannot go bellow the newly detected Z limit until restarted.

Following error message is shown:

> Error:Z-Probe_ failed to retract!
> [ERROR] Error:Z-Probe failed to retract!
> 
> Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
> [ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
> 
> echo:endstops hit:  Z:246.51

Note that my actual Z limit should be 296.5, not 246.51

My Configuration:

```
#define USE_ZMIN_PLUG  // a Z probe
const bool Z_MIN_ENDSTOP_INVERTING = true; 
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define min_software_endstops true
#define max_software_endstops true 
#define X_PROBE_OFFSET_FROM_EXTRUDER -1
#define Y_PROBE_OFFSET_FROM_EXTRUDER 16
#define Z_PROBE_OFFSET_FROM_EXTRUDER -3.5 
#define Z_RAISE_BEFORE_PROBING 15
#define Z_RAISE_BETWEEN_PROBINGS 5
#define Z_RAISE_AFTER_PROBING 50
#define MANUAL_Z_HOME_POS 296.9 
```

Blue-Marlin, 2016-05-10:
What is you setting for `ENDSTOPS_ONLY_FOR_HOMING`?

MartinPeartree, 2016-05-10:
`#define ENDSTOPS_ONLY_FOR_HOMING`

It's enabled!

Blue-Marlin, 2016-05-10:
That makes it worse. :-(
Then likely our ability to distinguish between z-min-endstp and probe is still/again broken:

https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/endstops.cpp#L304-L337

MartinPeartree, 2016-05-11:
If I dedicate a PIN to the probe, can it bypass the issue?

MartinPeartree, 2016-05-11:
Tested with
`#define ENDSTOPS_ONLY_FOR_HOMING`
Same behavior

MartinPeartree, 2016-05-11:
Having the same behavior with a dedicated PIN for the probe

```
//#define USE_ZMIN_PLUG
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_ENDSTOP
```

Blue-Marlin, 2016-05-11:
What version/branch do you use? If RCBugFix - Is #3672 in or not? That would help us to derermin where to search for the problem.

MartinPeartree, 2016-05-11:
I have a DyzEnd hotend, I run on DyzeDesign/Marlin/tree/DYZE-1.1.0_RC-6
I'll test with RCBugFix, my thermistor won't work but I'l report back on the behavior of the probe.

Blue-Marlin, 2016-05-11:
Looked up https://github.com/DyzeDesign/Marlin/commits/DYZE-1.1.0_RC-6.
Their branch is very clothe to our RC6. So it is not 3672 but before.

MartinPeartree, 2016-05-11:
Yes it's very close to RC6. 
I just tested RCBugFix with 3672 and got the same behavior.

AnHardt, 2016-05-11:
What are your settings for
```
//#define DISABLE_MAX_ENDSTOPS
#define DISABLE_MIN_ENDSTOPS
```
Not having disabled min-endstops could be an other source for the problem on a delta.

MartinPeartree, 2016-05-11:
Ok, I did a funny test, at anytime during stowing, if I trigger the probe it instantly stop going further down. **BUT**, if I cold unplug the Endstop from the motherboard instead it stow successfully. But I get the following error only:

`[ERROR] Error:Z-Probe failed to retract!`

I would have expected manually triggering or unplugging the probe to produce the same behavior.

AnHardt, 2016-05-11:
That's normal. Because a Allen Key Probe is triggered when stowed, we can check if it is successfully stowed. If there is no signal we get this message. Here the pin is tested directly. This has nothing to do with the endstop tests in the IRQ what cause the early stop of stowing the probe and the shift in the coordinate system. Successfully stowing the probe is possible because the z-min-endstop is not trigged.

jbrazio, 2016-05-24:
Closing due to merged patch #3828.

closed 2016-05-24 by jbrazio